ducalex/retro-go #155: Slow SD card file system speed

raphatex, 2024-09-09:
Hey,
I've just downloaded and compiled the latest version of retro-go for my retro-ruler and it seems like SD card speed is much slower than it was on my previous version of retro-go. I was using retro-go-ST7789 fork wich was last updated 2 years ago. 
So I was wondering if it was normal (some kind of pre-loading stuff) or if it wasn't.
For example loading GBC games takes around 20sec instead of 3sec for the 2 years old version.

Thanks in advance !

raphatex, 2024-09-10:
Same version of esp-idf, I'm using SDSPI shared with the display. The problem don't seems to only be with GBC, the whole OS is wayyy slower.
After further inspection it seems like the problem appeared between V1.42 and V1.43 (latest)
does the change in storage definition in the config.h between V1.42 and V1.43 could have caused this problem ?

```
// Storage
#define RG_STORAGE_ROOT             "/sd"
#define RG_STORAGE_SDSPI_HOST       SPI2_HOST
#define RG_STORAGE_SDSPI_SPEED      SDMMC_FREQ_DEFAULT
// #define RG_STORAGE_SDMMC_HOST       SDMMC_HOST_SLOT_1
// #define RG_STORAGE_SDMMC_SPEED      SDMMC_FREQ_DEFAULT
// #define RG_STORAGE_FLASH_PARTITION  "vfs"
```

ducalex, 2024-09-10:
Can you try to narrow it down a bit more? There's way too many changes between 1.42 and 1.43 :(

What I would do is checkout a commit somewhere between the two tags and start from there.

I realize it's a bit annoying because you have custom patches to carry, but imho it will be faster than trying to locate the cause of the slowdown through code alone.

raphatex, 2024-09-10:
No problem, I will try to investigate on my side !

raphatex, 2024-09-13:
So after further inspection it seems like the problem appeared after theses changes, I will check each commit after the weekend but for now I only know that before it works flawlessly and after it become slow. You can start checking if you want or wait for me to edit the post !
jun-28-> OK
jun-29-> not OK
[image]

Also here is my config.h:
```
// Target definition
#define RG_TARGET_NAME             "RETRO-RULER"

// Storage and Settings
#define RG_STORAGE_DRIVER           1                   // 0 = Host, 1 = SDSPI, 2 = SDMMC, 3 = USB, 4 = Flash
#define RG_STORAGE_HOST             SPI2_HOST           // Used by SDSPI and SDMMC
#define RG_STORAGE_SPEED            SDMMC_FREQ_DEFAULT  // Used by SDSPI and SDMMC
#define RG_STORAGE_ROOT             "/sd"               // Storage mount point

// Audio
#define RG_AUDIO_USE_INT_DAC        2   // 0 = Disable, 1 = GPIO25, 2 = GPIO26, 3 = Both
#define RG_AUDIO_USE_EXT_DAC        0   // 0 = Disable, 1 = Enable

// Video
#define RG_SCREEN_DRIVER            0   // 0 = ILI9341
#define RG_SCREEN_HOST              SPI2_HOST
#define RG_SCREEN_SPEED             SPI_MASTER_FREQ_40M
#define RG_SCREEN_BACKLIGHT         1
#define RG_SCREEN_WIDTH             320
#define RG_SCREEN_HEIGHT            240
#define RG_SCREEN_ROTATE            0
#define RG_SCREEN_MARGIN_TOP        0
#define RG_SCREEN_MARGIN_BOTTOM     0
#define RG_SCREEN_MARGIN_LEFT       19
#define RG_SCREEN_MARGIN_RIGHT      19
#define RG_SCREEN_INIT()                                                                                         \
    ILI9341_CMD(0xCF, 0x00, 0xc3, 0x30);                                                                         \
    ILI9341_CMD(0xED, 0x64, 0x03, 0x12, 0x81);                                                                   \
    ILI9341_CMD(0xE8, 0x85, 0x00, 0x78);                                                                         \
    ILI9341_CMD(0xCB, 0x39, 0x2c, 0x00, 0x34, 0x02);                                                             \
    ILI9341_CMD(0xF7, 0x20);                                                                                     \
    ILI9341_CMD(0xEA, 0x00, 0x00);                                                                               \
    ILI9341_CMD(0xC0, 0x1B);                 /* Power control   //VRH[5:0] */                                    \
    ILI9341_CMD(0xC1, 0x12);                 /* Power control   //SAP[2:0];BT[3:0] */                            \
    ILI9341_CMD(0xC5, 0x32, 0x3C);           /* VCM control */                                                   \
    ILI9341_CMD(0xC7, 0x91);                 /* VCM control2 */                                                  \
    ILI9341_CMD(0x36, (0x40 | 0x80 | 0x08)); /* Memory Access Control */                                         \
    ILI9341_CMD(0xB1, 0x00, 0x10);           /* Frame Rate Control (1B=70, 1F=61, 10=119) */                     \
    ILI9341_CMD(0xB6, 0x0A, 0xA2);           /* Display Function Control */                                      \
    ILI9341_CMD(0xF6, 0x01, 0x30);                                                                               \
    ILI9341_CMD(0xF2, 0x00); /* 3Gamma Function Disable */                                                       \
    ILI9341_CMD(0x26, 0x01); /* Gamma curve selected */                                                          \
    ILI9341_CMD(0xE0, 0xD0, 0x00, 0x05, 0x0E, 0x15, 0x0D, 0x37, 0x43, 0x47, 0x09, 0x15, 0x12, 0x16, 0x19);	 \
    ILI9341_CMD(0xE1, 0xD0, 0x00, 0x05, 0x0D, 0x0C, 0x06, 0x2D, 0x44, 0x40, 0x0E, 0x1C, 0x18, 0x16, 0x19);

// Input
// Refer to rg_input.h to see all available RG_KEY_* and RG_GAMEPAD_*_MAP types

#define RG_GAMEPAD_GPIO_MAP {\
    {RG_KEY_UP,     GPIO_NUM_4,  GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_DOWN,   GPIO_NUM_32, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_LEFT,   GPIO_NUM_33, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_RIGHT,  GPIO_NUM_34, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_SELECT, GPIO_NUM_13, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_START,  GPIO_NUM_35, GPIO_FLOATING,    0},\
    {RG_KEY_A,      GPIO_NUM_14, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_B,      GPIO_NUM_27, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_MENU,   GPIO_NUM_25, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_OPTION, GPIO_NUM_21, GPIO_PULLUP_ONLY, 0},\
}

// Battery
#define RG_BATTERY_DRIVER           1
#define RG_BATTERY_ADC_UNIT         ADC_UNIT_1
#define RG_BATTERY_ADC_CHANNEL      ADC_CHANNEL_0
#define RG_BATTERY_CALC_PERCENT(raw) (((raw) * 2.f - 3500.f) / (4200.f - 3500.f) * 100.f)
#define RG_BATTERY_CALC_VOLTAGE(raw) ((raw) * 2.f * 0.001f)

// Status LED
#define RG_GPIO_LED                 GPIO_NUM_NC

// I2C BUS
// #define RG_GPIO_I2C_SDA             GPIO_NUM_15
// #define RG_GPIO_I2C_SCL             GPIO_NUM_4

// SPI Display
#define RG_GPIO_LCD_MISO            GPIO_NUM_NC
#define RG_GPIO_LCD_MOSI            GPIO_NUM_23
#define RG_GPIO_LCD_CLK             GPIO_NUM_18
#define RG_GPIO_LCD_CS              GPIO_NUM_5
#define RG_GPIO_LCD_DC              GPIO_NUM_2
#define RG_GPIO_LCD_BCKL            GPIO_NUM_12

// SPI SD Card
#define RG_GPIO_SDSPI_MISO          GPIO_NUM_19
#define RG_GPIO_SDSPI_MOSI          GPIO_NUM_23
#define RG_GPIO_SDSPI_CLK           GPIO_NUM_18
#define RG_GPIO_SDSPI_CS            GPIO_NUM_22

// External I2S DAC
#define RG_GPIO_SND_I2S_BCK         GPIO_NUM_NC
#define RG_GPIO_SND_I2S_WS          GPIO_NUM_NC
#define RG_GPIO_SND_I2S_DATA        GPIO_NUM_NC
// #define RG_GPIO_SND_AMP_ENABLE      GPIO_NUM_NC

```

ducalex, 2024-09-13:
The rg_storage commits in that period do not seem to change anything that could affect you, so I don't think it's related (could be wrong of course).

Looking at the other commits, few ideas:

- f512606e3ba5723c3a53897e42b798c3d2056345 : Maybe this broke something that slows down emulation (audio regulates the emulation speed). Try deleting your config file?
- 3f3faff8f4f816e43e565d56ee9ac7e31639c994 : After this change the led pin is no longer checked for GPIO_NUM_NC, so gpio_set_level causes an error which might cause some delay when toggling the (non-existent) disk LED? Try commenting RG_GPIO_LED instead of defining it to GPIO_NUM_NC?
- 9849f3e69ffdd67fc17bf9d01379d6f2039fe552 : Maybe this breaks something, causing the display to stall longer which would affect overall draw speed and sd card access?
- cb262d160c0b1af05fc46e392149907f56db4372 and d925dc32cda5672eb884554e6ac4ae02c55896b7 : Maybe this causes an overflow somewhere, making it sleep too long somewhere?

raphatex, 2024-09-16:
Well, commenting RG_GPIO_LED did the trick ! thanks a lot for your help !

ducalex, 2024-09-17:
Good! To avoid anyone else having the issue in the future I have done:

- Added back a GPIO_NUM_NC check because why not?
- Only the low battery indicator is enabled by default now. Disk indicator is very retro but realistically no other handheld has one which means it might not be desirable to most user anyway.

closed 2024-09-17 by ducalex