ducalex/retro-go #161: JSON File for Theme

32teeth, 2024-10-16:
## Describe the bug ##
Although there is a good description in THEMING.json
* Customers jumping from 1.3 to 1.4 with new `theme.json` file, no clue what to do.
* In 1.3 my theme selection and setting was put into the `retro-go.json` config

## Having themes in my theme folder produces wonky resuls

- GBZ
- .GBZ
- Pixel
- .Pixel

Selecting any root folder in my themes folder doesn't do anything

ducalex, 2024-10-23:
I agree with you, retro-go should have a default/example theme provided as a starting point, so I'll add one.

I'll look into the theme selector issue soon and test it thoroughly, I'll keep you updated.

32teeth, 2024-10-28:
That would be awesome. Even a sample template to get me going locally would be helpful

ducalex, 2024-10-29:
I'm not having your listing issue, I see my test theme correctly in the menu. I've only tested the dev branch though, it's possible the issue is in a release! There was a typo in one of the launcher's colors that is now fixed.

This is the structure I have:

````
$ tree /sdcard/retro-go/themes
/sdcard/retro-go/themes
└── test
    └── theme.json

1 directory, 1 file
````

I'm attaching a zip of the theme I've tested. It's garbage but it's enough to see that everything seems to be working as intended on my side. [test.zip](https://github.com/user-attachments/files/17559783/test.zip)

Soon I'll make a better sample theme to add to the repo and make sure the documentation matches reality!

If you still have the listing issue in the last release or dev branch, maybe you could share the exact commit you're running and also the contents of your retro-go folder (config and themes) so I can replicate?

32teeth, 2024-11-01:
Thank you so much. I'll take a look and report back

ducalex, 2024-12-13:
I've added a `themes` folder to the project. Currently containing a `default` folder that can be used as a template to create new themes. I've also update the instructions to clarify some things, hopefully it helps!

Also if you're willing to contribute/upstream your theme maybe I could start distributing retro-go with more than one built-in theme. And if you do that, we could add a special `#define` so that targets can specify the default theme to use, so that retro-esp32 builds would use your theme rather than mine, for example.