ducalex/retro-go #165: GAMES NOT LAUNCHING!!!!!

SSStarku, 2024-11-02:
##SCREEN ##
[image]
[image]
[image]
[image]

## PARTS ##
-Display:ILI9431
-Microcontroller: esp32s3
-SDcard: 64gb formated to FAT32
Config: default of the esp32s3devmodule + #define RG_GPIO_LCD_CS              GPIO_NUM_14

## To Reproduce ##
Steps to reproduce the behavior:
1. Installing the .img that i build in the esp32s3
2.formating sd card 64gb to FAT32 and installing some games
3. error message about the sd card in screen but pressing A
4. esp32s3 detecting the games and the sd card
5. Launching a game
6. seeing the last error message:
- The crash log:[crash.log](https://github.com/user-attachments/files/17607163/crash.log).
- The serial monitor:[serialmonitor.txt](https://github.com/user-attachments/files/17607164/serialmonitor.txt)

## unexpected behaviour##
-When I turn it on I get an error telling me thet the sd is not connected
-The games do not start and the machine restarts.
-I keep getting an error :[0m[31m[error] rg_surface_load_image: PNG decoding failed: 83

SSStarku, 2024-11-02:
[image]
ok now some games launch but in somes  i also have new error:
[crash.log](https://github.com/user-attachments/files/17607950/crash.log)
this one is about the proom

ducalex, 2024-11-03:
Retro-Go requires [PSRAM (external RAM) ](https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32s3/api-guides/external-ram.html) to work but it's not detected, that's what the error on your screenshot is telling you.

If your esp module doesn't have PSRAM then there's no way you will be able to run DOOM unfortunately. You might be able to run smaller ROMs in some of the emulators but most will likely not work, there's just not enough memory :(

SSStarku, 2024-11-03:
What is the minimum I need? Isn't an 8PSRAM enough?

ducalex, 2024-11-03:
8MB is more than enough! Then I guess it means it isn't detected during boot. The detection is up to esp-idf before retro-go starts so this is more difficult for me to help you.

Can you share your `retro-go/targets/<name>/sdkconfig` file?

SSStarku, 2024-11-03:
i have used the default Fri3D Camp 2024 sdkconfig because the hardware is the same

ducalex, 2024-11-03:
You can increase logging level, replace the "Log output" section with this:

````
CONFIG_LOG_DEFAULT_LEVEL_NONE=n
CONFIG_LOG_DEFAULT_LEVEL_ERROR=n
CONFIG_LOG_DEFAULT_LEVEL_WARN=n
CONFIG_LOG_DEFAULT_LEVEL_INFO=y
CONFIG_LOG_DEFAULT_LEVEL_DEBUG=n
CONFIG_LOG_DEFAULT_LEVEL_VERBOSE=n
CONFIG_LOG_DEFAULT_LEVEL=3
CONFIG_LOG_COLORS=y
````

You will need to run `python rg_tool.py clean` after any change to the sdkconfig.

Then run the new version and please share all serial output from the moment you power it on, there will hopefully be some messages about spiram/psram init failure...

pilesofspam, 2024-11-04:
Little late to the party, but I have a similar hardware setup (esp32S3 with 16MB flash, 8MB psram), and spent some time troubleshooting.  For me (on esp-idf-5.1.1), I went into each application directory (retro-core, prboom-go, etc) and ran "idf.py menuconfig".  From the first screen, scrolled on down to "Component Config" and again down to "esp psram".  Enable support for spiram there.  Also, for your device, it's probably octal, so make sure to mark that under spi ram config.

SSStarku, 2024-11-05:
Thanks for the answers. : )
 I'm trying to fix it but I have a problem, why does it appear that I don't have an SD connected but it does recognize it?

pilesofspam, 2024-11-05:
Just comparing notes, the first SD card I used was a Sandisk Industrial microSD 16GB and it wasn't recognized.  I didn't proceed to test any further.  I switched to a full size 32GB Sandisk Ultra (both of these were formatted to fat32, identical directory structure) and it works fine, no other changes and no more warnings.  For the card holder, I'm using hardware similar to yours, SD card holder mounted on the LCD board.

SSStarku, 2024-11-05:
cna you share your sdkconfig and config file?
Thank you so much

pilesofspam, 2024-11-05:
Sure, these are from ./components/targets/esp32s3-devkit-c:

config.h:
`// REF: https://wiki.odroid.com/odroid_go/odroid_go

// Target definition
#define RG_TARGET_NAME             "ESP32S3-DEVKIT-C"

// Storage
#define RG_STORAGE_ROOT             "/sd"
#define RG_STORAGE_SDSPI_HOST       SPI3_HOST
#define RG_STORAGE_SDSPI_SPEED      SDMMC_FREQ_DEFAULT
// #define RG_STORAGE_SDMMC_HOST       SDMMC_HOST_SLOT_1
// #define RG_STORAGE_SDMMC_SPEED      SDMMC_FREQ_DEFAULT
// #define RG_STORAGE_FLASH_PARTITION  "vfs"

// Audio
#define RG_AUDIO_USE_INT_DAC        0   // 0 = Disable, 1 = GPIO25, 2 = GPIO26, 3 = Both
#define RG_AUDIO_USE_EXT_DAC        1   // 0 = Disable, 1 = Enable

// Video
#define RG_SCREEN_DRIVER            0   // 0 = ILI9341
#define RG_SCREEN_HOST              SPI2_HOST
#define RG_SCREEN_SPEED             SPI_MASTER_FREQ_40M // SPI_MASTER_FREQ_80M
#define RG_SCREEN_BACKLIGHT         1
#define RG_SCREEN_WIDTH             320
#define RG_SCREEN_HEIGHT            240
#define RG_SCREEN_ROTATE            1
#define RG_SCREEN_MARGIN_TOP        0
#define RG_SCREEN_MARGIN_BOTTOM     0
#define RG_SCREEN_MARGIN_LEFT       0
#define RG_SCREEN_MARGIN_RIGHT      0
#define RG_SCREEN_INIT()                                                                                           \
    ILI9341_CMD(0xCF, 0x00, 0xc3, 0x30);                                                                         \
    ILI9341_CMD(0xED, 0x64, 0x03, 0x12, 0x81);                                                                   \
    ILI9341_CMD(0xE8, 0x85, 0x00, 0x78);                                                                         \
    ILI9341_CMD(0xCB, 0x39, 0x2c, 0x00, 0x34, 0x02);                                                             \
    ILI9341_CMD(0xF7, 0x20);                                                                                     \
    ILI9341_CMD(0xEA, 0x00, 0x00);                                                                               \
    ILI9341_CMD(0xC0, 0x1B);                 /* Power control   //VRH[5:0] */                                    \
    ILI9341_CMD(0xC1, 0x12);                 /* Power control   //SAP[2:0];BT[3:0] */                            \
    ILI9341_CMD(0xC5, 0x32, 0x3C);           /* VCM control */                                                   \
    ILI9341_CMD(0xC7, 0x91);                 /* VCM control2 */                                                  \
    ILI9341_CMD(0x36, (0x60|0x08));          /* Memory Access Control  (0x20|0x80|0x08) */                       \
    ILI9341_CMD(0xB1, 0x00, 0x10);           /* Frame Rate Control (1B=70, 1F=61, 10=119) */                     \
    ILI9341_CMD(0xB6, 0x0A, 0xA2);           /* Display Function Control */                                      \
    ILI9341_CMD(0xF6, 0x01, 0x30);                                                                               \
    ILI9341_CMD(0xF2, 0x00); /* 3Gamma Function Disable */                                                       \
    ILI9341_CMD(0x26, 0x01); /* Gamma curve selected */                                                          \
    ILI9341_CMD(0xE0, 0x0F, 0x31, 0x2B, 0x0C, 0x0E, 0x08, 0x4E, 0xF1, 0x37, 0x07, 0x10, 0x03, 0x0E, 0x09, 0x00); \
    ILI9341_CMD(0xE1, 0x00, 0x0E, 0x14, 0x03, 0x11, 0x07, 0x31, 0xC1, 0x48, 0x08, 0x0F, 0x0C, 0x31, 0x36, 0x0F);

// Input
// Refer to rg_input.h to see all available RG_KEY_* and RG_GAMEPAD_*_MAP types
#define RG_GAMEPAD_ADC_MAP {\
    {RG_KEY_UP,    ADC_UNIT_1, ADC_CHANNEL_5, ADC_ATTEN_DB_11, 3072, 4096},\
    {RG_KEY_RIGHT, ADC_UNIT_1, ADC_CHANNEL_6, ADC_ATTEN_DB_11, 1024, 3071},\
    {RG_KEY_DOWN,  ADC_UNIT_1, ADC_CHANNEL_5, ADC_ATTEN_DB_11, 1024, 3071},\
    {RG_KEY_LEFT,  ADC_UNIT_1, ADC_CHANNEL_6, ADC_ATTEN_DB_11, 3072, 4096},\
}
#define RG_GAMEPAD_GPIO_MAP {\
    {RG_KEY_SELECT, GPIO_NUM_16, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_START,  GPIO_NUM_17, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_MENU,   GPIO_NUM_18, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_OPTION, GPIO_NUM_8,  GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_A,      GPIO_NUM_15, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_B,      GPIO_NUM_5,  GPIO_PULLUP_ONLY, 0},\
}

// Battery
[... 178 lines cut ...]

#
# HTTP Server
#
CONFIG_HTTPD_MAX_REQ_HDR_LEN=1024
CONFIG_HTTPD_MAX_URI_LEN=1024

#
# TLS
#
# Not ideal but I don't want to deal with CAs right now :(
CONFIG_ESP_TLS_INSECURE=y
CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY=y

#
# SPI Flash driver
#
CONFIG_SPI_FLASH_ENABLE_ENCRYPTED_READ_WRITE=n

#
# Silence deprecations because I have no intention of fixing them right now
#
CONFIG_ADC_SUPPRESS_DEPRECATE_WARN=y
CONFIG_I2S_SUPPRESS_DEPRECATE_WARN=y
`

SSStarku, 2024-11-05:
thanks i will try it

ducalex, 2024-11-06:
> I went into each application directory (retro-core, prboom-go, etc) and ran "idf.py menuconfig".

You just have to do this once and then when it's working correctly just grab the sdkconfig in the current app folder and copy it to your target folder.  (In fact you pretty much have to copy it, because `rg_tool.py clean` will wipe the apps' sdkconfig)

It might be nice to document this process better in retro-go's documentation...

SSStarku, 2024-11-06:
Ok, I made my own sdkconfig because @pileofspam sdkconfig did not work, but this is loot and recognize the Spiram!!! and the error of the SD card has disappeared, but now I do not know why the SD card does not detect it. Can you help me? @ducalex i have this error rg_storage_init: Storage mounting failed! err=0x107
[sdkconfig.txt](https://github.com/user-attachments/files/17653726/sdkconfig.txt)

ducalex, 2024-11-07:
Error `0x107` is `ESP_ERR_TIMEOUT` ([see here for all error codes](https://github.com/espressif/esp-idf/blob/master/components/esp_common/include/esp_err.h)) so I guess your card doesn't respond?

Maybe it is a wiring issue? Did it work correctly with your old sdkconfig?

SSStarku, 2024-11-17:
Hello, thank you very much for your help. I was able to make it work!! But I have a problem. I want to connect a jack port, but in the config.h only gpio25 and 26 appear, but the esp32s3 does not have any of those.

SSStarku, 2024-11-17:
I have seen that the esp32s3 does not have a dac pin so what could I do, can someone help me?

closed 2024-11-17 by SSStarku