ducalex/retro-go #170: ESP32-S3 SPI ram not configured on fri3d-2024 on IDF-5.3

raphatex, 2024-11-27:
So I don't know if this issue needs to be fixed but I wanted to share my experience to help others!

On IDF-5.3 it seems that sdkconfig is a bit different for the S3, so the Octal SPI RAM is not configured as it should be. Also, the default frequency is 160 MHz for the ESP32 instead of 240 MHz. There are probably other differences but I didn't dig deeper.

SSStarku, 2024-11-28:
i had the same error in my esp32s3 8psram that the PSRAM was not recognized in ESPTool 5.3 and I created a new one and this one works fine. Check if the Ducalex one works and if not, try this one.

[sdkconfig.txt](https://github.com/user-attachments/files/17951946/sdkconfig.txt)

raphatex, 2024-11-28:
Oh yeah sorry i just realised you already mentioned this issue

raphatex, 2024-11-28:
> I've updated the sdkconfig for CPU frequency and it seems to now be working correctly on at least 4.4 5.0 5.3, can you confirm? If it doesn't work, I've left some notes in the commit message and you can experiment and we can use whichever names you find work best.
> 
> For SPIRAM it might be more challenging (but not impossible) because 5.3 seems to have entirely dropped many of the old names.
> 
> If we can find values that work consistently across 5.x at least that would be good.

Frequency in working fine for me too !

raphatex, 2024-11-28:
So I edited the SPI RAM config section in the SDKconfig file with this:
```
#
# SPI RAM config
#
CONFIG_SPIRAM=y
# CONFIG_SPIRAM_MODE_QUAD is not set
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_TYPE_AUTO=y
# CONFIG_SPIRAM_TYPE_ESPPSRAM64 is not set
CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY=y
CONFIG_SPIRAM_CLK_IO=30
CONFIG_SPIRAM_CS_IO=26
# CONFIG_SPIRAM_XIP_FROM_PSRAM is not set
# CONFIG_SPIRAM_FETCH_INSTRUCTIONS is not set
# CONFIG_SPIRAM_RODATA is not set
# CONFIG_SPIRAM_SPEED_40M is not set
CONFIG_SPIRAM_SPEED_80M=y
CONFIG_SPIRAM_SPEED=80
# CONFIG_SPIRAM_ECC_ENABLE is not set
CONFIG_SPIRAM_BOOT_INIT=y
# CONFIG_SPIRAM_IGNORE_NOTFOUND is not set
# CONFIG_SPIRAM_USE_MEMMAP is not set
# CONFIG_SPIRAM_USE_CAPS_ALLOC is not set
CONFIG_SPIRAM_USE_MALLOC=y
CONFIG_SPIRAM_MEMTEST=y
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=16384
# CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP is not set
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=32768
# CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY is not set

CONFIG_ESP32_SPIRAM_SUPPORT=y
CONFIG_SPIRAM_BOOT_INIT=y
CONFIG_SPIRAM_IGNORE_NOTFOUND=y
CONFIG_SPIRAM_USE_MEMMAP=n
CONFIG_SPIRAM_USE_CAPS_ALLOC=n
CONFIG_SPIRAM_USE_MALLOC=y
CONFIG_SPIRAM_TYPE_AUTO=y
CONFIG_SPIRAM_TYPE_ESPPSRAM32=n
CONFIG_SPIRAM_TYPE_ESPPSRAM64=n
CONFIG_SPIRAM_SIZE=-1
CONFIG_SPIRAM_SPEED_40M=n
CONFIG_SPIRAM_SPEED_80M=y
CONFIG_SPIRAM_MEMTEST=y
CONFIG_SPIRAM_CACHE_WORKAROUND=n
CONFIG_SPIRAM_BANKSWITCH_ENABLE=n
#CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=16384
#CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=16384
```

(I simply added the IDF-5.3 specific on top of the old configuration) 
It works fine on IDF-5.3.
But I do not have older version of IDF to test it

ducalex, 2024-11-29:
Thanks to you two, based on your two versions I have identified what seems to be the main culprit and I've pushed a change. There were more differences but the values picked by the target's author likely had a reason to be so I prefer to leave them alone for now.

The changes seem to maintain SPIRAM being enabled on 4.4.8 5.0 5.3 (at least in the menu, I can't test on real hardware myself).

raphatex, 2024-12-03:
It seems to work fine on the S3 here ! (IDF-5.3)

ducalex, 2024-12-17:
The issue seems to be resolved so I will close it. Feel free to reopen, comment, or make a new issue if there are any outstanding problems!

closed 2024-12-17 by ducalex