ducalex/retro-go #175: Gameboy: Donkey Kong (World) shows incorrect rendering on pause and other screens

yaustar, 2025-01-04:
## Describe the bug ##
On [Donkey Kong (World)](https://www.mariowiki.com/Donkey_Kong_(Game_Boy)) on Gameboy, when pausing the game it renders the screen incorrectly 

[image]

It should look like 

[image]

There are some other screens that have this issue such as the results when completing a level.

## To Reproduce ##
Steps to reproduce the behavior:
1. Play the Donkey Kong game on Gameboy
2. Start a game and enter a level
3. Press start to pause

## Expected behavior ##
The pause screen to be rendered correctly

## Additional context ##
I used the same ROM file on another emulator and it shows the screen correctly so it's not an issue with the ROM

ducalex, 2025-01-10:
This is caused by a hack that I've added to fix the score displayed after clearing a stage.

https://github.com/ducalex/retro-go/blob/de5e8ae45a483b83c1733df704aec3a3c5aa65ac/retro-core/components/gnuboy/lcd.c#L657-L662

We obviously have some emulation accuracy issue but unfortunately I've forgotten everything I knew about GB emulation myself so I'd need an expert to point me in the right direction...

ducalex, 2025-01-10:
I mean I can improve the hack so that it kicks in only on necessary screens, but it would be nice if someone could tell me what's the actual problem :(.