ducalex/retro-go #181: Issue with NES Emulator (not retro-go related)

simplyrohan, 2025-01-24:
This isn't an issue with retro-go, I'm just hear asking for some advice with one of the emulators I am using separately. I've downloaded the `nofrendo` component from `retro-core` and want to use it separately from the retro-go project, specifically with the Arduino framework. I've had a lot of success doing this with the `gnuboy` and `smsplus` emulators.

I am running into a problem my project where the NES emulators graphics seem to be glitched. From some observations, I am pretty sure that the wrong sprites are being drawn. The game logic itself seems to be working great, I can control and play the game as it should be (eg. in Super Mario Bros I am colliding with things correctly, its just the graphics show a different, glitched, screen which doesn't mirror the actual game).

I wanted to know if you've experienced anything like this when making the NES emulator, and what your solution was.

I understand this isn't your project, and I'm not expecting you to help me write this. I'm just seeking some advice if you've had this problem.

(I've attached images of what the glitched screen looks like)

[image]

[image]

simplyrohan, 2025-01-24:
Also, after checking the console, it seems that nofrendo is giving some warnings about the ROM
```
ROM: Game found in database.
 ! ROM: mirroring mismatch! (DB: 0, ROM: 3)
 ! ROM: prg_rom_banks mismatch! (DB: 0, ROM: 4)
 ! ROM: prg_ram_banks mismatch! (DB: 0, ROM: 1)
 ! ROM: chr_rom_banks mismatch! (DB: 0, ROM: 1)
 ! ROM: chr_ram_banks mismatch! (DB: 0, ROM: 1)
```

ducalex, 2025-01-26:
I think the problem is that you didn't implement a CRC function and then nofrendo (erroneously) applies corrections based on the final entry of the database, which is invalid.

I've pushed a change to the dev branch that should fix it but I suggest that you also fix the `CRC32()` define in `nes/utils.h` to point to `crc32_le` from esp-idf or your own CRC32 implementation because a few games do need the database corrections.

> I understand this isn't your project, and I'm not expecting you to help me write this. I'm just seeking some advice if you've had this problem.

Most of retro-go's emulators were heavily modified by me so any problem with them is fine to report here, even if you don't use them inside retro-go :)

simplyrohan, 2025-01-26:
Pointing `CRC32` to the `crc32_le` worked perfectly! And thank you for porting all these emulators to work on the ESP32

closed 2025-01-26 by simplyrohan