ducalex/retro-go #182: Porting for T-Deck Pluis need help

kazz2020, 2025-01-27:
Hi i want to port this to LilyGo T-Deck Plus. 
`// Target definition
#define RG_TARGET_NAME             "ESPLAY-S3"

// Storage
#define RG_STORAGE_DRIVER           1                   // 0 = Host, 1 = SDSPI, 2 = SDMMC, 3 = USB, 4 = Flash
#define RG_STORAGE_HOST             SDMMC_HOST_SLOT_1   // Used by driver 1 and 2
#define RG_STORAGE_SPEED            SDMMC_FREQ_DEFAULT  // Used by driver 1 and 2
#define RG_STORAGE_ROOT             "/sd"               // Storage mount point

// Audio
#define RG_AUDIO_USE_INT_DAC        0   // 0 = Disable, 1 = GPIO25, 2 = GPIO26, 3 = Both
#define RG_AUDIO_USE_EXT_DAC        1   // 0 = Disable, 1 = Enable

// Video
#define RG_SCREEN_DRIVER            0   // 0 = ILI9341
#define RG_SCREEN_HOST              SPI2_HOST
#define RG_SCREEN_SPEED             SPI_MASTER_FREQ_80M
#define RG_SCREEN_TYPE              4   // 4 = ESPLAY-ST7789V2
#define RG_SCREEN_WIDTH             320
#define RG_SCREEN_HEIGHT            240
#define RG_SCREEN_ROTATE            0
#define RG_SCREEN_MARGIN_TOP        0
#define RG_SCREEN_MARGIN_BOTTOM     0
#define RG_SCREEN_MARGIN_LEFT       0
#define RG_SCREEN_MARGIN_RIGHT      0

// Input
#define RG_GAMEPAD_DRIVER           3   // 1 = ODROID-GO, 2 = Serial, 3 = I2C
#define RG_GAMEPAD_HAS_MENU_BTN     1
#define RG_GAMEPAD_HAS_OPTION_BTN   1
// Note: Depending on the driver, the button map can represent bits, registers, keys, or gpios.
#define RG_GAMEPAD_MAP_MENU         (0)
#define RG_GAMEPAD_MAP_OPTION       (0)
#define RG_GAMEPAD_MAP_START        (1<<0)
#define RG_GAMEPAD_MAP_SELECT       (1<<1)
#define RG_GAMEPAD_MAP_UP           (1<<2)
#define RG_GAMEPAD_MAP_RIGHT        (1<<5)
#define RG_GAMEPAD_MAP_DOWN         (1<<3)
#define RG_GAMEPAD_MAP_LEFT         (1<<4)
#define RG_GAMEPAD_MAP_A            (1<<6)
#define RG_GAMEPAD_MAP_B            (1<<7)
#define RG_GAMEPAD_MAP_X            (0)
#define RG_GAMEPAD_MAP_Y            (0)

// Battery
#define RG_BATTERY_ADC_CHANNEL      ADC1_CHANNEL_3
#define RG_BATTERY_CALC_PERCENT(raw) (((raw) * 2.f - 3500.f) / (4200.f - 3500.f) * 100.f)
#define RG_BATTERY_CALC_VOLTAGE(raw) ((raw) * 2.f * 0.001f)

// Status LED
//#define RG_GPIO_LED                 GPIO_NUM_2

// I2C BUS
#define RG_GPIO_I2C_SDA             GPIO_NUM_18
#define RG_GPIO_I2C_SCL             GPIO_NUM_8

// Built-in gamepad
#define RG_GPIO_GAMEPAD_L           GPIO_NUM_2
#define RG_GPIO_GAMEPAD_R           GPIO_NUM_1
#define RG_GPIO_GAMEPAD_MENU        GPIO_NUM_3
#define RG_GPIO_GAMEPAD_OPTION      GPIO_NUM_15

// SNES-style gamepad
// #define RG_GPIO_GAMEPAD_LATCH       GPIO_NUM_NC
// #define RG_GPIO_GAMEPAD_CLOCK       GPIO_NUM_NC
// #define RG_GPIO_GAMEPAD_DATA        GPIO_NUM_NC

// SPI Display
#define RG_GPIO_LCD_MISO            GPIO_NUM_38
#define RG_GPIO_LCD_MOSI            GPIO_NUM_41
#define RG_GPIO_LCD_CLK             GPIO_NUM_40
#define RG_GPIO_LCD_CS              GPIO_NUM_12
#define RG_GPIO_LCD_DC              GPIO_NUM_11
#define RG_GPIO_LCD_BCKL            GPIO_NUM_42
//#define RG_GPIO_LCD_RST             GPIO_NUM_NC

// SPI SD Card
#define RG_GPIO_SDSPI_MISO          GPIO_NUM_38
#define RG_GPIO_SDSPI_MOSI          GPIO_NUM_41
#define RG_GPIO_SDSPI_CLK           GPIO_NUM_40
#define RG_GPIO_SDSPI_CS            GPIO_NUM_39

// External I2S DAC
#define RG_GPIO_SND_I2S_BCK         7
#define RG_GPIO_SND_I2S_WS          5
#define RG_GPIO_SND_I2S_DATA        6
//#define RG_GPIO_SND_AMP_ENABLE      18`
This is my target but i dont know how to axctivate keyboard. Can someone guide me for this? I dont really know if that works to.

ducalex, 2025-02-03:
Retro-go has no driver for the keyboard, so you'll have to write your own unfortunately!

kazz2020, 2025-02-03:
> Retro-go has no driver for the keyboard, so you'll have to write your own unfortunately!

OK but the screen doesn't work either. Maybe you can add support only for screen?

ducalex, 2025-02-05:
It's difficult to add support for a device that I do not have but I can try to help you. Can you show us the datasheet and schematics for your exact device? I'd need to know the pinout and lcd controller type and screen model and all that!

kazz2020, 2025-02-05:
everything is here https://lilygo.cc/products/t-deck-plus-1

rapha-tech, 2025-02-05:
for your display here is a piece of code that should make it work !
```
#define RG_SCREEN_DRIVER            0   // 0 = ILI9341
#define RG_SCREEN_HOST              SPI2_HOST
#define RG_SCREEN_SPEED             SPI_MASTER_FREQ_40M
#define RG_SCREEN_BACKLIGHT         1
#define RG_SCREEN_WIDTH             320
#define RG_SCREEN_HEIGHT            240
#define RG_SCREEN_ROTATE            0
#define RG_SCREEN_MARGIN_TOP        0
#define RG_SCREEN_MARGIN_BOTTOM     0
#define RG_SCREEN_MARGIN_LEFT       0
#define RG_SCREEN_MARGIN_RIGHT      0
#define RG_SCREEN_INIT()                                                                                         \
    ILI9341_CMD(0xCF, 0x00, 0xc3, 0x30);                                                                         \
    ILI9341_CMD(0xED, 0x64, 0x03, 0x12, 0x81);                                                                   \
    ILI9341_CMD(0xE8, 0x85, 0x00, 0x78);                                                                         \
    ILI9341_CMD(0xCB, 0x39, 0x2c, 0x00, 0x34, 0x02);                                                             \
    ILI9341_CMD(0xF7, 0x20);                                                                                     \
    ILI9341_CMD(0xEA, 0x00, 0x00);                                                                               \
    ILI9341_CMD(0xC0, 0x1B);                 /* Power control   //VRH[5:0] */                                    \
    ILI9341_CMD(0xC1, 0x12);                 /* Power control   //SAP[2:0];BT[3:0] */                            \
    ILI9341_CMD(0xC5, 0x32, 0x3C);           /* VCM control */                                                   \
    ILI9341_CMD(0xC7, 0x91);                 /* VCM control2 */                                                  \
    ILI9341_CMD(0x36, (0x40 | 0x80 | 0x08)); /* Memory Access Control */                                         \
    ILI9341_CMD(0xB1, 0x00, 0x10);           /* Frame Rate Control (1B=70, 1F=61, 10=119) */                     \
    ILI9341_CMD(0xB6, 0x0A, 0xA2);           /* Display Function Control */                                      \
    ILI9341_CMD(0xF6, 0x01, 0x30);                                                                               \
    ILI9341_CMD(0xF2, 0x00); /* 3Gamma Function Disable */                                                       \
    ILI9341_CMD(0x26, 0x01); /* Gamma curve selected */                                                          \
    ILI9341_CMD(0xE0, 0xD0, 0x00, 0x02, 0x07, 0x0a, 0x28, 0x32, 0x44, 0x42, 0x06, 0x0e, 0x12, 0x14, 0x17);       \
    ILI9341_CMD(0xE1, 0xD0, 0x00, 0x02, 0x07, 0x0a, 0x28, 0x31, 0x54, 0x47, 0x0E, 0x1C, 0x17, 0x1b, 0x1e);       
```

kazz2020, 2025-02-05:
I have this [image]

kazz2020, 2025-02-05:
[image]

rapha-tech, 2025-02-05:
Yes this means a key is being pressed down

put this line in your target.h
`#define RG_RECOVERY_BTN RG_KEY_MENU // Keep this button pressed to open the recovery menu`
you can change the button to a button is known not to be pressed so you could actually boot into launcher

kazz2020, 2025-02-05:
#define RG_GAMEPAD_GPIO_MAP {\
    {RG_KEY_SELECT, GPIO_NUM_0, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_START,  GPIO_NUM_1, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_MENU,   GPIO_NUM_2, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_OPTION, GPIO_NUM_3,  GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_A,      GPIO_NUM_15, GPIO_PULLUP_ONLY, 0},\ 
that are buttons for dpad on t-deck. is it correct place to put pin numbers?

rapha-tech, 2025-02-05:
Yes seems nice ! just make sure you got '\' at the end of each line

kazz2020, 2025-02-05:
thx a lot. any clue how to make keyboard to work?

kazz2020, 2025-02-05:
[image]

[image]

It's like the down button is pressing itself.

rapha-tech, 2025-02-05:
Well this issue seems a bit trickier I would recommend you take a look at the documentation on your device's keyboard + check all the availables modes in rg_input.c and see the one that might be compatible/easier to adapt

kazz2020, 2025-02-05:
ill take a look tomorrow. thanks again for help

rapha-tech, 2025-02-05:
No problem !

kazz2020, 2025-02-06:
`// Target definition
#define RG_TARGET_NAME             "TDECK"

// Storage
#define RG_STORAGE_ROOT             "/sd"
#define RG_STORAGE_SDSPI_HOST       SPI3_HOST
#define RG_STORAGE_SDSPI_SPEED      SDMMC_FREQ_DEFAULT
// #define RG_STORAGE_SDMMC_HOST       SDMMC_HOST_SLOT_1
// #define RG_STORAGE_SDMMC_SPEED      SDMMC_FREQ_DEFAULT
// #define RG_STORAGE_FLASH_PARTITION  "vfs"

// Audio
#define RG_AUDIO_USE_INT_DAC        0   // 0 = Disable, 1 = GPIO25, 2 = GPIO26, 3 = Both
#define RG_AUDIO_USE_EXT_DAC        1   // 0 = Disable, 1 = Enable

#define RG_SCREEN_DRIVER            0   // 0 = ILI9341
#define RG_SCREEN_HOST              SPI2_HOST
#define RG_SCREEN_SPEED             SPI_MASTER_FREQ_40M
#define RG_SCREEN_BACKLIGHT         1
#define RG_SCREEN_WIDTH             320
#define RG_SCREEN_HEIGHT            240
#define RG_SCREEN_ROTATE            0
#define RG_SCREEN_MARGIN_TOP        0
#define RG_SCREEN_MARGIN_BOTTOM     0
#define RG_SCREEN_MARGIN_LEFT       0
#define RG_SCREEN_MARGIN_RIGHT      0
#define RG_SCREEN_INIT()                                                                                         \
    ILI9341_CMD(0xCF, 0x00, 0xc3, 0x30);                                                                         \
    ILI9341_CMD(0xED, 0x64, 0x03, 0x12, 0x81);                                                                   \
    ILI9341_CMD(0xE8, 0x85, 0x00, 0x78);                                                                         \
    ILI9341_CMD(0xCB, 0x39, 0x2c, 0x00, 0x34, 0x02);                                                             \
    ILI9341_CMD(0xF7, 0x20);                                                                                     \
    ILI9341_CMD(0xEA, 0x00, 0x00);                                                                               \
    ILI9341_CMD(0xC0, 0x1B);                 /* Power control   //VRH[5:0] */                                    \
    ILI9341_CMD(0xC1, 0x12);                 /* Power control   //SAP[2:0];BT[3:0] */                            \
    ILI9341_CMD(0xC5, 0x32, 0x3C);           /* VCM control */                                                   \
    ILI9341_CMD(0xC7, 0x91);                 /* VCM control2 */                                                  \
    ILI9341_CMD(0x36, (0x40 | 0x80 | 0x08)); /* Memory Access Control */                                         \
    ILI9341_CMD(0xB1, 0x00, 0x10);           /* Frame Rate Control (1B=70, 1F=61, 10=119) */                     \
    ILI9341_CMD(0xB6, 0x0A, 0xA2);           /* Display Function Control */                                      \
    ILI9341_CMD(0xF6, 0x01, 0x30);                                                                               \
    ILI9341_CMD(0xF2, 0x00); /* 3Gamma Function Disable */                                                       \
    ILI9341_CMD(0x26, 0x01); /* Gamma curve selected */                                                          \
    ILI9341_CMD(0xE0, 0xD0, 0x00, 0x02, 0x07, 0x0a, 0x28, 0x32, 0x44, 0x42, 0x06, 0x0e, 0x12, 0x14, 0x17);       \
    ILI9341_CMD(0xE1, 0xD0, 0x00, 0x02, 0x07, 0x0a, 0x28, 0x31, 0x54, 0x47, 0x0E, 0x1C, 0x17, 0x1b, 0x1e);

#define RG_RECOVERY_BTN RG_KEY_MENU // Keep this button pressed to open the recovery menu
// Input
// Refer to rg_input.h to see all available RG_KEY_* and RG_GAMEPAD_*_MAP types
#define RG_GAMEPAD_ADC_MAP {\
    {RG_KEY_UP,    ADC_UNIT_1, ADC_CHANNEL_5, ADC_ATTEN_DB_11, 3072, 4096},\
    {RG_KEY_RIGHT, ADC_UNIT_1, ADC_CHANNEL_6, ADC_ATTEN_DB_11, 1024, 3071},\
    {RG_KEY_DOWN,  ADC_UNIT_1, ADC_CHANNEL_5, ADC_ATTEN_DB_11, 1024, 3071},\
    {RG_KEY_LEFT,  ADC_UNIT_1, ADC_CHANNEL_6, ADC_ATTEN_DB_11, 3072, 4096},\
}
#define RG_GAMEPAD_GPIO_MAP {\
    {RG_KEY_SELECT, GPIO_NUM_0, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_START,  GPIO_NUM_1, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_MENU,   GPIO_NUM_2, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_OPTION, GPIO_NUM_3,  GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_A,      GPIO_NUM_15, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_B,      GPIO_NUM_10,  GPIO_PULLUP_ONLY, 0},\
}
#define RG_RECOVERY_BTN RG_KEY_MENU // Keep this button pressed to open the recovery menu
// Battery
#define RG_BATTERY_DRIVER           1
#define RG_BATTERY_ADC_UNIT         ADC_UNIT_1
#define RG_BATTERY_ADC_CHANNEL      ADC_CHANNEL_3
#define RG_BATTERY_CALC_PERCENT(raw) (((raw) * 2.f - 3500.f) / (4200.f - 3500.f) * 100.f)
#define RG_BATTERY_CALC_VOLTAGE(raw) ((raw) * 2.f * 0.001f)

// Status LED
//#define RG_GPIO_LED                 GPIO_NUM_38

// SPI Display (back up working)
#define RG_GPIO_LCD_MISO            GPIO_NUM_38
#define RG_GPIO_LCD_MOSI            GPIO_NUM_41
#define RG_GPIO_LCD_CLK             GPIO_NUM_40
#define RG_GPIO_LCD_CS              GPIO_NUM_12
#define RG_GPIO_LCD_DC              GPIO_NUM_11
#define RG_GPIO_LCD_BCKL            GPIO_NUM_42
//#define RG_GPIO_LCD_RST             GPIO_NUM_3

#define RG_GPIO_SDSPI_MISO          GPIO_NUM_38
#define RG_GPIO_SDSPI_MOSI          GPIO_NUM_41
#define RG_GPIO_SDSPI_CLK           GPIO_NUM_40
#define RG_GPIO_SDSPI_CS            GPIO_NUM_39

// External I2S DAC
#define RG_GPIO_SND_I2S_BCK         7
#define RG_GPIO_SND_I2S_WS          5
#define RG_GPIO_SND_I2S_DATA        6
// #define RG_GPIO_SND_AMP_ENABLE      18

// T-Deck Keyboard Support
#define RG_TARGET_TDECK
#define RG_GAMEPAD_TDECK_MAP {\
    {RG_KEY_UP,     'w'},\
    {RG_KEY_DOWN,   's'},\
    {RG_KEY_LEFT,   'a'},\
    {RG_KEY_RIGHT,  'd'},\
    {RG_KEY_A,      'j'},\
    {RG_KEY_B,      'k'},\
    {RG_KEY_X,      'u'},\
    {RG_KEY_Y,      'i'},\
    {RG_KEY_START,  '\r'},\
    {RG_KEY_SELECT, '\e'},\
    {RG_KEY_MENU,   'm'},\
    {RG_KEY_OPTION, 'o'},\
}
`

kazz2020, 2025-02-06:
[image]

SD card is OK and fat32 so I dint understand. Pins are OK so what could be wrong?

rapha-tech, 2025-02-06:
change `#define RG_STORAGE_SDSPI_HOST SPI3_HOST`
with `#define RG_STORAGE_SDSPI_HOST SPI2_HOST`

> [image]
> 
> SD card is OK and fat32 so I dint understand. Pins are OK so what could be wrong?

rapha-tech, 2025-02-06:
Also, for you keyboard as @ducalex said you'll have to write your own driver for it, take a look at the datasheet of your keyboard driver and compare with the availables drivers in rg_input.c to create your own

kazz2020, 2025-02-06:
> change `#define RG_STORAGE_SDSPI_HOST SPI3_HOST` with `#define RG_STORAGE_SDSPI_HOST SPI2_HOST` Since your are sharing the same SPI bus (pin MOSI, MISO and CLK)
> 
> > [image]
> > SD card is OK and fat32 so I dint understand. Pins are OK so what could be wrong?

did not help with sd. any idea why when i define trackball it acts like crazy?

kazz2020, 2025-02-17:
I give up. Nothing works.

kazz2020, 2025-02-22:
> Also, for you keyboard as [@ducalex](https://github.com/ducalex) said you'll have to write your own driver for it, take a look at the datasheet of your keyboard driver and compare with the availables drivers in rg_input.c to create your own

`// Target definition
#define RG_TARGET_NAME             "RETRO-ESP32"

// Storage
#define RG_STORAGE_ROOT             "/sd"
#define RG_STORAGE_SDSPI_HOST       SPI2_HOST
#define RG_STORAGE_SDSPI_SPEED      SDMMC_FREQ_DEFAULT
// #define RG_STORAGE_SDMMC_HOST       SDMMC_HOST_SLOT_1
// #define RG_STORAGE_SDMMC_SPEED      SDMMC_FREQ_DEFAULT
// #define RG_STORAGE_FLASH_PARTITION  "vfs"

// Audio
#define RG_AUDIO_USE_INT_DAC        0   // 0 = Disable, 1 = GPIO25, 2 = GPIO26, 3 = Both
#define RG_AUDIO_USE_EXT_DAC        0   // 0 = Disable, 1 = Enable

// Video
#define RG_SCREEN_DRIVER            0   // 0 = ILI9341
#define RG_SCREEN_HOST              SPI2_HOST
#define RG_SCREEN_SPEED             SPI_MASTER_FREQ_40M
#define RG_SCREEN_BACKLIGHT         0   // Set to 0 since LED is connected to VCC
#define RG_SCREEN_WIDTH             320
#define RG_SCREEN_HEIGHT            240
#define RG_SCREEN_ROTATE            0
#define RG_SCREEN_MARGIN_TOP        0
#define RG_SCREEN_MARGIN_BOTTOM     0
#define RG_SCREEN_MARGIN_LEFT       0
#define RG_SCREEN_MARGIN_RIGHT      0
#define RG_SCREEN_INIT()                                                                                         \
    ILI9341_CMD(0x01);                       /* Software reset */                                                \
    vTaskDelay(pdMS_TO_TICKS(100));          /* Delay 100ms */                                                  \
    ILI9341_CMD(0xCF, 0x00, 0xC3, 0x30);                                                                         \
    ILI9341_CMD(0xED, 0x64, 0x03, 0x12, 0x81);                                                                   \
    ILI9341_CMD(0xE8, 0x85, 0x00, 0x78);                                                                         \
    ILI9341_CMD(0xCB, 0x39, 0x2C, 0x00, 0x34, 0x02);                                                             \
    ILI9341_CMD(0xF7, 0x20);                                                                                     \
    ILI9341_CMD(0xEA, 0x00, 0x00);                                                                               \
    ILI9341_CMD(0xC0, 0x21);                 /* Power control VRH[5:0] */                                        \
    ILI9341_CMD(0xC1, 0x11);                 /* Power control SAP[2:0];BT[3:0] */                                \
    ILI9341_CMD(0xC5, 0x3F, 0x3C);           /* VCM control */                                                   \
    ILI9341_CMD(0xC7, 0xB5);                 /* VCM control2 */                                                  \
    ILI9341_CMD(0x36, 0x48);                 /* Memory Access Control */                                         \
    ILI9341_CMD(0xB1, 0x00, 0x1B);           /* Frame Rate Control */                                            \
    ILI9341_CMD(0xB6, 0x0A, 0xA2);           /* Display Function Control */                                      \
    ILI9341_CMD(0x3A, 0x55);                 /* COLMOD: Pixel Format Set */                                      \
    ILI9341_CMD(0xF2, 0x00);                 /* 3Gamma Function Disable */                                       \
    ILI9341_CMD(0x26, 0x01);                 /* Gamma curve selected */                                          \
    ILI9341_CMD(0xE0, 0x0F, 0x26, 0x24, 0x0B, 0x0E, 0x09, 0x54, 0xA8, 0x46, 0x0C, 0x17, 0x09, 0x0F, 0x07, 0x00); \
    ILI9341_CMD(0xE1, 0x00, 0x19, 0x1B, 0x04, 0x10, 0x07, 0x2A, 0x47, 0x39, 0x03, 0x06, 0x06, 0x30, 0x38, 0x0F); \
    ILI9341_CMD(0x11);                       /* Sleep out */                                                     \
    vTaskDelay(pdMS_TO_TICKS(120));          /* Delay 120ms */                                                  \
    ILI9341_CMD(0x29);                       /* Display on */                                                    \
    vTaskDelay(pdMS_TO_TICKS(20));           /* Delay 20ms */

// Input
// Refer to rg_input.h to see all available RG_KEY_* and RG_GAMEPAD_*_MAP types
#define RG_GAMEPAD_ADC_MAP {\
    {RG_KEY_UP,    ADC_UNIT_1, ADC_CHANNEL_7, ADC_ATTEN_DB_11, 2048, 4096},\
    {RG_KEY_DOWN,  ADC_UNIT_1, ADC_CHANNEL_7, ADC_ATTEN_DB_11, 1024, 2047},\
    {RG_KEY_LEFT,  ADC_UNIT_1, ADC_CHANNEL_6, ADC_ATTEN_DB_11, 2048, 4096},\
    {RG_KEY_RIGHT, ADC_UNIT_1, ADC_CHANNEL_6, ADC_ATTEN_DB_11, 1024, 2047},\
}
#define RG_GAMEPAD_GPIO_MAP {\
    {RG_KEY_UP,     GPIO_NUM_34,  GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_DOWN,   GPIO_NUM_33, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_LEFT,   GPIO_NUM_39, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_RIGHT,  GPIO_NUM_32, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_SELECT, GPIO_NUM_17, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_START,  GPIO_NUM_21, GPIO_PULLUP_ONLY, 0},\
[... 26 lines cut ...]
#define RG_GPIO_LCD_MOSI            GPIO_NUM_23  // Your MOSI pin
#define RG_GPIO_LCD_CLK             GPIO_NUM_18  // Your SCK pin
#define RG_GPIO_LCD_CS              GPIO_NUM_5   // Your CS pin
#define RG_GPIO_LCD_DC              GPIO_NUM_2   // Your DC pin
//#define RG_GPIO_LCD_BCKL            GPIO_NUM_14  // Commented out since LED is on VCC
#define RG_GPIO_LCD_RST             GPIO_NUM_0   // Reset pin (adjust if different)

// SPI SD Card
#define RG_GPIO_SDSPI_MISO          GPIO_NUM_12
#define RG_GPIO_SDSPI_MOSI          GPIO_NUM_13
[... cut ...]
// #define RG_UPDATER_ENABLE               1
// #define RG_UPDATER_APPLICATION          RG_APP_FACTORY
// #define RG_UPDATER_DOWNLOAD_LOCATION    RG_STORAGE_ROOT "/odroid/firmware"
`
can You check this config? the pins are good my LED pin in ILI9431 320x240 touch version is connected to VCC i changed to 4MB disabled psram and i use  python rg_tool.py --target retro-esp32 build-img launcher retro-core when i flash the image i see white screen with little flickering. No serial monitor no nothing.

kazz2020, 2025-02-23:
i try now to connect ESP32S3 Devkitc to my ILI9341 2.8 display and i cant :/ https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-devkitc-1/user_guide.html#hardware-reference 
`// Target definition
#define RG_TARGET_NAME             "ESPLAY-S3"

// Storage
#define RG_STORAGE_ROOT             "/sd"
#define RG_STORAGE_SDSPI_HOST       SPI2_HOST
#define RG_STORAGE_SDSPI_SPEED      SDMMC_FREQ_DEFAULT
#define RG_STORAGE_ALLOW_FAILURE    1
//#define RG_STORAGE_SDMMC_HOST       SDMMC_HOST_SLOT_1
//#define RG_STORAGE_SDMMC_SPEED      SDMMC_FREQ_DEFAULT
// #define RG_STORAGE_FLASH_PARTITION  "vfs"

// Audio
#define RG_AUDIO_USE_INT_DAC        0   // 0 = Disable, 1 = GPIO25, 2 = GPIO26, 3 = Both
#define RG_AUDIO_USE_EXT_DAC        0   // 0 = Disable, 1 = Enable

// Video
#define RG_SCREEN_DRIVER            0   // 0 = ILI9341
#define RG_SCREEN_HOST             SPI2_HOST    // Keep SPI2 for LCD
#define RG_SCREEN_SPEED            SPI_MASTER_FREQ_20M  // Reduced speed to 20MHz
#define RG_SCREEN_BACKLIGHT         1   // Enable backlight control
#define RG_SCREEN_WIDTH            320
#define RG_SCREEN_HEIGHT           240
#define RG_SCREEN_ROTATE            1    // Keep at 1 for correct orientation
#define RG_SCREEN_MARGIN_TOP        0
#define RG_SCREEN_MARGIN_BOTTOM     0
#define RG_SCREEN_MARGIN_LEFT       0
#define RG_SCREEN_MARGIN_RIGHT      0

#define RG_SCREEN_INIT() \
     ILI9341_CMD(0xCF, 0x00, 0xc3, 0x30);                                                                         \
    ILI9341_CMD(0xED, 0x64, 0x03, 0x12, 0x81);                                                                   \
    ILI9341_CMD(0xE8, 0x85, 0x00, 0x78);                                                                         \
    ILI9341_CMD(0xCB, 0x39, 0x2c, 0x00, 0x34, 0x02);                                                             \
    ILI9341_CMD(0xF7, 0x20);                                                                                     \
    ILI9341_CMD(0xEA, 0x00, 0x00);                                                                               \
    ILI9341_CMD(0xC0, 0x1B);                 /* Power control   //VRH[5:0] */                                    \
    ILI9341_CMD(0xC1, 0x12);                 /* Power control   //SAP[2:0];BT[3:0] */                            \
    ILI9341_CMD(0xC5, 0x32, 0x3C);           /* VCM control */                                                   \
    ILI9341_CMD(0xC7, 0x91);                 /* VCM control2 */                                                  \
    ILI9341_CMD(0x36, (0x40 | 0x80 | 0x08)); /* Memory Access Control */                                         \
    ILI9341_CMD(0xB1, 0x00, 0x10);           /* Frame Rate Control (1B=70, 1F=61, 10=119) */                     \
    ILI9341_CMD(0xB6, 0x0A, 0xA2);           /* Display Function Control */                                      \
    ILI9341_CMD(0xF6, 0x01, 0x30);                                                                               \
    ILI9341_CMD(0xF2, 0x00); /* 3Gamma Function Disable */                                                       \
    ILI9341_CMD(0x26, 0x01); /* Gamma curve selected */                                                          \
    ILI9341_CMD(0xE0, 0xD0, 0x00, 0x05, 0x0E, 0x15, 0x0D, 0x37, 0x43, 0x47, 0x09, 0x15, 0x12, 0x16, 0x19);	 \
    ILI9341_CMD(0xE1, 0xD0, 0x00, 0x05, 0x0D, 0x0C, 0x06, 0x2D, 0x44, 0x40, 0x0E, 0x1C, 0x18, 0x16, 0x19);

// Gamepad
#define RG_GAMEPAD_GPIO_MAP {\
    {RG_KEY_UP,     GPIO_NUM_18, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_DOWN,   GPIO_NUM_19, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_LEFT,   GPIO_NUM_20, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_RIGHT,  GPIO_NUM_21, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_SELECT, GPIO_NUM_38, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_START,  GPIO_NUM_39, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_A,      GPIO_NUM_40,  GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_B,      GPIO_NUM_1, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_MENU,   GPIO_NUM_2, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_OPTION, GPIO_NUM_8, GPIO_PULLUP_ONLY, 0},\
}

// Battery
#define RG_BATTERY_DRIVER           1
#define RG_BATTERY_ADC_UNIT         ADC_UNIT_1
#define RG_BATTERY_ADC_CHANNEL      ADC_CHANNEL_3
#define RG_BATTERY_CALC_PERCENT(raw) (((raw) * 2.f - 3500.f) / (4200.f - 3500.f) * 100.f)
#define RG_BATTERY_CALC_VOLTAGE(raw) ((raw) * 2.f * 0.001f)
[... 122 lines cut ...]

[0;32mI (130) boot:  7 fmsx             OTA app          00 14 003a0000 00090000[0m

[0;32mI (137) boot: End of partition table[0m

[0;32mI (142) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=2c748h (182088) map[0m

[0;32mI (178) esp_image: segment 1: paddr=0003c770 vaddr=3fc944c0 size=038a8h ( 14504) load[0m

[0;32mI (181) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=9d1e0h (643552) map[0m

[0;32mI (281) esp_image: segment 3: paddr=000dd208 vaddr=3fc97d68 size=00270h (   624) load[0m

[0;32mI (282) esp_image: segment 4: paddr=000dd480 vaddr=40374000 size=104c0h ( 66752) load[0m

[0;32mI (306) boot: Loaded app from partition at offset 0x10000[0m

[0;32mI (306) boot: Disabling RNG early entropy source...[0m

========================================================

launcher unknown (Feb 23 2025 10:08:42)

 built for: ESPLAY-S3. type: dev_`
can You help me becouse i am little frustarted. it supouse to be simple task but i cant do it :(

kazz2020, 2025-02-23:
> > did not help with sd. any idea why when i define trackball it acts like crazy?
> 
> In the latest config you've posted you have this:
> 
> ```
> #define RG_GPIO_SDSPI_MISO GPIO_NUM_12
> #define RG_GPIO_SDSPI_MOSI GPIO_NUM_13
> #define RG_GPIO_SDSPI_CLK GPIO_NUM_14
> #define RG_GPIO_SDSPI_CS GPIO_NUM_15
> ```
> 
> but according to this image the trackpad uses GPIO 15 so you have a conflict.
> 
> [image]
> 
> Based on the image I shared (I don't know how accurate it is, or if it's even the same device but I took it from your link), your SD card section should look like this:
> 
> ```
> #define RG_GPIO_SDSPI_MISO GPIO_NUM_38
> #define RG_GPIO_SDSPI_MOSI GPIO_NUM_41
> #define RG_GPIO_SDSPI_CLK GPIO_NUM_40
> #define RG_GPIO_SDSPI_CS GPIO_NUM_39
> ```

``#define RG_TARGET_NAME             "ESPLAY-S3"

// Storage
#define RG_STORAGE_ROOT             "/sd"
#define RG_STORAGE_SDSPI_HOST       SPI3_HOST
#define RG_STORAGE_SDSPI_SPEED      SDMMC_FREQ_DEFAULT
#define RG_STORAGE_ALLOW_FAILURE    1
//#define RG_STORAGE_SDMMC_HOST       SDMMC_HOST_SLOT_1
//#define RG_STORAGE_SDMMC_SPEED      SDMMC_FREQ_DEFAULT
// #define RG_STORAGE_FLASH_PARTITION  "vfs"

// Audio
#define RG_AUDIO_USE_INT_DAC        0   // 0 = Disable, 1 = GPIO25, 2 = GPIO26, 3 = Both
#define RG_AUDIO_USE_EXT_DAC        0   // 0 = Disable, 1 = Enable

// Video
#define RG_SCREEN_DRIVER            0   // 0 = ILI9341
#define RG_SCREEN_HOST             SPI2_HOST    // Keep SPI2 for LCD
#define RG_SCREEN_SPEED            SPI_MASTER_FREQ_20M  // Reduced speed to 20MHz
#define RG_SCREEN_BACKLIGHT         1   // Enable backlight control
#define RG_SCREEN_WIDTH            320
#define RG_SCREEN_HEIGHT           240
#define RG_SCREEN_ROTATE            1    // Keep at 1 for correct orientation
#define RG_SCREEN_MARGIN_TOP        0
#define RG_SCREEN_MARGIN_BOTTOM     0
#define RG_SCREEN_MARGIN_LEFT       0
#define RG_SCREEN_MARGIN_RIGHT      0

#define RG_SCREEN_INIT() \
     ILI9341_CMD(0xCF, 0x00, 0xc3, 0x30);                                                                         \
    ILI9341_CMD(0xED, 0x64, 0x03, 0x12, 0x81);                                                                   \
    ILI9341_CMD(0xE8, 0x85, 0x00, 0x78);                                                                         \
    ILI9341_CMD(0xCB, 0x39, 0x2c, 0x00, 0x34, 0x02);                                                             \
    ILI9341_CMD(0xF7, 0x20);                                                                                     \
    ILI9341_CMD(0xEA, 0x00, 0x00);                                                                               \
    ILI9341_CMD(0xC0, 0x1B);                 /* Power control   //VRH[5:0] */                                    \
    ILI9341_CMD(0xC1, 0x12);                 /* Power control   //SAP[2:0];BT[3:0] */                            \
    ILI9341_CMD(0xC5, 0x32, 0x3C);           /* VCM control */                                                   \
    ILI9341_CMD(0xC7, 0x91);                 /* VCM control2 */                                                  \
    ILI9341_CMD(0x36, (0x40 | 0x80 | 0x08)); /* Memory Access Control */                                         \
    ILI9341_CMD(0xB1, 0x00, 0x10);           /* Frame Rate Control (1B=70, 1F=61, 10=119) */                     \
    ILI9341_CMD(0xB6, 0x0A, 0xA2);           /* Display Function Control */                                      \
    ILI9341_CMD(0xF6, 0x01, 0x30);                                                                               \
    ILI9341_CMD(0xF2, 0x00); /* 3Gamma Function Disable */                                                       \
    ILI9341_CMD(0x26, 0x01); /* Gamma curve selected */                                                          \
    ILI9341_CMD(0xE0, 0xD0, 0x00, 0x05, 0x0E, 0x15, 0x0D, 0x37, 0x43, 0x47, 0x09, 0x15, 0x12, 0x16, 0x19);	 \
    ILI9341_CMD(0xE1, 0xD0, 0x00, 0x05, 0x0D, 0x0C, 0x06, 0x2D, 0x44, 0x40, 0x0E, 0x1C, 0x18, 0x16, 0x19);

// Gamepad
#define RG_GAMEPAD_GPIO_MAP {\
    {RG_KEY_UP,     GPIO_NUM_18, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_DOWN,   GPIO_NUM_19, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_LEFT,   GPIO_NUM_20, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_RIGHT,  GPIO_NUM_21, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_SELECT, GPIO_NUM_38, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_START,  GPIO_NUM_39, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_A,      GPIO_NUM_40,  GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_B,      GPIO_NUM_1, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_MENU,   GPIO_NUM_2, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_OPTION, GPIO_NUM_8, GPIO_PULLUP_ONLY, 0},\
}

// Battery
#define RG_BATTERY_DRIVER           1
#define RG_BATTERY_ADC_UNIT         ADC_UNIT_1
#define RG_BATTERY_ADC_CHANNEL      ADC_CHANNEL_3
#define RG_BATTERY_CALC_PERCENT(raw) (((raw) * 2.f - 3500.f) / (4200.f - 3500.f) * 100.f)
#define RG_BATTERY_CALC_VOLTAGE(raw) ((raw) * 2.f * 0.001f)

// SPI Display
#define RG_GPIO_LCD_MISO            GPIO_NUM_13
#define RG_GPIO_LCD_MOSI            GPIO_NUM_11
#define RG_GPIO_LCD_CLK             GPIO_NUM_12
#define RG_GPIO_LCD_CS              GPIO_NUM_10
#define RG_GPIO_LCD_DC              GPIO_NUM_15
#define RG_GPIO_LCD_BCKL            GPIO_NUM_14

// SPI SD Card
#define RG_GPIO_SDSPI_MISO          GPIO_NUM_7
#define RG_GPIO_SDSPI_MOSI          GPIO_NUM_5
#define RG_GPIO_SDSPI_CLK           GPIO_NUM_6
#define RG_GPIO_SDSPI_CS            GPIO_NUM_9

maybe this?

kilrah, 2025-06-11:
Some cleanup needed, but 

https://github.com/user-attachments/assets/c9f691a5-b1e3-4b80-b71b-57ed3ad3be4f

kilrah, 2025-06-11:
Pretty much done, PR updated

kilrah, 2025-06-26:
Guess this can be closed!