ducalex/retro-go #183: Ingame low battery indicator ~basic osd

rapha-tech, 2025-02-05:
I want to add an ingame low battery indicator (at <10% for example). After some research I found this TODO in rg_display.c:
```c
if (osd != NULL)
    {
        // TODO: Draw on screen display. By default it should be bottom left which is fine
        // for both virtual keyboard and info labels. Maybe make it configurable later...
    }
```

I was wondering if this was close to what I wanted to do ?

I'd like to tackle this, but I don't really know where to start:
1°) should I use the ```rg_surface_t osd``` to draw my battery icon on it ?
2°) should I combine the ```rg_surface_t update``` and ```rg_surface_t osd``` in ```write_update(const rg_surface_t *update)```  and then draw everything on screen ?

Also, I realized it would be really long to copy the whole rg_surface_t osd over the rg_surface_t update. So I though we could specify wich area of the osd should be copied over the screen.
I'm thinking about some struct like this:
```c
typedef struct {
   int x0, y0;
   int width, height;
} osd_frame_t;
```
To hold thoses "rectangular frames"

ducalex, 2025-02-05:
1. Yes you should draw your icon on a surface. Drawing directly on the screen would be very wasteful because you'd have to do it 30/60 times per seconds, versus once every many seconds for a surface. In other words copying a surface is very fast, drawing shapes or text is very slow.

2. I think it's best/simpler to let the game display update be sent to the LCD fully, as it is currently, and then send the OSD separately at the end. Compositing both together might save 1-2 SPI transactions but I suspect it will take a lot more CPU. It would also be more complicated because the update doesn't always cover the entire screen so the OSD might be outside its boundaries. 
 
3. Your osd_frame_t struct already exists in retro-go it's called rg_rect_t (defined in rg_surface.h). But I don't know if it's really needed because in my mind OSD will always stick to one corner and those coordinates can be calculated based on screen size and the osd surface size. (eg `left = display.screen.width - osd->width`). 

This is how I intended OSD to be implemented:
- Add a way to make rg_gui_draw_* work on surfaces rather than just the screen. It has to be done in a way that we don't have to change all the code that use the functions though, so it's probably creating wrappers which is annoying boilerplate. I think that's why I never finished the OSD system, actually, being unable to draw anything without big changes.
- Add a function to rg_display to set the osd surface and its position. Position could be a corner, or left/top if you prefer (though personally I would stick to corners). Then you have to figure how you want `osd` ownership to work, because we need a way to set/free/replace it without leaking. We might also need a locking mechanism because we don't want to modify the osd object if it's currently being sent to the display. But we might get away without it, so eh, wait and see :)
- Then in the block you've quoted it would contain code to send the osd to the screen. I seem to remember that calling rg_display_write_rect inside write_update isn't particularly safe or efficient so it will likely have to be the lcd_* functions. But the gist would be like `rg_display_write_rect(left, top, osd->width, osd->height, osd->stride, osd->buffer, RG_DISPLAY_WRITE_NOSYNC);`.
- And finally to actually put something on the osd it would be like `rg_surface_t *osd = rg_surface_create(16, 16); rg_gui_draw_...(...osd...); rg_display_put_the_cool_osd_on_screen(osd, top_right);`. Then of course you have to decide where to put that osd updating code. I suppose `system_monitor_task` would be the obvious place to have a battery or fps indicator, but I'm not sure I like it. Though I'm not opposed to it either because I don't have a better idea at the moment.

Note that this is just how I remember having planned things to work. I'm sure you'll encounter issues that I haven't considered, so feel free to diverge if needed or do things entirely differently if you have a better idea! Just keep in mind that it's best to keep changes to existing code to a minimum. For example if you refactor how rg_display or rg_gui works entirely to accommodate your cool new OSD, it will likely take us a lot longer before it can be merged (even if it's objectively better!).

rapha-tech, 2025-02-11:
So I improved the code a little, now it produces decent results:

[image]

I also modified the "osd_init" function to work with corners !

Now I just have to adapt the code to work with all formats like "scaling", "zoom"... right now only "full" is working properly
Also I still need to find a way to reduce flickering.
https://github.com/rapha-tech/retro-go/tree/dev
Let me know if you want me to open a PR ?

ducalex, 2025-03-04:
Very nice to see it working!

I worry a bit that supporting transparency introduces a lot of complexity (as you've seen). I can think of a lot of ways to make it work better and reduce or even eliminate the flicker you've mentioned (and the performance hit that likely goes with it) if that's the direction you want to go, and all in all it would result in a higher quality rendering system so it's a worthwhile pursuit.

But it seems like a big price to pay just for OSD transparency, no? Because it wouldn't stop there, for transparency to work correctly we'd probably also need to introduce something like text shadow otherwise it would often end up unreadable in many scenes.

Whatever you decide maybe it's time to open a PR indeed. You have working code that does exactly what you proposed, that's more than many PRs :) It will also allow me to help you more directly with the code.

rapha-tech, 2025-03-13:
Yes I realised that transparency involved a lot of complexity but if that can upgrade the rendering system I feel like it should be worth to take the time to do it properly!
Anyway, I'm closing this issue since I opened the PR

closed 2025-03-13 by rapha-tech