ducalex/retro-go #185: keep restarting while running on ESP32s3 need help

hellohale, 2025-02-19:
keep restarting while running on ESP32s3
I have been using Arduino before. When compiling a project with ESP idf for the first time, I encountered some problems. I request your help to investigate the issue. The details are as follows:
While running, an hourglass appeared on the screen, and the LCD should be running normally. According to the serial port output information, the SD card is running normally, but the config file is missing and it keeps restarting
The first question is, what are the total configuration files that need to be placed on SD, what is the path structure of the files, and where can I download those configuration files.
The second question is about keep restarting.
My environment is esp-idf v5.08, with hardware including esp32s3-cevkit-c (n16r8), LCD, Joystick, and SDSPI modules, but not connected to I2S DAC
The command I used for compilation is: python rg_tool.py --target esp32s3-devkit-c --port COM3 install launcher retro-core
Due to the original code failing to compile directly, I changed the '% d' in one place to '% ld' according to the compilation prompt.

E:/Program/esp32s3-idf/retro-go-1/retro-core/components/snes9x/src/memmap.c:612:62: error: format '%d' expects argument of type 'int', but argument 5 has type 'long unsigned int' [-Werror=format=]
  612 |    printf("Rom loaded: name: %s, id: %s, company: %s, size: %dKB\n", Memory.ROMName, Memory.ROMId, Memory.CompanyId, Memory.CalculatedSize / 1024);
      |                                                             ~^                                                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                              |                                                                             |
      |                                                              int                                                                           long unsigned int
      |                                                             %ld

Compiled successfully after modifying according to the prompts.

In addition, I made the following modifications to the config. h under esp32s3-devkit-c: there may be a problem with the settings of the pins in the original code of the sdSPI module, which cannot access the SD card. Therefore, I made some adjustments to the ADC button and GPIO button.
When I didn't modify the DAC button parameters, the LCD displayed three options for Recoverymode, and the cursor kept moving down and looping. I modified the voltage range to keep the cursor still. But after changing the voltage, the hourglass is displayed directly, and the menu for Recoverymode does not appear, and continuous restarts.

#define RG_GAMEPAD_ADC_MAP {\
    {RG_KEY_UP,    ADC_UNIT_1, ADC_CHANNEL_0, ADC_ATTEN_DB_11, 3072, 4096},\
    {RG_KEY_RIGHT, ADC_UNIT_1, ADC_CHANNEL_3, ADC_ATTEN_DB_11, 0, 1024},\
    {RG_KEY_DOWN,  ADC_UNIT_1, ADC_CHANNEL_0, ADC_ATTEN_DB_11, 0, 1024},\
    {RG_KEY_LEFT,  ADC_UNIT_1, ADC_CHANNEL_3, ADC_ATTEN_DB_11, 3072, 4096},\
}
/*
#define RG_GAMEPAD_ADC_MAP {\
   {RG_KEY_UP,    ADC_UNIT_1, ADC_CHANNEL_0, ADC_ATTEN_DB_11, 3072, 4096},\
    {RG_KEY_RIGHT, ADC_UNIT_1, ADC_CHANNEL_3, ADC_ATTEN_DB_11, 1024, 3071},\
   {RG_KEY_DOWN,  ADC_UNIT_1, ADC_CHANNEL_0, ADC_ATTEN_DB_11, 1024, 3071},\
    {RG_KEY_LEFT,  ADC_UNIT_1, ADC_CHANNEL_3, ADC_ATTEN_DB_11, 3072, 4096},\
}
*/

#define RG_GAMEPAD_GPIO_MAP {\
    {RG_KEY_SELECT, GPIO_NUM_42, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_START,  GPIO_NUM_6, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_MENU,   GPIO_NUM_41, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_OPTION, GPIO_NUM_15,  GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_A,      GPIO_NUM_7, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_B,      GPIO_NUM_40,  GPIO_PULLUP_ONLY, 0},\
}

/*
#define RG_GAMEPAD_GPIO_MAP {\
    {RG_KEY_SELECT, GPIO_NUM_16, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_START,  GPIO_NUM_17, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_MENU,   GPIO_NUM_18, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_OPTION, GPIO_NUM_8,  GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_A,      GPIO_NUM_15, GPIO_PULLUP_ONLY, 0},\
    {RG_KEY_B,      GPIO_NUM_5,  GPIO_PULLUP_ONLY, 0},\
}
*/

// SPI Display (back up working)
#define RG_GPIO_LCD_MISO            GPIO_NUM_NC
#define RG_GPIO_LCD_MOSI            GPIO_NUM_9//12
#define RG_GPIO_LCD_CLK             GPIO_NUM_48
#define RG_GPIO_LCD_CS              GPIO_NUM_NC
#define RG_GPIO_LCD_DC              GPIO_NUM_47
#define RG_GPIO_LCD_BCKL            GPIO_NUM_39
#define RG_GPIO_LCD_RST             GPIO_NUM_3

#define RG_GPIO_SDSPI_MISO          GPIO_NUM_13 //9
#define RG_GPIO_SDSPI_MOSI          GPIO_NUM_11
#define RG_GPIO_SDSPI_CLK           GPIO_NUM_12 //13
#define RG_GPIO_SDSPI_CS            GPIO_NUM_10

The serial port output information is as follows, continuously looping:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x14b0
load:0x403c9700,len:0xd70
load:0x403cc700,len:0x2ec0
entry 0x403c9920
I (24) boot: ESP-IDF HEAD-HASH-NOTFOUND 2nd stage bootloader
I (25) boot: compile time 15:43:00
I (25) boot: Multicore bootloader
I (28) boot: chip revision: v0.2
I (32) qio_mode: Enabling default flash chip QIO
I (37) boot.esp32s3: Boot SPI Speed : 80MHz
I (42) boot.esp32s3: SPI Mode       : QIO
I (46) boot.esp32s3: SPI Flash Size : 16MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (75) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (82) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  3 launcher         OTA app          00 10 00010000 000f0000
I (97) boot:  4 retro-core       OTA app          00 11 00100000 00100000
I (105) boot: End of partition table
I (109) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=2fb48h (195400) map
I (147) esp_image: segment 1: paddr=0003fb70 vaddr=3fc99400 size=004a8h (  1192) load
I (148) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=a6284h (680580) map
I (256) esp_image: segment 3: paddr=000e62ac vaddr=3fc998a8 size=03a60h ( 14944) load
I (259) esp_image: segment 4: paddr=000e9d14 vaddr=40374000 size=153d8h ( 87000) load
I (287) boot: Loaded app from partition at offset 0x10000
I (287) boot: Disabling RNG early entropy source...
W (733) i2s(legacy): legacy i2s driver is deprecated, please migrate to use driver/i2s_std.h, driver/i2s_pdm.h or driver/i2s_tdm.h
W (734) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`

========================================================
launcher unknown (Feb 19 2025 17:11:39)
 built for: ESP32S3-DEVKIT-C. type: dev
========================================================

[info] rg_storage_init: Looking for SD Card using SDSPI...
[info] rg_storage_init: Storage mounted at /sd.
[info] rg_input_init: Initializing ADC gamepad driver...
[info] rg_input_init: Initializing GPIO gamepad driver...
[info] rg_input_init: Initializing ADC battery driver...
[info] rg_input_init: Input ready. state=00000000 00000000
[error] rg_storage_read_file: Fopen failed (2): '/sd/retro-go/config/global.json'
[error] rg_storage_read_file: Fopen failed (2): '/sd/retro-go/config/boot.json'
[info] rg_display_init: Initialization...
[error] rg_storage_read_file: Fopen failed (2): '/sd/retro-go/config/launcher.json'
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fcae500
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fcaef04
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fcaf908
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fcb030c
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fcb0d10
[info] lcd_set_backlight: backlight set to 80%
[info] rg_display_init: Display ready.
[info] rg_alloc: SIZE=11520, CAPS=SPIRAM|8BIT, PTR=0x3c0e97b4
[info] rg_gui_set_language_id: Language set to: English (0)
[info] rg_gui_set_font: Font set to: points=12, scaling=1.00
[info] rg_gui_set_theme: Using built-in theme!
[info] rg_audio_init: Audio ready. sink='Ext DAC', samplerate=32000, volume=50

hellohale, 2025-02-22:
After testing, it was found that there was an error reading from the SD card. However, according to the records, the file had been successfully read from the SD card multiple times during runtime, and the error began at the step of reading clock.bin, causing the system to crash and restart. Could you please advise on the reason? Thank you!
===================================
The source code is:
bool rg_storage_read_file(const char *path, void **data_out, size_t *data_len, uint32_t flags)
{
    printf("[DEBUG]File path %s\n",path);
    RG_ASSERT_ARG(data_out && data_len);
    printf("[DEBUG]Check path \n");
    CHECK_PATH(path);
    size_t output_buffer_alloc_size;
    size_t output_buffer_size;
    void *output_buffer;
    size_t file_size;
    printf("[DEBUG]Test File open\n");
    FILE *ftest = fopen("/sd/retro-go/test.txt", "rb");
    if (!ftest)
    {
        RG_LOGE("Test Fopen failed\n");
    }
    fclose(ftest);
    printf("[DEBUG]File open\n");
    FILE *fp = fopen(path, "rb");
    if (!fp)
    {
        RG_LOGE("Fopen failed (%d): '%s'", errno, path);
        return false;
    }
    printf("[DEBUG]File open seccesful\n");
    fseek(fp, 0, SEEK_END);
    file_size = ftell(fp);

    fseek(fp, 0, SEEK_SET);

    if (flags & RG_FILE_USER_BUFFER)
    {
        output_buffer_alloc_size = *data_len;
        output_buffer_size = RG_MIN(*data_len, file_size);
        output_buffer = *data_out;
    }
    else
    {
        size_t blocksize = RG_MAX(0x400, (flags & 0xF) * 0x2000);
        output_buffer_alloc_size = (file_size + (blocksize - 1)) & ~(blocksize - 1);
        output_buffer_size = file_size;
        output_buffer = malloc(output_buffer_alloc_size);
    }

    if (!output_buffer)
    {
        RG_LOGE("Memory allocation failed: '%s'", path);
        fclose(fp);
        return false;
    }

    if (!fread(output_buffer, output_buffer_size, 1, fp))
    {
        RG_LOGE("File read failed (%d): '%s'", errno, path);
        fclose(fp);
        if (!(flags & RG_FILE_USER_BUFFER))
            free(output_buffer);
        return false;
    }
    printf("[DEBUG]Close File %s\n",path);
    fclose(fp);

    // Wipe the extra allocated space, if any
    if (output_buffer_alloc_size > output_buffer_size)
    {
        memset(output_buffer + output_buffer_size, 0, output_buffer_alloc_size - output_buffer_size);
    }

    *data_out = output_buffer;
    *data_len = output_buffer_size;
    return true;
}

==================================
The serial port output is:
========================================================
launcher unknown (Feb 22 2025 22:41:48)
 built for: ESP32S3-DEVKIT-C. type: dev
========================================================

[info] rg_storage_init: Looking for SD Card using SDSPI...
[info] rg_storage_init: Storage mounted at /sd.
[info] rg_input_init: Initializing ADC gamepad driver...
[info] rg_input_init: Initializing GPIO gamepad driver...
[info] rg_input_init: Initializing ADC battery driver...
[info] rg_input_init: Input ready. state=00000000 00000000
[DEBUG]File path /sd/retro-go/config/global.json
[DEBUG]Check path
[DEBUG]Test File open
[DEBUG]File open
[DEBUG]File open seccesful
[DEBUG]Close File /sd/retro-go/config/global.json
[info] json_root: Config file loaded: '/sd/retro-go/config/global.json'
[DEBUG]File path /sd/retro-go/config/boot.json
[DEBUG]Check path 
[DEBUG]Test File open
[DEBUG]File open
[DEBUG]File open seccesful
[DEBUG]Close File /sd/retro-go/config/boot.json
[info] json_root: Config file loaded: '/sd/retro-go/config/boot.json'
[info] rg_display_init: Initialization...
[DEBUG]File path /sd/retro-go/config/launcher.json
[DEBUG]Check path
[DEBUG]Test File open
[DEBUG]File open
[DEBUG]File open seccesful
[DEBUG]Close File /sd/retro-go/config/launcher.json
[info] json_root: Config file loaded: '/sd/retro-go/config/launcher.json'
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fcaea90
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fcaf494
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fcafe98
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fcb089c
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fcb12a0
[info] lcd_set_backlight: backlight set to 80%
[info] rg_display_init: Display ready.
[info] rg_alloc: SIZE=11520, CAPS=SPIRAM|8BIT, PTR=0x3c0e97b4
[info] rg_gui_set_language_id: Language set to: English (0)
[info] rg_gui_set_font: Font set to: points=12, scaling=1.00
[DEBUG]File path /sd/retro-go/themes/default/theme.json
[DEBUG]Check path
[DEBUG]Test File open
[DEBUG]File open
[DEBUG]File open seccesful
[DEBUG]Close File /sd/retro-go/themes/default/theme.json
[info] rg_gui_set_theme: Theme set to 'default'!
[DEBUG] start rg_audio_init
[DEBUG] mutex give start[info] rg_audio_init: Audio ready. sink='Ext DAC', samplerate=32000, volume=50
[Debug]audio init last setp[DEBUG] mutex give start[DEBUG] start set_timezone 
[DEBUG] system_load_time
[DEBUG]RG_MAX
[DEBUG]clock.bin
[DEBUG]File path /sd/retro-go/cache/clock.bin
[DEBUG]Check path
[DEBUG]Test File open

ducalex, 2025-02-22:
>  The first question is, what are the total configuration files that need to be placed on SD, what is the path structure of the files, and where can I download those configuration files.

There is nothing to do in that regard, retro-go will create everything that is missing!

Regarding your actual problem, I don't know why it crashes on the foppen (does it even crash? Do you see a stack trace before it reboots?) I can think of a few ways it could crash later in the rg_storage_read_file function but not on the fopen itself, this is weird. It kind of implies a crash on esp-idf's side?

Anyway I will do some testing and come back to you soon.

In the mean time you can try just removing the rg_storage_read_file(...clock.bin...) read entirely, it's not important and maybe it will allow you to boot for now.

hellohale, 2025-02-23:
I have found a solution, After I changed the External I2S DAC port, it was fine.
Although I don't know why this happened.

At First:
// External I2S DAC
#define RG_GPIO_SND_I2S_BCK         35//41  BCLK
#define RG_GPIO_SND_I2S_WS          36//42  LRC
#define RG_GPIO_SND_I2S_DATA        37//40  DIN

Change to:
// External I2S DAC
#define RG_GPIO_SND_I2S_BCK         17//41  BCLK
#define RG_GPIO_SND_I2S_WS          18//42  LRC
#define RG_GPIO_SND_I2S_DATA        16//40  DIN

hellohale, 2025-02-23:
After reading the ESP document, GPIO35, 36,37 have already been used for PSRAM, which may have caused issues. A foolish mistake

ducalex, 2025-03-04:
Thanks for sharing your solution! So the issue has been resolved?

AshrafHanyy, 2025-05-24:
I have same issue with esp32 s3 wroom n16r8 and your fix didn't solve this issue to me

my log
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x16e0
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (27) boot: ESP-IDF v5.1 2nd stage bootloader
I (27) boot: compile time May 22 2025 13:25:19
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (72) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (79) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  3 launcher         OTA app          00 10 00010000 00130000
I (94) boot:  4 retro-core       OTA app          00 11 00140000 00100000
I (102) boot:  5 prboom-go        OTA app          00 12 00240000 000d0000
I (109) boot:  6 gwenesis         OTA app          00 13 00310000 000f0000
I (117) boot:  7 fmsx             OTA app          00 14 00400000 000a0000
I (124) boot: End of partition table
I (129) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=43b48h (277320) map
I (187) esp_image: segment 1: paddr=00053b70 vaddr=3fc9a300 size=04a74h ( 19060) load
I (191) esp_image: segment 2: paddr=000585ec vaddr=40374000 size=07a2ch ( 31276) load
I (200) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=c5720h (808736) map
I (346) esp_image: segment 4: paddr=00125748 vaddr=4037ba2c size=0e7e0h ( 59360) load
I (369) boot: Loaded app from partition at offset 0x10000
I (369) boot: Disabling RNG early entropy source...
I (380) cpu_start: Multicore app
I (381) octal_psram: vendor id    : 0x0d (AP)
I (381) octal_psram: dev id       : 0x02 (generation 3)
I (384) octal_psram: density      : 0x03 (64 Mbit)
I (389) octal_psram: good-die     : 0x01 (Pass)
I (395) octal_psram: Latency      : 0x01 (Fixed)
I (400) octal_psram: VCC          : 0x01 (3V)
I (405) octal_psram: SRF          : 0x01 (Fast Refresh)
I (411) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (417) octal_psram: BurstLen     : 0x01 (32 Byte)
I (422) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (428) octal_psram: DriveStrength: 0x00 (1/1)
I (434) esp_psram: Found 8MB PSRAM device
I (438) esp_psram: Speed: 40MHz
I (442) cpu_start: Pro cpu up.
I (446) cpu_start: Starting app cpu, entry point is 0x40375750
--- 0x40375750: call_start_cpu1 at /Users/ashrafhany/esp/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (1183) esp_psram: SPI SRAM memory test OK
I (1192) cpu_start: Pro cpu start user code
I (1193) cpu_start: cpu freq: 160000000 Hz
I (1193) cpu_start: Application information:
I (1196) cpu_start: Project name:     launcher
I (1201) cpu_start: App version:      1.44-dirty
I (1206) cpu_start: Compile time:     May 22 2025 15:33:33
I (1212) cpu_start: ELF file SHA256:  3cda8722ffa2ef79...
I (1218) cpu_start: ESP-IDF:          v5.1
I (1223) cpu_start: Min chip rev:     v0.0
I (1228) cpu_start: Max chip rev:     v0.99
I (1233) cpu_start: Chip rev:         v0.2
[... 81 lines cut ...]
I (2355) wifi:wifi certification version: v7.0
I (2365) wifi:config NVS flash: enabled
I (2365) wifi:config nano formating: disabled
I (2365) wifi:Init data frame dynamic rx buffer num: 32
I (2375) wifi:Init management frame dynamic rx buffer num: 32
I (2375) wifi:Init management short buffer num: 32
I (2385) wifi:Init static tx buffer num: 16
I (2385) wifi:Init tx cache buffer num: 32
I (2395) wifi:Init static tx FG buffer num: 2
I (2395) wifi:Init static rx buffer size: 1600
I (2405) wifi:Init static rx buffer num: 10
I (2405) wifi:Init dynamic rx buffer num: 32
I (2405) wifi_init: rx ba win: 6
I (2415) wifi_init: tcpip mbox: 32
I (2415) wifi_init: udp mbox: 6
I (2415) wifi_init: tcp mbox: 6
I (2425) wifi_init: tcp tx win: 5744
I (2425) wifi_init: tcp rx win: 5744
I (2435) wifi_init: tcp mss: 1440
I (2435) wifi_init: WiFi IRAM OP enabled
I (2445) wifi_init: WiFi RX IRAM OP enabled
[error] rg_storage_read_file: Fopen failed (22): '/SD/retro-go/config/wifi.json'
[debug] rg_network_wifi_read_config: Looking for 'ssid0' (slot 0)
[info] webui_start: Web server started
[info] update_viewport_scaling: 320x240@1.333 => 320x240@1.333 left:0 top:0 step_x:1.00 step_y:1.00