ducalex/retro-go #199: Unable to flash Rachel-ESP32

GameDevHowl, 2025-05-19:
## Describe the bug ##
Upon flashing the Rachel-ESP32 trying to turn it on just flashes the blacklight once and nothing happens.

## To Reproduce ##
Steps to reproduce the behavior:
1. Build with 4.4.8
2. Plug in while holding boot.  Flash using appropriate port and target, get positive results in CLI
3. Unplug, try to use, nothing. 

Am I missing a step?

ducalex, 2025-05-25:
I don't have a Rachel-ESP32 to test unfortunately.

Can you share the command used for flashing as well as a log of the device booting?

ioiotor, 2025-05-29:
Rachel-esp32 and Forairaaaa's Rachel are different devices

GameDevHowl, 2025-06-09:
> Rachel-esp32 and Forairaaaa's Rachel are different devices

Can you link to more info on the version of the Rachel-ESP32 you have?

GameDevHowl, 2025-06-14:
> Rachel-esp32 and Forairaaaa's Rachel are different devices

I literally cannot find any other Rachel-ESP32 that looks like this other than the one I linked. It matches the picture in the device list, minus the red button on top.

I'm at a loss with this.

GameDevHowl, 2025-06-14:
I give up.
ioiotor's assertion that these are inherently different devices doesn't help when they don't back it up. 
Giving up on esp32 in general thanks to this.

closed 2025-06-14 by GameDevHowl