ducalex/retro-go #200: 3.2" ili3941 screen rotate

crashedash32, 2025-05-21:
Hi, 

would someone be able to tell me how to finish this line so i get the screen rotated 180? 

ILI9341_CMD(0x36, 

I have tried

ILI9341_CMD(0x36, 0xC0, 0x48)
ILI9341_CMD(0x36, 0x48, 0xC0)

tried 0xC0 and 0x48 on their own, i just get a corrupt 90 degree rotate.

i have googled, but find nothing that really helps, sorry if it is obvious to you clever chaps..

but i am really stumped..

thanks in advance for any help offered..

ducalex, 2025-05-25:
0x36 takes only cares about the lower 8 bits argument so you should only have two values in your ILI9341_CMD. 

Possibly what you want is `ILI9341_CMD(0x36, 0xC0 | 0x48)` instead?

AshrafHanyy, 2025-05-25:
this got me a corrupt screen as well

AshrafHanyy, 2025-05-25:
working solutino for me:
`ILI9341_CMD(0x36, 0xA8); `

crashedash32, 2025-05-29:
Hi, sorry for the late reply, i have been away from home.

Thank you so much for coming back to me. @AshrafHanyy that suggestion worked perfectly for me.

i can now focus on why i cant get the SD card to work. LOL.. 

thanks again for your help!!

closed 2025-05-29 by crashedash32