ducalex/retro-go #203: Issues adding new I2C controller (MCP23017)

DynaMight1124, 2025-05-31:
Hi

As I mentioned in another thread, I have a Cheap Yellow Display (CYD) that I have been playing with trying to get RetroGo running. It doesnt have very many free GPIO so relies on I2C. I had built up a controller PCB using a MCP23017 IO expander, this was originally designed to work with Galagino and it works perfectly with that. I used AI to adapt the Wii remote code to work with the MCP. All buttons are pressed to ground, which seems to be perfectly fine for Galagino.

So I thought why cant I use it for RetroGo as well! Been taking 1 step forwards and 2 steps back for ages with AI now. I'm no coder so deffo something I wouldnt be able to do by myself.

I've uploaded the most recent 'working' code here: https://github.com/DynaMight1124/retro-go/tree/CYD

I've been trying to keep AI from totally re-writing all the code and just keep it basic, the more I did this the better it seemed to get. I am not sure if its re-written something that is now causing an issue.

It essentially recognises and connects to the MCP but I get phantom key presses (which dont happen at all in Galagino) and no keys register when I press them. 

Heres the serial output. The 'keypresses' at the bottom are the phantom ones. 

Any help would be amazing. 
[17:15:14]
[17:15:14]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[17:15:14]configsip: 0, SPIWP:0xee
[17:15:14]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[17:15:14]mode:DIO, clock div:1
[17:15:14]load:0x3fff0030,len:6004
[17:15:14]load:0x40078000,len:15564
[17:15:14]load:0x40080400,len:4472
[17:15:14]entry 0x400806a4
[17:15:14][0;32mI (27) boot: ESP-IDF v4.4.8-dirty 2nd stage bootloader[0m
[17:15:14][0;32mI (27) boot: compile time 17:14:02[0m
[17:15:15][0;32mI (27) boot: Multicore bootloader[0m
[17:15:15][0;32mI (31) boot: chip revision: v3.1[0m
[17:15:15][0;32mI (35) qio_mode: Enabling default flash chip QIO[0m
[17:15:15][0;32mI (40) boot.esp32: SPI Speed      : 80MHz[0m
[17:15:15][0;32mI (45) boot.esp32: SPI Mode       : QIO[0m
[17:15:15][0;32mI (49) boot.esp32: SPI Flash Size : 4MB[0m
[17:15:15][0;32mI (54) boot: Enabling RNG early entropy source...[0m
[17:15:15][0;32mI (59) boot: Partition Table:[0m
[17:15:15][0;32mI (63) boot: ## Label            Usage          Type ST Offset   Length[0m
[17:15:15][0;32mI (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[17:15:15][0;32mI (78) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[17:15:15][0;32mI (85) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[17:15:15][0;32mI (93) boot:  3 launcher         OTA app          00 10 00010000 000e0000[0m
[17:15:15][0;32mI (100) boot:  4 retro-core       OTA app          00 11 000f0000 000f0000[0m
[17:15:15][0;32mI (108) boot: End of partition table[0m
[17:15:15][0;32mI (112) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=27680h (161408) map[0m
[17:15:15][0;32mI (165) esp_image: segment 1: paddr=000376a8 vaddr=3ffb0000 size=033c4h ( 13252) load[0m
[17:15:15][0;32mI (169) esp_image: segment 2: paddr=0003aa74 vaddr=40080000 size=055a4h ( 21924) load[0m
[17:15:15][0;32mI (178) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=8d348h (578376) map[0m
[17:15:15][0;32mI (339) esp_image: segment 4: paddr=000cd370 vaddr=400855a4 size=0cbcch ( 52172) load[0m
[17:15:15][0;32mI (365) boot: Loaded app from partition at offset 0x10000[0m
[17:15:15][0;32mI (365) boot: Disabling RNG early entropy source...[0m
[17:15:15]
[17:15:15]========================================================
[17:15:15]launcher 1.44-dirty (May 31 2025 17:13:17)
[17:15:15] built for: CYD. type: release
[17:15:15]========================================================
[17:15:15]
[17:15:15][info] rg_storage_init: Looking for SD Card using SDSPI...
[17:15:15][0m[info] rg_storage_init: Storage mounted at /sd.
[17:15:15][0m[info] rg_input_init: [rg_input_init] Initializing I2C gamepad driver...
[17:15:15][0m[info] rg_i2c_init: I2C driver ready (SDA:22 SCL:27).
[17:15:15][0m[info] rg_input_init: [rg_input_init] I2C master init successful. Calling rg_i2c_gpio_init().
[17:15:15][0m[info] rg_input_init: [rg_input_init] rg_i2c_gpio_init() SUCCESSFUL.
[17:15:15][0m[info] rg_input_read_gamepad_raw: MCP23017 Raw: A=0x00 B=0x00 => Combined=0x0000
[17:15:15][0m[info] rg_input_init: Input ready. state=00000000 00000000
[17:15:15][0m[info] json_root: Config file loaded: '/sd/retro-go/config/global.json'
[17:15:15][0m[31m[error] rg_storage_read_file: Fopen failed (2): '/sd/retro-go/config/boot.json'
[17:15:15][0m[info] rg_display_init: Initialization...
[17:15:15][0m[info] json_root: Config file loaded: '/sd/retro-go/config/launcher.json'
[17:15:15][0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3ffc6bc0
[17:15:15][0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3ffc75c4
[17:15:15][0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3ffc7fc8
[17:15:15][0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3ffc89cc
[17:15:15][0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3ffc93d0
[17:15:15][0m[info] lcd_set_backlight: backlight set to 50%
[17:15:15][0m[info] rg_display_init: Display ready.
[17:15:15][0m[info] rg_alloc: SIZE=11520, CAPS=SPIRAM|8BIT, PTR=0x3f8061a8
[17:15:15][0m[info] rg_gui_set_language_id: Language set to: English (0)
[17:15:15][0m[info] rg_gui_set_font: Font set to: points=12, scaling=1.00
[17:15:15][0m[info] rg_gui_set_theme: Theme set to 'Art Book Next'!
[17:15:15][0m[info] rg_audio_init: Audio ready. sink='Speaker', samplerate=32000, volume=100
[17:15:15][0m[31m[error] rg_storage_read_file: Fopen failed (2): '/sd/retro-go/cache/clock.bin'
[17:15:15][0m[info] rg_system_load_time: Time is now: Sat May 31 16:13:08 2025
[17:15:15]
[17:15:15][0m[info] rg_system_init: Available memory: 225/317 + 4060/4093
[17:15:15][0m[info] rg_system_init: Retro-Go ready.
[17:15:15]
[17:15:15][0m[info] rg_alloc: SIZE=153632, CAPS=SPIRAM|8BIT, PTR=0x3f808eac
[17:15:15][0m[info] gui_add_tab: Tab 'nes' added at index 0
[17:15:15][0m[info] gui_add_tab: Tab 'snes' added at index 1
[17:15:15][0m[info] gui_add_tab: Tab 'gb' added at index 2
[17:15:15][0m[info] gui_add_tab: Tab 'gbc' added at index 3
[17:15:15][0m[info] gui_add_tab: Tab 'gw' added at index 4
[17:15:15][0m[info] gui_add_tab: Tab 'sms' added at index 5
[17:15:15][0m[info] gui_add_tab: Tab 'gg' added at index 6
[17:15:15][0m[info] application: Application 'Sega Mega Drive' (gwenesis) not present, skipping
[17:15:15][0m[info] gui_add_tab: Tab 'col' added at index 7
[17:15:15][0m[info] gui_add_tab: Tab 'pce' added at index 8
[17:15:15][0m[info] gui_add_tab: Tab 'lnx' added at index 9
[17:15:15][0m[info] application: Application 'DOOM' (prboom-go) not present, skipping
[17:15:15][0m[info] application: Application 'MSX' (fmsx) not present, skipping
[17:15:15][0m[info] gui_add_tab: Tab 'favorite' added at index 10
[17:15:15][0m[info] gui_add_tab: Tab 'recent' added at index 11
[17:15:15][0m[33m[warn] book_load: Unknown path form: '/sd/roms/md/Sonic the Hedgehog 2 (World).md'
[17:15:15][0m[33m[warn] book_load: Unknown path form: '/sd/roms/md/Aladdin.zip'
[17:15:15][0m[33m[warn] book_load: Unknown path form: '/sd/roms/doom/DOOM2.WAD'
[17:15:15][0m[33m[warn] book_load: Unknown path form: '/sd/roms/md/Streets of Rage 2.zip'
[17:15:15][0m[33m[warn] book_load: Unknown path form: '/sd/roms/md/Sonic the Hedgehog 2.zip'
[17:15:15][0m[info] gui_resize_list: Resized list 'recent' from 0 to 22 items (new capacity: 32)
[17:15:15][0m[info] json_root: Config file loaded: '/sd/retro-go/config/wifi.json'
[17:15:15][0m[info] webui_start: Web server started
[17:15:16][0m[info] update_viewport_scaling: 320x240@1.333 => 320x240@1.333 left:0 top:0 step_x:1.00 step_y:1.00
[17:15:17][0m[info] rg_input_read_gamepad_raw: MCP23017 Raw: A=0x08 B=0x00 => Combined=0x0008
[17:15:17][0m[info] rg_input_read_gamepad_raw: MCP23017 Raw: A=0x00 B=0x00 => Combined=0x0000
[17:15:17][0m[info] rg_input_read_gamepad_raw: MCP23017 Raw: A=0x08 B=0x00 => Combined=0x0008
[17:15:17][0m[info] rg_input_read_gamepad_raw: MCP23017 Raw: A=0x00 B=0x00 => Combined=0x0000
[17:15:17][0m[info] rg_input_read_gamepad_raw: MCP23017 Raw: A=0x08 B=0x00 => Combined=0x0008
[17:15:17][0m[info] rg_input_read_gamepad_raw: MCP23017 Raw: A=0x00 B=0x00 => Combined=0x0000
[17:15:17][0m[info] rg_input_read_gamepad_raw: MCP23017 Raw: A=0x0C B=0x00 => Combined=0x000C
[17:15:17][0m[info] rg_input_read_gamepad_raw: MCP23017 Raw: A=0x00 B=0x00 => Combined=0x0000
[17:15:17][0m[info] rg_input_read_gamepad_raw: MCP23017 Raw: A=0x0C B=0x00 => Combined=0x000C
[17:15:17][0m[info] rg_input_read_gamepad_raw: MCP23017 Raw: A=0x00 B=0x00 => Combined=0x0000

DynaMight1124, 2025-06-25:
Hi

Thank you for looking into this. I'll try to keep it concise :) very short story is I did bin off the MCP23017 and tried a PCF8575, had AI 'help' with the code again and it pretty much sorted it first time, I guess it was pretty close to the PCF9539 so I have been using that.

However I still have the MCP23017 board so I downloaded the latest dev, added the CYD target, changed to the above defines as you suggested, I did also change my controls based on the ones you amended already: 

> #define RG_GAMEPAD_I2C_MAP {\
>     {RG_KEY_UP,     2, 1},\
>     {RG_KEY_RIGHT,  1, 1},\
>     {RG_KEY_DOWN,   3, 1},\
>     {RG_KEY_LEFT,   0, 1},\
>     {RG_KEY_SELECT, 6, 1},\
>     {RG_KEY_START,  7, 1},\
>     {RG_KEY_MENU,   8, 1},\
>     {RG_KEY_OPTION, 9, 1},\
>     {RG_KEY_A,      4, 1},\
>     {RG_KEY_B,      5, 1},\
> }

Hopefully thats correct. I did originally use 0 instead of 1 but went straight to factory menu due to held buttons, I guess its inversion?

It does compile, although I do get this warning:

> [766/920] Building C object esp-idf/retro-go/CMakeFiles/__idf_retro-go.dir/rg_i2c.c.obj
C:/Espressif/frameworks/esp-idf-v4.4.8/retro-go-dev/components/retro-go/rg_i2c.c: In function 'rg_i2c_gpio_init':
C:/Espressif/frameworks/esp-idf-v4.4.8/retro-go-dev/components/retro-go/rg_i2c.c:178:26: warning: comparison of unsigned expression < 0 is always false [-Wtype-limits]
     for (size_t i = 0; i < RG_COUNT(gpio_init_sequence); ++i)
                          ^
C:/Espressif/frameworks/esp-idf-v4.4.8/retro-go-dev/components/retro-go/rg_i2c.c: In function 'rg_i2c_gpio_deinit':
C:/Espressif/frameworks/esp-idf-v4.4.8/retro-go-dev/components/retro-go/rg_i2c.c:196:30: warning: comparison of unsigned expression < 0 is always false [-Wtype-limits]
         for (size_t i = 0; i < RG_COUNT(gpio_deinit_sequence); ++i)

I plugged in my MCP controller, it boots to the menu fine but actually works quite similar to the AI version e.g. no buttons registered and has phantom inputs

Heres the serial log:

> [17:27:33][info] rg_i2c_init: I2C driver ready (SDA:22 SCL:27).
[17:27:33][0m[info] rg_i2c_gpio_init: GPIO Extender ready (driver:3, addr:0x20).
[17:27:33][0m[info] rg_storage_init: Looking for SD Card using SDSPI...
[17:27:33][0m[info] rg_storage_init: Storage mounted at /sd.
[17:27:33][0m[info] rg_input_init: Initializing I2C gamepad driver...
[17:27:33][0m[info] rg_input_init: Input ready. state=00000000 00000000
[17:27:33][0m[info] json_root: Config file loaded: '/sd/retro-go/config/global.json'
[17:27:33][0m[31m[error] rg_storage_read_file: Fopen failed (2): '/sd/retro-go/config/boot.json'
[17:27:33][0m[info] rg_display_init: Initialization...
[17:27:33][0m[info] json_root: Config file loaded: '/sd/retro-go/config/launcher.json'
[17:27:33][0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3ffc6a60
[17:27:33][0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3ffc7464
[17:27:33][0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3ffc7e68
[17:27:33][0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3ffc886c
[17:27:33][0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3ffc9270
[17:27:33][0m[info] lcd_set_backlight: backlight set to 80%
[17:27:33][0m[info] rg_display_init: Display ready.
[17:27:33][0m[info] rg_alloc: SIZE=11520, CAPS=SPIRAM|8BIT, PTR=0x3f8061a8
[17:27:33][0m[info] rg_gui_set_language_id: Language set to: English (0)
[17:27:33][0m[info] rg_gui_set_font: Font set to: points=12, scaling=1.00
[17:27:33][0m[info] rg_gui_set_theme: Theme set to 'Art Book Next'!
[17:27:33][0m[info] rg_audio_init: Audio ready. sink='Speaker', samplerate=32000, volume=100
[17:27:33][0m[31m[error] rg_storage_read_file: Fopen failed (2): '/sd/retro-go/cache/clock.bin'
[17:27:33][0m[info] rg_system_load_time: Time is now: Wed Jun 25 11:25:40 2025
[17:27:33]
[17:27:33][0m[info] rg_system_init: Available memory: 225/318 + 4060/4093
[17:27:33][0m[info] rg_system_init: Retro-Go ready.
[17:27:33]
[17:27:33][0m[info] rg_alloc: SIZE=153632, CAPS=SPIRAM|8BIT, PTR=0x3f808eac
[17:27:33][0m[info] gui_add_tab: Tab 'nes' added at index 0
[17:27:33][0m[info] gui_add_tab: Tab 'snes' added at index 1
[17:27:33][0m[info] gui_add_tab: Tab 'gb' added at index 2
[17:27:33][0m[info] gui_add_tab: Tab 'gbc' added at index 3
[17:27:33][0m[info] gui_add_tab: Tab 'gw' added at index 4
[17:27:33][0m[info] gui_add_tab: Tab 'sms' added at index 5
[17:27:33][0m[info] gui_add_tab: Tab 'gg' added at index 6
[17:27:33][0m[info] application: Application 'Sega Mega Drive' (gwenesis) not present, skipping
[17:27:33][0m[info] gui_add_tab: Tab 'col' added at index 7
[17:27:33][0m[info] gui_add_tab: Tab 'pce' added at index 8
[17:27:33][0m[info] gui_add_tab: Tab 'lnx' added at index 9
[17:27:33][0m[info] application: Application 'DOOM' (prboom-go) not present, skipping
[17:27:33][0m[info] application: Application 'MSX' (fmsx) not present, skipping
[17:27:33][0m[info] crc_cache_init: Loaded CRC cache (entries: 15)
[17:27:33][0m[info] gui_add_tab: Tab 'favorite' added at index 10
[17:27:33][0m[info] gui_add_tab: Tab 'recent' added at index 11
[17:27:33][0m[31m[error] rg_storage_read_file: Fopen failed (2): '/sd/retro-go/config/wifi.json'
[17:27:33][0m[info] webui_start: Web server started

I dont think its an issue with my controller board as I literally plugged in the PCF8575 board and it worked fine. I did also try the PCF8575 board with the new dev code using both MCP and PCF9539 set but that didnt work, goes to  [... cut ...]
I dont know if it helps but heres the AI code for 1.44 that works with the PCF8575: https://github.com/DynaMight1124/retro-go/tree/CYD-PCF8575 been working pretty flawless actually, however does break pretty much every other target!! 

Hopefully I havent rambled too much useless information, more than happy to test anything you need. I'm by no means an expert but when trying to get the MCP to work, it felt like AI was struggling with the 8 pins per bus and bank switching, whereas the PCF8575 had all 16 pins on a single bus. 

Thanks

ducalex, 2025-06-25:
Thanks for testing! I think I will add PCF8575 support but first I'd really like to get the currently "supported" devices working.

One thing you could try is add the following code in `rg_input_read_gamepad_raw`:

```c
    for (size_t addr = 0x20; addr < 0x28; ++addr)
    {
        uint8_t a0 = rg_i2c_read_byte(addr, 0x12); // GPIOA BANK = 0
        uint8_t b0 = rg_i2c_read_byte(addr, 0x13); // GPIOB BANK = 0
        uint8_t a1 = rg_i2c_read_byte(addr, 0x09); // GPIOA BANK = 1
        uint8_t b1 = rg_i2c_read_byte(addr, 0x19); // GPIOB BANK = 1
        if (a0 == 0xFF && b0 == 0xFF && a1 == 0xFF && b1 == 0xFF)
            continue;
        if (a0 == 0x00 && b0 == 0x00 && a1 == 0x00 && b1 == 0x00)
            continue;
        RG_LOGI("addr %02X:    a0: %02X  b0: %02X  a1: %02X  b1: %02X", addr, a0, b0, a1, b1);
    }
```
It will scan all addresses and possible registers for the MCP23017. You can try to see if it shows anything when you press buttons.

> Hopefully thats correct. I did originally use 0 instead of 1 but went straight to factory menu due to held buttons, I guess its inversion?

In your code [the values from the PCF8575 are inverted](https://github.com/DynaMight1124/retro-go/blob/CYD-PCF8575/components/retro-go/rg_input.c#L132C14-L132C75) which makes me think that you do pull buttons to ground when they're pressed? So 0 would have been the right value indeed. But if you try my snippet above you should leave it at 1 for now so that it can boot...

DynaMight1124, 2025-06-25:
Hi, I've added this code. I compiled with both 0 & 1

0 (    {RG_KEY_UP,     2, 0},\) outputted:
> [21:58:45][info] rg_i2c_init: I2C driver ready (SDA:22 SCL:27).
[21:58:45][0m[info] rg_i2c_gpio_init: GPIO Extender ready (driver:3, addr:0x20).
[21:58:45][0m[info] rg_storage_init: Looking for SD Card using SDSPI...
[21:58:45][0m[info] rg_storage_init: Storage mounted at /sd.
[21:58:45][0m[info] rg_input_init: Initializing I2C gamepad driver...
[21:58:45][0m[info] rg_input_read_gamepad_raw: GPIOA: 00  GPIOB: 02  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:45][0m[info] rg_input_read_gamepad_raw: GPIOA: 80  GPIOB: 02  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:45][0m[info] rg_input_read_gamepad_raw: GPIOA: 00  GPIOB: 00  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:45][0m[info] rg_input_read_gamepad_raw: GPIOA: 91  GPIOB: 03  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:45][0m[info] rg_input_read_gamepad_raw: GPIOA: 00  GPIOB: 00  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:45][0m[info] rg_input_read_gamepad_raw: GPIOA: 91  GPIOB: 03  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:45][0m[info] rg_input_read_gamepad_raw: GPIOA: 00  GPIOB: 00  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:45][0m[info] rg_input_read_gamepad_raw: GPIOA: FB  GPIOB: 03  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:45][0m[info] rg_input_read_gamepad_raw: GPIOA: 00  GPIOB: 00  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:45][0m[info] rg_input_read_gamepad_raw: GPIOA: B1  GPIOB: 03  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:45][0m[info] rg_input_read_gamepad_raw: GPIOA: 81  GPIOB: 00  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:45][0m[info] rg_input_read_gamepad_raw: GPIOA: 10  GPIOB: 03  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:45][0m[info] rg_input_read_gamepad_raw: GPIOA: F1  GPIOB: 00  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:45][0m[info] rg_input_read_gamepad_raw: GPIOA: 00  GPIOB: 03  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:45][0m[info] rg_input_read_gamepad_raw: GPIOA: FF  GPIOB: 00  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:45][0m[info] rg_input_read_gamepad_raw: GPIOA: 00  GPIOB: 00  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:45][0m[info] rg_input_init: Input ready. state=00000011 11111111
[21:58:45][0m[info] rg_input_read_gamepad_raw: GPIOA: 01  GPIOB: 00  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:45][0m[33m[warn] rg_system_init: Button 00000011 11111111 being held down...
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: F9  GPIOB: 03  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: 00  GPIOB: 03  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: FF  GPIOB: 03  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: 00  GPIOB: 00  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: FF  GPIOB: 03  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: 00  GPIOB: 00  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: FF  GPIOB: 03  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: 01  GPIOB: 03  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: FF  GPIOB: 03  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: 01  GPIOB: 00  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[33m[warn] rg_system_init: Button 00000011 11111111 being held down...
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: 11  GPIOB: 03  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: 01  GPIOB: 03  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: 85  GPIOB: 00  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: 01  GPIOB: 00  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: 81  GPIOB: 02  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: 01  GPIOB: 00  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: 85  GPIOB: 02  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: 01  GPIOB: 00  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: 85  GPIOB: 02  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: 01  GPIOB: 00  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[33m[warn] rg_system_init: Button 00000011 11110111 being held down...
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: 05  GPIOB: 03  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: 01  GPIOB: 02  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[21:58:46][0m[info] rg_input_read_gamepad_raw: GPIOA: FF  GPIOB: 01  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[... cut ...]
[22:02:51][0m[info] rg_input_read_gamepad_raw: GPIOA: 9D  GPIOB: 03  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[22:02:51][0m[info] rg_display_init: Initialization...
[22:02:51][0m[info] rg_input_read_gamepad_raw: GPIOA: 9D  GPIOB: 03  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00
[22:02:51][0m[info] json_root: Config file loaded: '/sd/retro-go/config/launcher.json'
[22:02:51][0m[info] rg_input_read_gamepad_raw: GPIOA: 00  GPIOB: 00  IODIRA: FF  IODIRB: FF  IOPOLA: 00  IOPOLB: 00

Hopefully that makes sense to you :)

DynaMight1124, 2025-06-26:
Hi, in both examples I wasnt pressing any buttons at that stage, I did press some buttons later on and looked like something was registering but the serial log was scrolling so quickly it was hard to tell for sure and then pin point it afterwards. I can re-do with lots of key presses if that helps?

The actual board I am using is here: https://www.aliexpress.com/item/1005008592833333.html in its current setup its essentially a breadboard style controller with buttons tied directly to the pins and a common ground (when pressed) on a perf board. Its the same board and setup that does work on the Galagino. When I got the PCF8575 I basically unplugged a Dupont at a time and plugged into the new IO expander and it worked first time.

ducalex, 2025-06-26:
I'm not sure why it works with the PCF8575 but I'm now convinced that your problem with the MCP23017 is a lack of pull-up resistors. [This page](https://learn.adafruit.com/adafruit-pcf8575/overview) suggests that the PCF8575 has weak pull-ups built-in but I can't confirm that with the datasheet...

Anyway the good news is that the MCP23017 also has internal pull-ups that can be enabled. I'll add proper support for configurable pull-ups to rg_i2c_gpio later but in the mean time you can try adding this code in `rg_i2c_gpio_init` before the `return true;`:

```c
    rg_i2c_write_byte(gpio_address, 0x0C, 0xFF);
    rg_i2c_write_byte(gpio_address, 0x0D, 0xFF);
```

Don't forget to set your values back to 0 in your `config.h` map before testing.

ducalex, 2025-06-27:
> Let me know if you want me to try the latest dev with the two lines from above and test the results.

If you have time yes I'd appreciate if you could try the latest dev commit so that we can close this issue!

The changes from this thread are no longer needed but I haven't added the pullup configuration to `RG_GAMEPAD_I2C_MAP` map yet. So after checking out the latest commit you will need this one change:

- In `rg_input.c` find `RG_GPIO_INPUT` and change it to `RG_GPIO_INPUT_PULLUP`.

ducalex, 2025-06-27:
I've made a quick and dirty test to support PCF8575 based on your code and the datasheet. Can you try it with your CYD? I'd like to know if I understood it properly.

It's in a temp branch: https://github.com/ducalex/retro-go/commits/PCF8575/

You will need to update your config.h:

```c
#define RG_I2C_GPIO_DRIVER          4
#define RG_I2C_GPIO_ADDR            0x20

...

#define RG_GAMEPAD_I2C_MAP {\
    {RG_KEY_UP,     2, 0},\
    {RG_KEY_RIGHT,  1, 0},\
    {RG_KEY_DOWN,   3, 0},\
    {RG_KEY_LEFT,   0, 0},\
    {RG_KEY_SELECT, 6, 0},\
    {RG_KEY_A,      4, 0},\
    {RG_KEY_B,      5, 0},\
    {RG_KEY_START,  7, 0},\
    {RG_KEY_MENU,   8, 0},\
    {RG_KEY_OPTION, 9, 0},\
}
```

DynaMight1124, 2025-06-28:
I've tested the PCF8575 branch with your suggested config.h settings and everything looks to be perfect. Tested for about 10mins and no issues so far

Serial looks fine and no i2c related errors and no compiling related errors/messages either.

> [08:54:29][info] rg_i2c_init: I2C driver ready (SDA:22 SCL:27).
[08:54:29][0m[info] rg_i2c_gpio_init: GPIO Extender ready (driver:4, addr:0x20).
[08:54:29][0m[info] rg_storage_init: Looking for SD Card using SDSPI...
[08:54:29][0m[info] rg_storage_init: Storage mounted at /sd.
[08:54:29][0m[info] rg_input_init: Initializing I2C gamepad driver...
[08:54:29][0m[info] rg_input_init: Input ready. state=00000000 00000000

DynaMight1124, 2025-06-30:
Hi

Thank you, it does make sense and will certainly help people adapt their system setups easily.

As for my use case, I think when I searched I couldnt locate any breakout boards with the two i2c controllers originally supported so I just wanted something that worked with the i2c breakout I had to hand (initially the MCP) and had hoped AI could work its magic but couldnt hence the original post :) my original plan wasnt to add the CYD as a device or merge any terrible AI code, just publish a FW that worked with my project. 

I'm not sure if I've actually shared this here yet but this is the CYD project: https://www.instructables.com/Retro-Handheld-Based-on-the-ESP32-CYD-and-RetroGo/ It does work out fairly cheap if you can get the CYD on an Aliexpress bundle deal, can be had for like £5/$6 then the rest of the components are not too expensive. No-one else may ever build one but I had fun learning and putting it together :)

ducalex, 2025-07-10:
I've merged support for PCF8575 (RG_I2C_GPIO_DRIVER 4, in d57d455e1dec49730208ff0edca173b8c368cca7). If you have time you can try it but otherwise I'll close this issue and assume it works until someone says otherwise :) .

> No-one else may ever build one but I had fun learning and putting it together :)

It's fantastic how our hobby has become so cheap and accessible! I'm curious how hard it would be to add touch support to the CYD. Not that it would be very fun to play that way...

DynaMight1124, 2025-07-10:
I've flashed the latest dev build, the controls dont work and I get the following in serial:

> [22:21:25][info] rg_i2c_init: I2C driver ready (SDA:22 SCL:27).
[22:21:25][0m[31m[error] rg_i2c_gpio_init: Failed to initialize extender (driver:4, addr:0x20).
[22:21:25][0m[info] rg_storage_init: Looking for SD Card using SDSPI...
[22:21:25][0m[info] rg_storage_init: Storage mounted at /sd.
[22:21:25][0m[info] rg_input_init: Initializing I2C gamepad driver...
[22:21:25][0m[33m[warn] rg_i2c_gpio_configure_port: PCF8575 mode cannot be set by pin. (mask is 0x04, expected 0xFF)
[22:21:25][0m[info] rg_input_init: Input ready. state=00000000 00000000

I used the most recent i2c map you suggested, not sure if I did something wrong.

As for the CYD touchscreen, I am not sure how useful it would be for gaming, it uses resistive similar to old phones from like 15-20+ years ago! there are CYD style devices that have capacitive touchscreens which would be better from proper touch controls. Yeah no fun to play, always hated emulators etc on my phone but could potentially have other uses in the menu system.

ducalex, 2025-07-10:
> I used the most recent i2c map you suggested, not sure if I did something wrong.

No I think my code is wrong, I overlooked something. Can you try replacing this line in rg_i2c.c:

`return PCF8575_mode == RG_GPIO_OUTPUT && rg_i2c_write(gpio_address, -1, &gpio_output_values, 2);`

with 

`return PCF8575_mode != RG_GPIO_OUTPUT || rg_i2c_write(gpio_address, -1, &gpio_output_values, 2);`

It doesn't make logical sense but it makes the behavior consistent with the other extenders.

DynaMight1124, 2025-07-10:
I've tried that code, I get the same serial log:

> [23:15:45][info] rg_i2c_init: I2C driver ready (SDA:22 SCL:27).
[23:15:45][0m[info] rg_i2c_gpio_init: GPIO Extender ready (driver:4, addr:0x20).
[23:15:45][0m[info] rg_storage_init: Looking for SD Card using SDSPI...
[23:15:45][0m[info] rg_storage_init: Storage mounted at /sd.
[23:15:45][0m[info] rg_input_init: Initializing I2C gamepad driver...
[23:15:45][0m[33m[warn] rg_i2c_gpio_configure_port: PCF8575 mode cannot be set by pin. (mask is 0x04, expected 0xFF)
[23:15:45][0m[info] rg_input_init: Input ready. state=00000000 00000000

Weirdly the controls do work somethings, maybe 50% of boots but only for a tiny bit and also ended up crashing something as the theme disappeared (I think the SD card vanished until I restarted) It's possible it was doing the same before too, just so happened it wasnt working when I tried. Both times the serial disconnected fairly soon after this so dont have any logs.

I'm off now, got an early start but happy to try anything you need tomorrow, just let me know

DynaMight1124, 2025-07-11:
I'm really sorry again, looks like it was an error at my end, when I took apart the unit, it looks like I dislodged the poor shrinkwrap job I did on the speaker cables and they were touching, this looked to be had been causing the crashes in the serial log and also the intermittently working controls!!

I have reverted back to the original dev code:

> return PCF8575_mode == RG_GPIO_OUTPUT && rg_i2c_write(gpio_address, -1, &gpio_output_values, 2);

Compiled and its currently working fine. I still get this in the serial log, not sure if that should be there or not?  

> [14:25:00][0m[33m[warn] rg_i2c_gpio_configure_port: PCF8575 mode cannot be set by pin. (mask is 0x04, expected 0xFF)

I'm sorry to waste your time again, I'm sure you've got better things to do :) I'll keep testing it the rest of the afternoon to make sure its working as before but its looking promising so far.

DynaMight1124, 2025-07-11:
Been testing for a few hours now and its working fine on the latest dev. Happy to supply a more extensive serial log if needed or let me know if you need anything else tested? but apart from the warning in the log, everything is working perfectly.

ducalex, 2025-07-11:
If all works correctly please close the issue, since we've successfully added support for MCP23017, PCF9539, and PCF8575 :)

DynaMight1124, 2025-07-11:
Thank you, I'll pull the latest change but also change to .pullup = 0 for good measure. I'll compile and test later/tomorrow and close when tested but I reckon its a good one!

Thank you for all your help. I'm sure you dont want another device added so I'll keep the CYD as a fork.