ducalex/retro-go #208: [Feature Request] Full screen artwork

DynaMight1124, 2025-07-07:
Hi

Not sure if this is possible or exactly how it could look on a smaller screen but what about rather than a boxart picture displayed over the top of the text, there could be some artwork with a gradient displayed behind the text.

It's pretty easy with tools like Skraper to get custom mixes with all sorts of artwork, screenshots, logos, even boxart as well but would need the launcher to display it properly. 

In my head I'm thinking something similar to this:

[image]

Theres some more pictures here: https://github.com/ebzero/garlic-onion-skraper-mix?tab=readme-ov-file#full-screen-mix

I dont know if there would be performance issues? backgrounds would probably need to be 320x240 at worse but obviously could be compressed etc however we could potentially half that if there was a stronger gradient to allow decent readability of the game list text, so maybe 160x240 for half the screen width wise and full length. 

If its not gonna look great on the small screen or take way too much work to implement then its totally fine, the current setup of the boxart etc is more than good enough imo. Just a thought.

ducalex, 2025-07-19:
The entire screen is redrawn on updates already (everything pixel is usually redrawn, it's just not always blitted) so I don't think that would change much in that regard.  But load times from the sd card (+ decompression + possibly resampling) might be an issue.

It might be worth generating a small batch of such images (possibly post-processed by pngquant) to see what kind of sizes we're looking at, and then perform some performance testing.

DynaMight1124, 2025-07-19:
I've quickly drawn up something rough, the actual gradient size can be adjusted as well as the whats shown. Currently its screenshot plus logo, but could also have boxart either replacing the logo or boxart at the bottom, it'll be one single image whatever it shown. With the limited screen size, probably dont want it too busy! 

I've processed with PNGQuant and by CRC for Atari Lynx since it has a fairly small library, I guess CRC would give the worst case scenario, my personal setup is matched via name and its lots quicker. If CRC is too slow but name match is ok, you could force caching maybe.

Heres some examples:

[image]

[image]

Logo size can be moved or made smaller etc. 

Heres the full collection zipped: 

[lnx romart.zip](https://github.com/user-attachments/files/21331901/lnx.romart.zip)

Let me know if you need anything else.

ducalex, 2025-07-27:
I've made a quick branch to try it out and it's definitely promising but there's a fair amount of tweaking to be done to get the full effect:
- replace our current top banner with a smaller icon like in your first post
- possibly update the themes for a lighter font color in the list
- decide how to render the old art (should it now be behind the text too)
- your images might need to be darkened a bit wider from the right
- decide if we show save preview (when enabled) in addition of the background, or if we show it as the background

But in any case, you can try it if you'd like: https://github.com/ducalex/retro-go/commits/background-previews/

I'd say performance is definitely not an issue with the small sizes you've achieved, as long as the CRC cache is prebuilt. Very promising imho!