ducalex/retro-go #209: fatal error: rom/miniz.h: No such file or directory   511 | #include <rom/miniz.h>

alkopop79, 2025-07-10:
When trying to build the image with `python rg_tool.py build-fw`, I get the following error:

```
fatal error: rom/miniz.h: No such file or directory
  511 | #include <rom/miniz.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
```

Reproducing the error:

ESP-IDF v5.4.2 on Mac OS X 15.0.1 - once I start ESP IDF in the terminal, navigate to the retro-go folder downloaded from GitHub and run `python rg_tool.py build-fw` or `python3 rg_tool.py build-fw`

Expected behavior: successful compilation, instead of the error

Any idea what could have gone wrong?

alkopop79, 2025-07-10:
Thank you! Do I just download the latest version of the software?

alkopop79, 2025-07-10:
I downloaded the dev version, and I still have the same issue. What am I doing wrong?

ducalex, 2025-07-10:
I'm not sure, I will have to install 5.4 and do some testing...

In the mean time you can try disabling zip support, it will remove the need for miniz. In `components/retro-go/config.h` find `RG_ZIP_SUPPORT 1` and replace with `RG_ZIP_SUPPORT 0`.

DynaMight1124, 2025-07-11:
I cant speak from a lot of experience but I know when I first tried to compile RetroGo, I installed the latest ESP-IDF, as you would generally do and got loads of errors trying to compile, get an error, google it, fix it, compile, get a further error. I know from other projects like VS Code, Picos etc its usually quite important to use the same version as it was originally developed for. Arduino esp is a nightmare for this.  

I removed the newer versions and installed 4.4.8, followed the sd fix https://github.com/ducalex/retro-go/blob/master/BUILDING.md and it compiled perfectly first time. I dont know if there are any benefits of compiling with newer versions? Sometimes, it can actually make things slower but thats only from what I've read on some Arduino projects I've kept tabs on.

alkopop79, 2025-07-11:
> I'm not sure, I will have to install 5.4 and do some testing...
> 
> In the mean time you can try disabling zip support, it will remove the need for miniz. In `components/retro-go/config.h` find `RG_ZIP_SUPPORT 1` and replace with `RG_ZIP_SUPPORT 0`.

Thank you for the reply; I will try that! I appreciate the effort

alkopop79, 2025-07-21:
Tried installing esp-idf-v4.4.8, and it didn't work. I give up and will use an RPI instead of the ESP32. Thanks for the help anyway

`
Traceback (most recent call last):
  File "/Users/greglorincz/esp-idf-v4.4.8/tools/idf_tools.py", line 1587, in action_install_python_env
    import virtualenv  # noqa: F401
    ^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'virtualenv'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/greglorincz/esp-idf-v4.4.8/tools/idf_tools.py", line 1996, in <module>
    main(sys.argv[1:])
    ~~~~^^^^^^^^^^^^^^
  File "/Users/greglorincz/esp-idf-v4.4.8/tools/idf_tools.py", line 1992, in main
    action_func(args)
    ~~~~~~~~~~~^^^^^^
  File "/Users/greglorincz/esp-idf-v4.4.8/tools/idf_tools.py", line 1590, in action_install_python_env
    subprocess.check_call([sys.executable, '-m', 'pip', 'install', '--user', 'virtualenv'],
    ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                          stdout=sys.stdout, stderr=sys.stderr)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.13/3.13.5/Frameworks/Python.framework/Versions/3.13/lib/python3.13/subprocess.py", line 419, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/opt/homebrew/opt/python@3.13/bin/python3.13', '-m', 'pip', 'install', '--user', 'virtualenv']' returned non-zero exit status 1.`