ducalex/retro-go #211: Retro-go on the ESP32-P4

rapha-tech, 2025-07-18:
After a day of trouble I finally got full retro-go working on the ESP32-P4 so I decided to share my codebase for others.
First of all, you might be wondering "are the performances better?" and yes they are ! I noticed great improvements from the ESP32 (S3):
- GENESIS runs at full speed 60fps with all the chips enabled (with the games I tested)
- with audio disabled SNES runs at full speed 55/60fps with frameskip at "2" and audio disabled (with the games I tested)
- with audio enabled SNES still struggle and will runs at 80% speed 40fps  with frameskip at "2" but the ESP is busy at 80% so weird (with the games I tested)
- GBA: for example MarioKart SuperCircuit runs at about 75% speed with frameskip at "5" (10fps) pokemon emerald will runs at full speed with frameskip "5" (10fps) most of the time (in the city for example) and busy at 90%. We might be able to get down to frameskip "4" at 400mhz. So almost playable.

Keep in mind this was tested at 360mhz, performance should be a little bit better when 400mhz chips will be available.
Also these are just some vague aproximations based on a few tests, I would recommend you to test your own games to get a better idea of the exact performance to expect.

Anyway, if you are interested, here is the branch for the P4 on my fork (it was built using ESP-IDF v5.5-rc1):
https://github.com/rapha-tech/retro-go/tree/ESP32-P4
Instructions for building are in the README inside the target's folder:
https://github.com/rapha-tech/retro-go/tree/ESP32-P4/components/retro-go/targets/esp32-p4

Here is a quick demo video on GBA:
https://youtu.be/vBlkORBRjsY
and SNES:
https://youtu.be/XWoIEtMfELU

I also decided to merge the GBA emulator for easier benchmarking/testing.

This fork is just a quick draft, lot of code for regular devices have been removed but I will probably make a PR later wich clean #ifdef.

DynaMight1124, 2025-07-18:
Thats awesome, I was wondering the other day if the P4 could be compiled without too much work. Looks very promising!

rapha-tech, 2025-07-19:
Thanks! the other day I saw a guy running the SNES emulator from retro-go (with some tweaks) on the P4 and it seems to run perfectly.
https://www.youtube.com/watch?v=ojtIx0t9SMg
At least from my testing it runs better than my simple adaptation of the original code.
So maybe some improvements could be made in retro-go to achieve the same level of performance as the guy.

DynaMight1124, 2025-07-19:
Yeah watched that video a while back and it really does show the potential of the P4, would be amazing to see some almost full speed SNES and GBA with RetroGo. I was a Sega guy growing up so full speed MegaDrive is amazing!

ducalex, 2025-07-22:
Assuming the youtube guy didn't modify my snes9x port too much and you use similar sdkconfig, then it's safe to assume that libretro-go is probably our bottleneck here. Presumably rg_display and/or rg_audio stuff being too wasteful? [His approach to scaling/blitting](https://github.com/fcipaq/snes9x_esp32) is definitely much much faster than whatever I do in rg_display but does it fully explain the difference? 

Of the top of my head you can try disabling filtering, scaling, and setting `RG_SCREEN_PARTIAL_UPDATES 0` in your config.h.

Maybe we should consider adding a faster rendering path for high performance apps that would support only integer scaling, no filtering, and no partial updates.

Very interesting possibilities with the P4!

DynaMight1124, 2025-07-23:
> Maybe we should consider adding a faster rendering path for high performance apps that would support only integer scaling, no filtering, and no partial updates.

Thats interesting, I'm assuming this would also benefit the OG ESP32 and S3 with MegaDrive and SNES? It feels like stripping out some of the 'nice to have' stuff to benefit performance wouldnt be a bad thing! imo, even if sound was disabled by default to get closer to playable would also be a suitable compromise, obviously imo.

From my testing with MD, its a little inconsistent and I'm not exactly sure what happens behind the scenes e.g. my usual test is Street of Rage 2, load up a game, plays fairly slow probably 60-70% speed (as shown by the pause menu), go into the menu, disable the sound and sometimes it'll then play fairly close to full speed with some frameskip but other times it'll actually play too fast like 130-150% speed. If I disable sound completely and load up a game, there wont be any sound but it'll still play at 60-70% speed. Sorry, not trying to hijack the post and happy to open a new issue if you want, didnt wanna bombard you with issues, esp for a system that may never be perfect on the current hardware however the P4 may change that!

rapha-tech, 2025-07-23:
Yes the P4 is so much capable they even managed to run quake 1 on it!

For SNES emulation I tried disabling filtering and scaling and it didn't changed a thing. So I believe libretro-go isn't our bottleneck here.

Also, I benchmarked Mariokart while he tried Mario world which is easier to runs hence the (very) big difference in performance.

With mario world:
BUSY:64%, FPS:50 (25+25+0), BATT:0
BUSY:64%, FPS:50 (25+18+6), BATT:0
BUSY:63%, FPS:50 (24+16+10), BATT:0
BUSY:63%, FPS:50 (26+3+22), BATT:0
BUSY:62%, FPS:50 (25+0+25), BATT:0
I get a reasonable frameskip of "1" but as you can see, we're only using 60/65% of the power : but still too much to disable frameskip completly.

Additionnaly, after tinkering I found out what were the 2 main differences:
- He didn't used "fixed" frameskip, instead he just skip some frames there and there allowing the game to runs at a float ish frameskip. And this is why when running games I was only using 60-80% of the power of the P4 with a fixed frameskip of 1 or 2.
- He used the 2 cores just like retro-go but did it a little differently. Basically, instead of running the whole emulator on one core he offloaded the `S9xMixSamples()` function to the "I/O core" taht's why he can achieve this level of performance with sound enabled!

After seeing that I tried implementing it in `main_snes.c` 
I also disabled `RG_SCREEN_PARTIAL_UPDATES` as you said.

I think the numbers speak for themselves...
Mario world (Europe) with sound enabled on core 1:

BUSY:100%, FPS:50 (6+0+43)   starting the game
BUSY:100%, FPS:49 (8+0+41)
BUSY:100%, FPS:50 (5+0+46)  Intro
BUSY:100%, FPS:49 (5+0+45) 
BUSY:100%, FPS:50 (3+0+46) 
BUSY:100%, FPS:50 (3+0+47) 
BUSY:100%, FPS:59 (0+0+59) 
BUSY:100%, FPS:52 (3+0+49)  Hit B to start
BUSY:100%, FPS:53 (0+0+53)
BUSY:100%, FPS:54 (0+0+54) 
BUSY:100%, FPS:54 (0+0+54) First text box / tutorial
BUSY:100%, FPS:54 (0+0+55)  
BUSY:100%, FPS:54 (0+0+53)  
BUSY:100%, FPS:54 (0+0+54) 
BUSY:100%, FPS:55 (0+0+55) 
BUSY:100%, FPS:55 (0+0+55) 
BUSY:100%, FPS:54 (0+0+54)
BUSY:100%, FPS:55 (0+0+55) 
BUSY:100%, FPS:56 (0+0+56) 
BUSY:100%, FPS:59 (0+0+59) 
BUSY:100%, FPS:59 (0+0+59) 
BUSY:100%, FPS:56 (1+0+55)
BUSY:100%, FPS:50 (6+0+44) 
BUSY:100%, FPS:50 (7+0+43)  Choose level
BUSY:100%, FPS:50 (7+0+43) 
BUSY:100%, FPS:54 (4+0+50) 
BUSY:100%, FPS:60 (0+0+60) 
BUSY:100%, FPS:54 (3+0+51)  Level starts
BUSY:100%, FPS:50 (2+0+49) 
BUSY:100%, FPS:50 (2+0+47)
BUSY:100%, FPS:50 (3+0+47) 
BUSY:100%, FPS:50 (2+0+48) 
BUSY:100%, FPS:50 (3+0+47) 
BUSY:100%, FPS:50 (5+0+46) 
BUSY:100%, FPS:50 (5+0+44) 
BUSY:100%, FPS:50 (4+0+46) 
BUSY:100%, FPS:50 (2+0+48) 
BUSY:100%, FPS:49 (1+0+48) 
BUSY:100%, FPS:51 (2+0+49) 
BUSY:100%, FPS:50 (0+0+51)
BUSY:100%, FPS:50 (0+0+49) 
BUSY:100%, FPS:50 (1+0+49)
BUSY:100%, FPS:50 (2+0+49)
BUSY:100%, FPS:50 (3+0+47)
BUSY:100%, FPS:50 (2+0+48)
BUSY:100%, FPS:50 (3+0+46)
[... 501 lines cut ...]
            rg_display_submit(currentUpdate, 0);
        }

        // Signal audio task to process samples using mutex
        if (rg_mutex_take(audio_mutex, 5)) // 5ms timeout
        {
            audio_ready = true;
            rg_mutex_give(audio_mutex);
        }

        rg_system_tick(rg_system_timer() - startTime);

        if (skipFrames > 0)
        {
            skipFrames--;
        }
    }

    // Cleanup on exit
    audio_shutdown = true;
    if (audio_mutex) {
        rg_mutex_free(audio_mutex);
    }
}
```

rapha-tech, 2025-07-23:
> Thats interesting, I'm assuming this would also benefit the OG ESP32 and S3 with MegaDrive and SNES? It feels like stripping out some of the 'nice to have' stuff to benefit performance wouldnt be a [...]

Hi dynamight,
From what I know on SNES, MD and basically all apps except PCE and DOOM the whole system emulation is run on core 0 while core 1 handle I/O like display (scaling, filtering, SPI...), input buttons, DAC, rg_system monitor...
On SNES and MD our main bottleneck is core 0 which runs the emulator so disabling core 1 functions like scaling probably won't help with performance.

What might help though would be to offload some of the chip emulation to the I/O core (1) (just like the youtube guy did with sound emulation on SNES) but I don't know if it is possible with MD. I haven't seen anyone done it. 
Plus I have no clue about what "busy %" we're at on the I/O core currently.

I'm not 100% sure about all that so @ducalex please correct me if I'm wrong.

DynaMight1124, 2025-07-23:
Hi

Yeah that makes sense now, I wasnt sure how it worked previously, I know from previous ESP32 projects I've seen such as the RetroTV's it was very important to maximise both cores for smooth operation so would totally make sense with this too and sounds like core 1 probably isnt doing as much as poor old core 0 hah

I'd guess the obvious would be sound being offloaded to core 1 as well as keeping the other stuff, I'm deffo no coder so no idea how tricky that would be! 

The improvements to the SNES speed also look impressive!

ducalex, 2025-07-27:
I've tried many times to get sound to run on the second core but I always end up with synchronization issues and the games would crash. I also tried doing the graphics rendering on the second core which has the same issues. Eventually I gave up. 

But I'll try your code, maybe the AI is smarter :) and also I'll look more closely at what fcipaq did. It's definitely an improvement worth merging if it works.

> For SNES emulation I tried disabling filtering and scaling and it didn't changed a thing. So I believe libretro-go isn't our bottleneck here.

I've done some tests (with partial update disabled to get a more stable reading) in SNES.

Scaling and filtering enabled: Blit time is 26ms
Scaling and filtering disabled: Blit time is 21ms
Rewrote write_update to send the raw buffer to the display: 21ms

21ms to transfer ~122880 bytes, so about 44Mbps which is about what you'd expect for a 40Mhz SPI display.

So the pixel manipulation code is likely fine, that's good news to me because I spent so much time on it over the years to make it fast... 

So as long as the snes audio task has higher priority than the display task, it should be able to keep running during the display transfer and use most of the processing power of the second core.

rapha-tech, 2025-07-29:
Okay, so after tinkering even more with the AI, I came up with this code:

I added back rg_display_sync() and the audio_task has been modified, now the sound isn't choppy anymore and the screen is in sync!
I also added double buffering but I don't know if that changed a thing
Currently the only issues are these:
- When the menu is opened and closed after a few seconds, the audio task try to spit out as fast as possible all the samples
- On certain games like f-zero and zelda a link to the past there is a weird line of ~20px at the bottom of the screen

```c
#include "shared.h"

#include <snes9x.h>
#include <math.h>

typedef struct
{
	char name[16];
	struct {
		uint16_t snes9x_mask;
		uint16_t local_mask;
		uint16_t mod_mask;
	} keys[16];
} keymap_t;

static const keymap_t KEYMAPS[] = {
	{"Type A", {
		{SNES_A_MASK, RG_KEY_A, 0},
		{SNES_B_MASK, RG_KEY_B, 0},
		{SNES_X_MASK, RG_KEY_START, 0},
		{SNES_Y_MASK, RG_KEY_SELECT, 0},
		{SNES_TL_MASK, RG_KEY_B, RG_KEY_MENU},
		{SNES_TR_MASK, RG_KEY_A, RG_KEY_MENU},
		{SNES_START_MASK, RG_KEY_START, RG_KEY_MENU},
		{SNES_SELECT_MASK, RG_KEY_SELECT, RG_KEY_MENU},
		{SNES_UP_MASK, RG_KEY_UP, 0},
		{SNES_DOWN_MASK, RG_KEY_DOWN, 0},
		{SNES_LEFT_MASK, RG_KEY_LEFT, 0},
		{SNES_RIGHT_MASK, RG_KEY_RIGHT, 0},
	}},
	{"Type B", {
		{SNES_A_MASK, RG_KEY_START, 0},
		{SNES_B_MASK, RG_KEY_A, 0},
		{SNES_X_MASK, RG_KEY_SELECT, 0},
		{SNES_Y_MASK, RG_KEY_B, 0},
		{SNES_TL_MASK, RG_KEY_B, RG_KEY_MENU},
		{SNES_TR_MASK, RG_KEY_A, RG_KEY_MENU},
		{SNES_START_MASK, RG_KEY_START, RG_KEY_MENU},
		{SNES_SELECT_MASK, RG_KEY_SELECT, RG_KEY_MENU},
		{SNES_UP_MASK, RG_KEY_UP, 0},
		{SNES_DOWN_MASK, RG_KEY_DOWN, 0},
		{SNES_LEFT_MASK, RG_KEY_LEFT, 0},
		{SNES_RIGHT_MASK, RG_KEY_RIGHT, 0},
	}},
	{"Type C", {
		{SNES_A_MASK, RG_KEY_A, 0},
		{SNES_B_MASK, RG_KEY_B, 0},
		{SNES_X_MASK, 0, 0},
		{SNES_Y_MASK, 0, 0},
		{SNES_TL_MASK, 0, 0},
		{SNES_TR_MASK, 0, 0},
		{SNES_START_MASK, RG_KEY_START, 0},
		{SNES_SELECT_MASK, RG_KEY_SELECT, 0},
		{SNES_UP_MASK, RG_KEY_UP, 0},
		{SNES_DOWN_MASK, RG_KEY_DOWN, 0},
		{SNES_LEFT_MASK, RG_KEY_LEFT, 0},
		{SNES_RIGHT_MASK, RG_KEY_RIGHT, 0},
	}},
};

static const size_t KEYMAPS_COUNT = (sizeof(KEYMAPS) / sizeof(keymap_t));
[... 429 lines cut ...]
            slowFrame = !rg_display_sync(false);
            rg_display_submit(currentUpdate, 0);
            currentUpdate = (currentUpdate == updates[0]) ? updates[1] : updates[0];
            GFX.Screen = currentUpdate->data;
        }
        
        rg_system_tick(rg_system_timer() - startTime);

        if (skipFrames > 0)
        {
            skipFrames--;
        }
        else if (slowFrame || app->frameskip > 1)
        {
            skipFrames = slowFrame ? 1 : app->frameskip;
        }
    }

    // Cleanup on exit
    audio_shutdown = true;
    if (audio_mutex) {
        rg_mutex_free(audio_mutex);
    }
}
```

ducalex, 2025-08-10:
I've now merged some of your ideas to SNES, thanks for the work!

- Double buffering for display and audio are both included but disabled by default. (I'm pretty sure audio double buffering does nothing because rg_audio_submit creates a copy anyway, and we skip too many frames for double display to be worth it)

- Audio task is your code but a bit simplified. It is enabled by default because it does provide a decent speed up but we'll have to check if it breaks any games or gets out of sync too much...

I have not yet merged your frame skip code. I can confirm that with it SMW runs full speed even on base esp32 (with an obviously very very choppy display) so I will definitely be looking into either merging it, or improving the frameskip code of retro-go itself.

Let me know if my current code isn't good enough for your use case or if I broke anything :)

rapha-tech, 2025-08-11:
The current code is great! I couldn't do any better so I'm happy that this improvement could be merged.
Maybe in the future someone will find a solution to have the same performance as the free running task without the games breaking...
Still thanks for taking the time to review/upgrade the code!

DynaMight1124, 2025-08-11:
Improvements sound awesome! Are any of these possible for the MegaDrive emulator? Was always a Sega kid growing up, even got a Saturn trying to convince everyone it was better than the PSX ha :)

ducalex, 2025-08-11:
With the free running task, the main emulation frequently ends up 20-50 frames ahead of audio which is a big problem. With my current code, the main emulation can get at most 1 frame ahead.

I think ideally we should make the code allow for some wiggle room, like +/-N frames where N is configurable. This would allow us to find the sweet spot between utilising as much CPU as possible and not breaking games by running too far ahead. It should be pretty easy to do with some tweaks to rg_task, allowing for configurable message queue depth.

For the MegaDrive emulator I have tried several times running the Z80 on core 1. But I faced the same sync issues so I had to give up. It might be worth trying again.

Could also try running SN76489/ym2612 (audio chips) but not the Z80, maybe it'll cause less sync issues.

DynaMight1124, 2025-08-11:
Hi

Just had a quick test so far, only on Sonic 2 and Streets of Rage 2. It is faster, brings it almost up to levels without any sound, but still with lots of frameskip etc. Sonic is super choppy, not really playable but SOR2 is ok.

If I disable both sound chips and Z80 it actually runs too fast, about 120% speed. 

I'll need to do some more testing to make sure theres no major drawbacks. I'll test it on a few devices over the next few days and let you know.

Personally, I'd take fairly(ish!) smooth gameplay without any audio than choppy gameplay and scratchy audio :)

ducalex, 2025-08-13:
The branch I shared moves graphics rendering and audio processing to core 1.

I'm fairly confident that the crashes are caused by the rendering and not the sound. It's hard to say without a log but I'm guessing the fact that cpu emulation runs at the same time might cause bogus pointers. Or it's simply a stack overflow.

I've added menu options to control what is offloaded to core 1, you can change at any time during game play to compare. Let me know if either of you find a combo that is stable but also improves speed enough to be worth merging to dev!

DynaMight1124, 2025-08-14:
Hi

Yes rendering on core 1 was causing the graphical glitching in Outrun, disable and the glitches go away. It does bring some performance boost however, enabling rendering on core 1 does help with my two test games, Sonic 2 and SOR2. I didnt get any crashes on Sonic 2 this time either.

Also I saw you tied the game speed to the display rather than sound now, this does keep the game at a maximum of 100%. Although not an ideal scenario, but sound disabled and render on core 1, allows Sonic 2 to play at pretty much full speed (60/1 or 60/2 depending on when you check) disabling rendering on core 1 introduces some more frameskip, usually around 60/4 or 5 and doesnt feel anywhere as smooth. These were done on a original ESP32, I will do the same on a S3 but I'd imagine it will be a little quicker/smoother.

I will run some further tests but as it stands, both on core 1 is faster but rendering will cause problems with certain games.

ducalex, 2025-08-16:
> Also I saw you tied the game speed to the display rather than sound now, this does keep the game at a maximum of 100%. 

It's still tied to sound. Before, when sound was disabled, there was no speed control at all. Now when sound is disabled silence is fed to rg_audio_submit instead of never calling it. The alternative is to use busy waits but I suspect it would impact performance more than sending silence. It might be worth measuring, though.

I think I'll merge the new core 1 menu to the dev branch so we can keep experimenting.