ducalex/retro-go #217: ESP32-S3-WROOM On board SD card issues can't load sd card

kingmaxmmxweisun, 2025-08-03:
Hi my board is ESP32-S3-WROOM  Cam 16mb 8mb spram on board sd 

the config.h showed 

#define RG_GPIO_SDSPI_MISO          GPIO_NUM_40  // SD_DATA (MISO)
#define RG_GPIO_SDSPI_MOSI          GPIO_NUM_40  // SD_DATA (MOSI - 双向)
#define RG_GPIO_SDSPI_CLK           GPIO_NUM_39  // SD_CLK
#define RG_GPIO_SDSPI_CS            GPIO_NUM_38  // SD_CMD

Serial Output errors

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1494
load:0x403c9700,len:0xe60
load:0x403cc700,len:0x3060
entry 0x403c9970
I (24) boot: ESP-IDF v4.4.8-dirty 2nd stage bootloader
I (24) boot: compile time 17:18:42
I (25) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (31) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32s3: Boot SPI Speed : 80MHz
I (41) boot.esp32s3: SPI Mode       : QIO
I (46) boot.esp32s3: SPI Flash Size : 16MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (82) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  3 launcher         OTA app          00 10 00010000 000f0000
I (97) boot:  4 retro-core       OTA app          00 11 00100000 00100000
I (104) boot:  5 prboom-go        OTA app          00 12 00200000 000e0000
I (112) boot:  6 gwenesis         OTA app          00 13 002e0000 00110000
I (119) boot:  7 fmsx             OTA app          00 14 003f0000 000b0000
I (127) boot: End of partition table
I (131) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=2e53ch (189756) map
I (168) esp_image: segment 1: paddr=0003e564 vaddr=3fc97a20 size=01ab4h (  6836) load
I (170) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=9a160h (631136) map
I (270) esp_image: segment 3: paddr=000da188 vaddr=3fc994d4 size=020bch (  8380) load
I (272) esp_image: segment 4: paddr=000dc24c vaddr=40374000 size=13a14h ( 80404) load
I (298) boot: Loaded app from partition at offset 0x10000
I (299) boot: Disabling RNG early entropy source...

========================================================
launcher unknown (Aug  3 2025 17:16:37)
 built for: ESP32S3-WROOM. type: dev
========================================================

[info] rg_storage_init: Looking for SD Card using SDSPI...
E (823) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[warn] rg_storage_init: SD Card mounting failed (0x107), retrying at lower speed...
E (843) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[warn] rg_storage_init: SD Card mounting still failed (0x107), retrying at minimum speed...
E (863) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
[error] rg_storage_init: Storage mounting failed! err=0x107
[info] rg_input_init: Initializing ADC gamepad driver...
[info] rg_input_init: Initializing GPIO gamepad driver...
[info] rg_input_init: Initializing ADC battery driver...
[info] rg_input_init: Input ready. state=00000000 00000000
[error] rg_storage_read_file: Fopen failed (2): '/sd/retro-go/config/global.json'
[error] rg_storage_read_file: Fopen failed (2): '/sd/retro-go/config/boot.json'
[info] rg_display_init: Initialization...
[error] rg_storage_read_file: Fopen failed (2): '/sd/retro-go/config/launcher.json'
[info] rg_alloc: SIZE=1024, CAPS=DMA|8BIT, PTR=0x3fceb0d0
[info] rg_alloc: SIZE=1024, CAPS=DMA|8BIT, PTR=0x3fceb4d4
[info] rg_alloc: SIZE=1024, CAPS=DMA|8BIT, PTR=0x3fceb8d8
[info] rg_alloc: SIZE=1024, CAPS=DMA|8BIT, PTR=0x3fcebcdc
[info] rg_alloc: SIZE=1024, CAPS=DMA|8BIT, PTR=0x3fcec0e0
[info] lcd_set_backlight: backlight set to 80%
[info] rg_display_init: Display ready.
[info] rg_alloc: SIZE=4608, CAPS=SPIRAM|8BIT, PTR=0x3d800a44
[info] rg_gui_set_language_id: Language set to: English (0)
[info] rg_gui_set_font: Font set to: VeraBold 11 (height=13, scaling=1.00)
[info] rg_gui_set_theme: Using built-in theme!
[info] rg_audio_init: Audio ready. sink='Dummy', samplerate=32000, volume=50
[error] rg_storage_read_file: Fopen failed (2): '/sd/retro-go/cache/clock.bin'
[info] rg_system_load_time: Time is now: Sun Aug  3 02:16:24 2025

[info] rg_system_init: Available memory: 298/366 + 8184/8189
[info] rg_system_init: Retro-Go ready.

[debug] rg_gui_draw_text: Texbox (pos: 188x15, size: 62x0) will be truncated!
[debug] rg_gui_draw_text: Texbox (pos: 188x15, size: 62x0) will be truncated!
[debug] STACK:13424, HEAP:298+8184 (272+8064), BUSY:0%, FPS:0 (0+0+0), BATT:12
[debug] STACK:13424, HEAP:298+8184 (272+8064), BUSY:0%, FPS:50 (50+0+0), BATT:12
[debug] STACK:13424, HEAP:298+8184 (272+8064), BUSY:0%, FPS:50 (50+0+0), BATT:12
---- Closed the serial port /dev/tty.wchusbserial5A4E1051401 ----

rapha-tech, 2025-08-03:
Your RG_GPIO_SDSPI_MISO and RG_GPIO_SDSPI_MOSI pins are the same in your config.h

rapha-tech, 2025-08-09:
It seems like you still have problem with sd. Can you share your full config.h ?

rapha-tech, 2025-08-09:
Have you tried it without
```c
// Backup SPI mode config (in case SDMMC fails)
#define RG_STORAGE_SDSPI_HOST SPI3_HOST
#define RG_STORAGE_SDSPI_SPEED SPI_MASTER_FREQ_10M
#define RG_STORAGE_SDSPI_MISO GPIO_NUM_40 // reuse SD_DATA as MISO
#define RG_STORAGE_SDSPI_MOSI GPIO_NUM_40 // reuse SD_DATA as MOSI
#define RG_STORAGE_SDSPI_CLK GPIO_NUM_39 // reuse SD_CLK
#define RG_STORAGE_SDSPI_CS GPIO_NUM_38 // reuse SD_CMD as CS

// Flash storage as fallback
#define RG_STORAGE_FLASH_PARTITION "storage"
```

kingmaxmmxweisun, 2025-08-09:
thx [rapha]  it seems sd is working ,it's a big save for my life. thank you 

I (1428) main: Application started successfully
I (1438) main: Starting simple launcher interface...
I (1448) main: Starting simple launcher main loop
---- Closed serial port /dev/tty.usbmodem14101 due to disconnection from the machine ----
---- Reopened serial port /dev/tty.usbmodem14101 ----
I (762) esp_psram: SPI SRAM memory test OK
I (771) cpu_start: Pro cpu start user code
I (771) cpu_start: cpu freq: 160000000 Hz
I (772) cpu_start: Application information:
I (774) cpu_start: Project name:     retro-go-esp32s3
I (780) cpu_start: App version:      1
I (785) cpu_start: Compile time:     Aug  9 2025 14:53:44
I (791) cpu_start: ELF file SHA256:  451c2f1a5...
I (796) cpu_start: ESP-IDF:          v5.2.1-dirty
I (801) cpu_start: Min chip rev:     v0.0
I (806) cpu_start: Max chip rev:     v0.99 
I (811) cpu_start: Chip rev:         v0.2
I (816) heap_init: Initializing. RAM available for dynamic allocation:
I (823) heap_init: At 3FC9BFA8 len 0004D768 (309 KiB): RAM
I (829) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (835) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (841) heap_init: At 600FE014 len 00001FD4 (7 KiB): RTCRAM
I (848) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (856) spi_flash: detected chip: generic
I (860) spi_flash: flash io: dio
I (864) sleep: Configure to isolate all GPIO pins in sleep state
I (871) sleep: Enable automatic switching of GPIO sleep configuration
I (878) main_task: Started on CPU0
I (888) main_task: Calling app_main()
I (888) main: Starting application with ESP-IDF v5.2.1
I (888) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (898) gpio: GPIO[39]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (908) gpio: GPIO[47]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (918) main: GPIO configuration completed - avoiding gamepad GPIO conflicts
I (928) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (938) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (948) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (958) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 

========================================================
launcher unknown (Aug  9 2025 14:53:49)
 built for: ESP32S3-WROOM. type: release
========================================================

[info] rg_storage_init: === 🔧 Full test of SDMMC native mode ===
[info] rg_storage_init: ✅ Hardware status: 10kΩ pull-up resistor (3.28V) verified successfully
[info] rg_storage_init: 🔧 GPIO config: CMD=38, CLK=39, D0=40 (SDMMC dedicated pins)
[info] rg_storage_init: 📊 System status: PSRAM 8MB + 32KB DMA memory reserved
[info] rg_storage_init: Initializing SDMMC native mode...
[info] rg_storage_init: SDMMC config details:
[info] rg_storage_init:   - Frequency: 20000KHz
[info] rg_storage_init:   - Width: 1-bit mode
[info] rg_storage_init:   - GPIO CMD: 38
[info] rg_storage_init:   - GPIO CLK: 39
[info] rg_storage_init:   - GPIO D0:  40
I (1038) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1048) gpio: GPIO[38]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1058) gpio: GPIO[40]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[info] rg_storage_init: 🎉 SDMMC mode SD card mounted successfully! Frequency=20000KHz
[info] rg_input_init: Initializing ADC gamepad driver...
[info] rg_input_init: Initializing ADC battery driver...
[info] rg_input_init: ADC1 calibration initialized successfully
[info] rg_input_init: Input ready. state=00000000 00000000
[info] json_root: Config file loaded: '/sd/retro-go/config/global.json'
[info] json_root: Config file loaded: '/sd/retro-go/config/boot.json'
[info] rg_display_init: Initialization...
[info] json_root: Config file loaded: '/sd/retro-go/config/launcher.json'
[info] lcd_set_backlight: backlight set to 80%
[info] rg_display_init: Display ready.
[info] rg_alloc: SIZE=4608, CAPS=SPIRAM|8BIT, PTR=0x3c07774c
[info] rg_gui_set_language_id: Language set to: English (0)
[info] rg_gui_set_font: Font set to: VeraBold 11 (height=13, scaling=1.00)
[error] rg_storage_read_file: Fopen failed (2): '/sd/retro-go/themes/default/theme.json'
[error] rg_gui_set_theme: Failed to load theme JSON from '/sd/retro-go/themes/default/theme.json'!
[info] rg_gui_set_theme: Using built-in theme!
[info] rg_audio_init: Audio ready. sink='Dummy', samplerate=22050, volume=50
[error] rg_storage_read_file: Fopen failed (2): '/sd/retro-go/cache/clock.bin'
[info] rg_system_load_time: Time is now: Fri Aug  8 23:53:35 2025

[info] rg_system_init: Available memory: 323/366 + 8157/8189
[info] rg_system_init: Retro-Go ready.

I (1428) main: Application started successfully
I (1438) main: Starting simple launcher interface...
I (1448) main: Starting simple launcher main loop

closed 2025-08-09 by kingmaxmmxweisun