ducalex/retro-go #218: Built in updater - Should it work on other devices?

DynaMight1124, 2025-08-03:
Hi

I understand that the Odroid and G32 devices allow onboard updating via the .fw firmware files, are other devices able to update in such a way? 

For instance the VMU and Null Nano share the Odroid reference design mostly so I had assumed that they would be able to update via this method also? I was kinda hoping as both are not easy to flash since they have no onboard flash circuitry :) 

I hadnt really looked into it too much as they were not part of an official release (although there were no official .fw files compiled, so not sure if thats an indication or not!)

I have compiled my own .fw files and on the VMU for instance, I placed it in /vmu/firmware. Am I correct that I should be able to turn on with a button held and in the recovery menu get an 'reboot to updater' option? I dont get that option for some reason. I can connect to the internet, download the latest vmu.img file and I noticed that does get correctly placed into the vmu/firmware directory but a reboot didnt work.

I've spent some time comparing odroid target files and the main config.h and I cant see anywhere that may cause a difference.

Apologies if I misunderstood and that method is only available on the 2 above devices maybe due to some additional onboard hardware.

Thanks

P.S. on the 1.45 changelog there were some missing updates :)

New device support: VMU
New device support: CrokPocket by Megazoid
Added support for additional GPIO Extenders: MCP23017 & PCF8575

rapha-tech, 2025-08-03:
Hey Dyna,
You new a special firmware on your device to "install" retro-go's .fw files. Here is the link to ducalex's version of it:
https://github.com/ducalex/odroid-go-multi-firmware
It basically allows you to install multiple apps (such as retro-go's) on your device as the name implies

DynaMight1124, 2025-08-03:
Hi

Oh yeah, I understand now. I wasnt aware of this before now! I never had a Odroid Go but do kinda remember seeing videos people loading apps in a similar way, so must had been that!

Looks like it could maybe be adapted for the VMU etc, assuming the display driver works as thats pretty much the only main difference compared to the original Odroid. Sounds like a future project for a (very) rainy day :)

ducalex, 2025-09-07:
I've pushed the first working prototype to the dev branch!

To test if you need to do a few things:

Make sure your target's config.h has these:
```c
#define RG_UPDATER_ENABLE               1
#define RG_UPDATER_APPLICATION          "updater" // <--- Specifically this
#define RG_UPDATER_DOWNLOAD_LOCATION    RG_STORAGE_ROOT "/nullnano/firmware"
```

Then add "updater" to your target's env.py DEFAULT_APPS:

- You can either list all the apps you want: `DEFAULT_APPS = "launcher ... updater"`
- Or you can just add updater to the default value: `DEFAULT_APPS = f"{DEFAULT_APPS} updater"`

Then build a new full image and flash it through serial.

The updater is accessible through the launcher's about menu. 

Several limitations remain:
- The updater will ignore new or bigger partitions in the .img file
- If the updater fails during the launcher update, there will be no way of going back in it so the device will be "bricked" until you flash over serial. 
- No integrity check (but I do embed a CRC32 in .img now)
- The progress display is a bit messy with big font/small screen
- No rollback mechanism. It would be nice to add but will "waste" 2MB of reserved flash
- Probably tons more...

All fixable issues that I have already overcome in my other project, odroid-go-multi-firmware, but I think I'll wait for the updater to work reliably before working on these issues.

DynaMight1124, 2025-09-07:
That sounds awesome, I won't be able to test for a week or so as I'm away from home currently but I'll look forward to testing.

Few thoughts and this might be covered and I'd normally scan the code to try to understand first but not doing that on my phone 😁 you mentioned about issues due to partition sizes, does it currently have ample space for the current apps in that partition to grow? I don't know how close they are or how much additional space maybe required etc, e.g would changing to a few IDF version increase the size? Just thinking that giving extra space at this stage may help with seamless future updates if there big changes to certain apps while keeping the 4mb limit in mind but a much bigger tolerance for 16mb devices.

rapha-tech, 2025-09-07:
Hi @ducalex ,
The updater would be really nice to have!
I don't know if this is possible or a viable solution, but maybe merging all emulators into a single partition could be a bit simpler for updating, like instead of having to leave headroom for gwenesis, retro-core, pr-boom-go.... We would just have a big "emulator" partition taking up all the space left after launcher and updater + headroom.
Also it would be possible to update the firmware even if an emulator is added. 
But I was wondering if there was a reason for some emulators being separate apps?

DynaMight1124, 2025-09-24:
Looks like some great improvements to the updater! 

Its cool about the CYD and stuff, at the time I was more thinking if the updater was excluded it would had still been nice to be able to keep the same packages as v1.45 but there comes a time where you cant keep making comprises for the sake of the odd device, as long as core and launcher fits then that covers most of the major systems imo. 

If having SNES standalone makes it faster then its certainly worth the extra space and like you said its getting very close to fully playable now!

ducalex, 2025-09-24:
I understand your disappointment entirely, trying to fit as much as possible in 4MB was always a main goal of retro-go and I spent so much (enjoyable) time experimenting with ways to achieve it!

There is still https://github.com/ducalex/retro-go/issues/167 that I'm hopeful to achieve one day. I almost did it earlier this year but I had to undo it when I found performance issues in some specific SMS games, where the standalone retro-core normally never frame skips more than 1.

Now that snes9x is no longer part of retro-core I will probably give launcher-retro-core another shot, it might be just enough freed RAM to make it work.

DynaMight1124, 2025-09-24:
I was thinking the other day that fitting everything into the space, speed and memory constraints of a ESP32 must be pretty challenging. When I built up my first RetroGo device (Null Nano) probably the first thing that stood out was how great the software side was, felt like something from a much more power device. Everything was very integrated and felt seamless, something I wasnt really expecting from such a low powered device if I'm honest. 

I guess having more standalone cores/apps etc means duplicating a lot of the core functionality? Which takes up additional space duplicating it. Interesting. Honestly I'm not too worried about the CYD, I knew going in it was gonna be limited, strangely its been a pretty popular project but if needed it can be left with v1.45.

I have been working on an updated VMU with an S3 so at some stage I may do a pull request to update that as long as the new revision works as intended.