ducalex/retro-go #224: Working on Doom Multiplayer

jefthebuilder, 2025-08-27:
Hi,

I am working on a working version of doom multiplayer using a central server.
I already have the multiplayer working but i am still fixing some small errors like position desync.
I am now using a way to start multiplayer mode by adding to the wad path a | -net <ip_address>
where the | is a seperator so the path and the extra args are seperated.
Now the problem how do you all see how the way to start multiplayer is implemented my first thought was to add a seperate launcher menu instead of the standard doom menu and then ask for a server ip and port
but if i want to use that i need to update the theme and was asking how you got the pictures.
or if there is a better way pls tell me.

Greetings,

Jef

ducalex, 2025-08-29:
> I am working on a working version of doom multiplayer using a central server. I already have the multiplayer working but i am still fixing some small errors like position desync.

Impressive work! I always thought there wasn't enough RAM for a good multiplayer gameplay. Happy to be proven wrong :).

> Now the problem how do you all see how the way to start multiplayer is implemented my first thought was to add a seperate launcher menu instead of the standard doom menu and then ask for a server ip [...]

I'm not sure what you mean by the theme and pictures?

If you mean adding a menu to prboom, the DOOM WADs do not contain the entire alphabet in the menu's big font unfortunately. It's certainly possible to extend the already bundled PWAD and add the missing images but I wouldn't know how.

The simplest way to add menu items is to use the regular smaller font with `M_WriteText`. That is what I did for my own additions like cheats and level select (it's all in `m_menu.c`).

But I think it might make more sense to use the retro-go menu system to control the multiplayer aspect, because it can directly control wifi and we now have a virtual keyboard to type an IP and all that.

jefthebuilder, 2025-08-30:
Oh sorry for the confussion, I mean that in the launcher there is a seperate menu called multiplayer doom seperate from the normal doom but that has currently no textures/theme

also currently i have a changed the prboom server a lot because of some desync like player rotation and position gets over time out of sync because the standard prboom send input data instead of actual positions etc
 
Also i want to try to port and minimize lzdoom for a friend of my, i think it wont work but i will only know when i tried;)

Ps: sorry for some typos i am on phone currently.