ducalex/retro-go #226: I2S audio doesn't work when rebooting from an emulator

rapha-tech, 2025-08-31:
## Describe the bug ##
When launching an emulator from the launcher, the sound is enabled (expected behavior). However, when rebooting (powering off then on the device) from an emulator like GBC, `RG_GPIO_SND_AMP_ENABLE` is not high, which disables the amplifier

## To Reproduce ##
Steps to reproduce the behavior:
1. Power on the device
2. Go into the launcher
3. Launch a GBC game for example
4. You should have sound enabled
5. Power off the device
6. Power on the device
7. The device should automatically resume the game but no sound is present (RG_GPIO_SND_AMP_ENABLE is low)

## Expected behavior ##
RG_GPIO_SND_AMP_ENABLE should be high

## Additional context ##
When going into the menu and changing the audio out to dummy and then Ext DAC then going out of the menu (basically reinit the driver), RG_GPIO_SND_AMP_ENABLE is enabled and sound is working.

Also, editing `rg_system.c` with these lines solve the issue:

```c
    memset(&panicTrace, 0, sizeof(panicTrace));
    panicTraceCleared = true;

    update_memory_statistics();
    app.lowMemoryMode = statistics.totalMemoryExt == 0;

    app.indicatorsMask = rg_settings_get_number(NS_GLOBAL, SETTING_INDICATOR_MASK, app.indicatorsMask);
    app.romPath = app.bootArgs ?: ""; // For whatever reason some of our code isn't NULL-aware, sigh..

    rg_gui_draw_hourglass();

    // edited
    rg_audio_init(sampleRate);
    rg_audio_deinit();
    rg_audio_init(sampleRate);
    // edited

    rg_system_set_timezone(rg_settings_get_string(NS_GLOBAL, SETTING_TIMEZONE, "EST+5"));
    rg_system_load_time();
```

ducalex, 2025-08-31:
Have you checked the log? The amp is enabled only after the i2s driver is successfully initialized, so maybe the real problem is that it fails to initialize?

If you can't read the log for whatever reason I suggest you try replacing your edit with: `rg_audio_set_mute(false);`. If it doesn't work, then we have to assume it's an init problem and not so much a mute problem.

rapha-tech, 2025-08-31:
This is the log I get when rebooting from an emulator:

```
W (713) i2s(legacy): legacy i2s driver is deprecated, please migrate to use driver/i2s_std.h, driver/i2s_pdm.h or driver/i2s_tdm.h
W (715) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
[info] platform_init: Running platform-specific init...

========================================================
retro-core 1.46-dev-3-g1edec-dirty (Aug 31 2025 23:00:10)
 built for: RETRO-RULER. type: dev
========================================================

[info] rg_storage_init: Looking for SD Card using SDSPI...
[info] rg_storage_init: Storage mounted at /sd.
[info] rg_input_init: Initializing GPIO gamepad driver...
[info] rg_input_init: Initializing ADC battery driver...
[info] rg_input_init: Input ready. state=00000000 00000000
[info] json_root: Config file loaded: '/sd/retro-go/config/global.json'
[info] json_root: Config file loaded: '/sd/retro-go/config/boot.json'
[info] rg_display_init: Initialization...
[info] json_root: Config file loaded: '/sd/retro-go/config/gb.json'
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fcebe08
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fcec80c
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fced210
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fcedc14
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fcb5ac4
E (1022) spi: spi_bus_initialize(809): SPI bus already initialized.
[info] lcd_set_backlight: backlight set to 80%
[info] rg_display_init: Display ready.
[info] rg_alloc: SIZE=10080, CAPS=SPIRAM|8BIT, PTR=0x3c0e6dcc
[info] rg_gui_set_language_id: Language set to: Francais (1)
[info] rg_gui_set_font: Font set to: VeraBold 11 (height=13, scaling=1.00)
[info] rg_gui_set_theme: Using built-in theme!
[info] rg_audio_init: Audio ready. sink='Ext DAC', samplerate=32000, volume=50
[info] rg_system_load_time: Time loaded from storage
[info] rg_system_load_time: Time is now: Wed Dec 31 19:00:03 1969
```

Adding `rg_audio_set_mute(false);` after `rg_audio_init(sampleRate);` doesn't work the pin is still low.

ducalex, 2025-09-01:
I've moved the gpio configuration to `driver_init` rather than do it every time in `set_mute`. I've also added your `gpio_reset_pin(RG_GPIO_SND_AMP_ENABLE)` there.

Let me know if it's insufficient, we can use your new fix instead and reset the pin in set_mute even though it seems "wrong" to reconfigure the pin all the time as we're doing.

rapha-tech, 2025-09-01:
> I've moved the gpio configuration to `driver_init` rather than do it every time in `set_mute`. I've also added your `gpio_reset_pin(RG_GPIO_SND_AMP_ENABLE)` there.

This fixed the issue!
thanks ;)

closed 2025-09-01 by rapha-tech