ducalex/retro-go #230: Crashing with doit esp32 devkit v1

SzaboPeterBence1, 2025-09-06:
I want to install retro-go on a doit esp32 devkit v1 but it keeps crashing with errors, I already have my own target folder (maybe it's wrong), but somehow it always crashes during the build, so I'd like to ask for help.

I use 320x240 ILI9341 display.

closed 2025-09-06 by SzaboPeterBence1

ducalex, 2025-09-06:
The DOIT DEVIT V1 seems to use the ESP32-WROOM module which doesn't have enough RAM for retro-go. So even if you get it to build most emulators won't work.

But it should still build successfully, can you share the errors you're getting?

ducalex, 2025-09-13:
I don't understand how this error could occur, but you can try working around it by modifying rg_storage.h:

Change that line: `char path[RG_PATH_MAX + 1];` to `char path[256];`