ducalex/retro-go #231: ESP32-P4 port

fcipaq, 2025-09-06:
Hi,

first of all: thank you very much for you great work and the excellent porting guide!

I'm currently porting Retro-Go to the ESP32-P4. I'm using my open-source DIY ESP32-P4 handheld (if you're interested you can find an introduction here: [https://www.youtube.com/watch?v=5BC4VqG1UPU](url)). Now, the RG graphics hardware abstraction is focuses on transmitting a buffer to an LCD with GRAM and I can see why. However, my handheld has a MIPI-Display that relies on timing-sensitive buffer transmission so I had to make some changes to rg_display.c
Further changes may be necessary in the future to increase emulation speed (I already ported Snex9x only from the retro-go repo and it runs many games at full speed).

Now, here's my question: Is there general interest in merging these changes? (My repo is currently private, but I'm more than happy to grant access if you'd like to see what I modified).

ducalex, 2025-09-09:
Someone shared your video in another issue/PR related to the P4, happy to see you here!

As far as adding a new display driver and new device target, I typically accepts anything as long as: the drivers be self contained as much as possible, and the device be available for purchase or that build instructions be available.

But merging significant changes to core retro-go files we'd likely have to discuss it and iterate on it in a PR. Alternatively if you don't want this long process you can just show me your code privately so I can better understand your needs and I'm sure I'd be able to make some tweaks to at least make your life easier! Things like adding new hooks or ways to bypass part of the rg_display code or inject code from the driver file should be no problem at all.

Or maybe you just need first-class esp_lcd support in retro-go? Because I have a stash somewhere in my local tree doing just that. It was never merged because, iirc, it didn't work well (at least with SPI). It frequently crashed due to bus conflicts with the SD card and it also didn't have an API to set low level things like gamma or voltage curves or at least issue the raw commands so it looked like crap. Also I think the draw/send function worked by copy so it created twice the memory transfers. But it's been a while, maybe it improved?

And finally, any improvement to snes9x or other emulators I would likely accept without any question. :)

fcipaq, 2025-09-10:
> Someone shared your video in another issue/PR related to the P4, happy to see you here!

I had no idea there was going on so much with the P4, great to hear that! 
Next to the things you've already discussed, I also put some of the hot paths in IRAM, especially the CPU Ops. Moreover, GFX.Screen, GFX.SubScreen, GFX.ZBuffer and GFX.SubZBuffer need to be in IRAM or that'll hurt the performance big time. I ran out of RAM so I reduced the tile cache to have more RAM available and also GFX.SubScreen = GFX.Screen and to my surprise all games I tried didn't show any glitches but my understanding of Snes9x internals is too little to really get all the implications.
Things done by the 2nd core won't eat into CPU0's performance, however, it will use some bandwidth (especially true for PSRAM) and increase the probability of cache misses.

> As far as adding a new display driver and new device target, I typically accepts anything as long as: the drivers be self contained as much as possible, and the device be available for purchase or that build instructions be available.

Okay, I'm planning on fully open-sourcing it so that everyone can build one. :) Regarding the containment of the drivers: Currently, the app's performance is significantly impacted by the creation of an "rg_surface" buffer, which is then processed by rg_display and eventually submitted to the display. And that is a formidable way of doing it when you have an SPI/MCU attached. However, with a DSI or RGB display connected, the preferred approach is probably for the app to create two surfaces for double buffering.
Now, one could modify the apps using a double-buffering approach only when using a DSI/RGB display. In Snes9x, I see you already had something like that in mind with `static rg_surface_t *updates[2]`. Alternatively, the apps could query the surface's data buffer pointer after submission to access the other framebuffer like `GFX.Screen = rg_surface_get_buffer(my_surface)`, effectively flipping the buffer. It would probably only be a couple lines of code that would need to be changed. I'll check that out.

> But merging significant changes to core retro-go files we'd likely have to discuss it and iterate on it in a PR. Alternatively if you don't want this long process you can just show me your code [...]

Btw.: The code is already public (apparently you cannot have a private repo that was forked from a public one)

> Or maybe you just need first-class esp_lcd support in retro-go? Because I have a stash somewhere in my local tree doing just that. It was never merged because, iirc, it didn't work well (at least [...]

You are right about that. The original driver copies everything upon writing, which is very slow. I modified it so that any given buffer can be scanned out. The supplied buffer does not have to match the screen resolution. So if you're, say, using a 720x720 screen (like I do) you can still have a 320x240 buffer submitted. You can then have the image integer scaled and rotated. However, this is not ideal either since the CPU does the heavy lifting and it takes up quite some memory bandwidth as well. Maybe this is a good job for the ESP32-P4's "2D graphics accelerator" - had no time to look into it.

> And finally, any improvement to snes9x or other emulators I would likely accept without any question. :)

Great! It's a great project. You and the community have really achieved something! And I'd be more than happy to contribute if I can. Question is, what would be the best way to collaborate? If you - or someone else - wants to help port this to the "Picoheld 2" - I have a spare prototype lying around that waits to be tinkered with :) - it only has a plastic shell, but I'd be happy to give it away (for free :) ).

rapha-tech, 2025-09-10:
Hey, 
I saw you used the master branch but the most recent changes are on the dev branch, make sure to check it out ;)
Also a few of the fixes for the P4 have been done in here (like the bootloader offset, the 4bit SDMMC with the on chip LDO, the L/R/X/Y support on SNES9X, and the general speed improvements on SNES9X)

I saw you were using a fairly slow SDMMC speed and I had the same issue before using the on chip LDO (but I was personnaly using a devkit so maybe it won't change anything on your console).

ducalex, 2025-09-13:
> Moreover, GFX.Screen, GFX.SubScreen, GFX.ZBuffer and GFX.SubZBuffer need to be in IRAM or that'll hurt the performance big time. 

Just to be clear with IRAM you mean Internal RAM and not Instruction RAM here? I'm asking because although espressif themselves sometimes call internal ram IRAM, IRAM usually refers to the instruction RAM which, on xtensa esp32s at least, can only be accessed in 32bit words so that won't work for those 8/16 bit buffers.

Anything that isn't explicitly allocated in internal RAM already is either because it didn't fit (most likely) or I rely on `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL` to make it happen implicitly. Reason is simply that I try to avoid `rg_*` calls in the apps themselves to keep them more "portable". Obviously if you find that calling rg_alloc(..., MEM_FAST) explicitly from the app makes a difference I will be pragmatic about it :).

> In Snes9x, I see you already had something like that in mind with static rg_surface_t *updates[2]

About half my apps use double buffering. When they don't it's always because:
- Emulation is so slow that we have plenty of time to scale and send it over SPI before the app gets to overwrite it, so it doesn't actually help avoid artifacts or improve speed, it just wastes RAM.
- The second buffer doesn't fit in internal RAM and I found that having a framebuffer in SPIRAM is, in many cases, worse than having a single framebuffer.

In the dev branch there is a define to enable double buffering and `GFX.Screen` is flipped every time a frame is sent to the display.

From what I'm understanding, for your device to work well you just need a way to ditch the display_task mechanism entirely and have rg_display_submit let your driver do whatever with the raw framebuffer? Because all apps with double buffering already flip the buffer after calling rg_display_submit, so I don't think any modification to the apps is required. Although many apps use paletted framebuffers (nes, sms, pce, doom, genesis) so your driver would have to handle that (it's currently handled by display_task/write_update). I'm not opposed to adding 565 renderers directly to those apps either (behind a flag, because the og esp32 currently benefits more from the smaller framebuffers than it would from a simpler blitting pipeline).

ducalex, 2025-09-13:
> it also didn't have an API to set low level things like gamma or voltage curves or at least issue the raw commands so it looked like crap

Also to anyone reading my earlier comment I've found that I am wrong about that, there is [esp_lcd_panel_io_tx_param](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/lcd/index.html#_CPPv425esp_lcd_panel_io_tx_param25esp_lcd_panel_io_handle_tiPKv6size_t). But the rest of my issues with esp_lcd still seem to apply as of esp-idf 5.4 unfortunately.

fcipaq, 2025-09-14:
> Just to be clear with IRAM you mean Internal RAM and not Instruction RAM here? I'm asking because although espressif themselves sometimes call internal ram IRAM, IRAM usually refers to the [...]

Yeah, sorry, my bad. I meant internal RAM as in `MALLOC_CAP_INTERNAL`

> Anything that isn't explicitly allocated in internal RAM already is either because it didn't fit (most likely) or I rely on `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL` to make it happen implicitly. Reason [...]

The emulated RAM is usually relatively slow whereas the emulator's RAM must be fast. When relying solely on `malloc`, some memory may be allocated in internal RAM, even though it doesn't need to be fast. This takes up space that could be used for faster memory. For example, in `memmap.c` `Memory.RAM` is allocated at the beginning, potentially taking up all the space in internal RAM and `IPPU.ScreenColors`, which might be frequently accessed, won't fit into internal RAM and will be allocated in PSRAM. Therefore I would prefer to explicitly control what memory is allocated where. Moreover, the default value for `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL` is 32 kB, with anything larger than that will be allocated in PSRAM. For example `rg_surface_create(SNES_WIDTH, SNES_HEIGHT_EXTENDED, RG_PIXEL_565_LE, 0);` would always be allocated in external memory.

> > In Snes9x, I see you already had something like that in mind with static rg_surface_t *updates[2]
> 
> About half my apps use double buffering. When they don't it's always because:
> 
>     * Emulation is so slow that we have plenty of time to scale and send it over SPI before the app gets to overwrite it (due to frameskip or just not reaching 100% speed anyhow), so it doesn't actually help avoid artifacts or improve speed, it just wastes RAM.
> 
>     * The second buffer doesn't fit in internal RAM and I found that having a framebuffer in SPIRAM is, in many cases, worse than having a single framebuffer.
> 
> 
> In the dev branch there is a define to enable double buffering and `GFX.Screen` is flipped every time a frame is sent to the display.
> 
> From what I'm understanding, for your device to work well you just need a way to ditch the display_task mechanism entirely and have rg_display_submit let your driver do whatever with the raw framebuffer? Because all apps with double buffering already flip the buffer after calling rg_display_submit, so I don't think any modification to the apps is required. Although many apps use paletted framebuffers (nes, sms, pce, doom, genesis) so your driver would have to handle that (it's currently handled by display_task/write_update). I'm not opposed to adding 565 renderers directly to those apps either (behind a flag, because the og esp32 currently benefits more from the smaller framebuffers than it would from a simpler blitting pipeline).

What I'm trying is to implement a `rg_display_submit_direct` next to the "regular" `rg_display_submit` allowing a direct write to the framebuffer in SNES. I think the other emulators are less critical and may use the "regular" way as does the menu. There are still some minor issues when switching between the two modes. But it generally works.
I now get 50 fps out of Super Mario World's intro with every frame being rendered, which is good I guess. To do this I had to:

- Put two screen buffers in internal memory (and that means Subscreen = Screen and yes, that is a hack *gulp*)
- Put the ZBuffer in RAM
- Disable tile buffering (which does no good due to the slow external RAM)
- Put the CPU Ops along some other stuff in IRAM

If you're interested you can take a look in the repo (dev branch), but it's a WIP :)

fcipaq, 2025-09-15:
> > Oh, and it looks like there's an issue with the audio sampling rate. The audio task's queuing limits the main thread to 50 FPS (`xQueueSend(task->queue, msg, portMAX_DELAY) == pdTRUE;`) The target video FPS rate, however, seems to be 60. Consequently, the video can never catch up with the set rate, which triggers massive frame dropping.
> 
> Is it because I mix the entire AUDIO_BUFFER_LENGTH every frame? Which is set to accomodate 50 fps, so the audio code currently always renders a bit more than 1 frame when playing NTSC games?

I think the issue might be: `#define AUDIO_BUFFER_LENGTH (AUDIO_SAMPLE_RATE / 50 + 1)` (`shared.h`) vs `.frameTime = 1000000 / 60` (`rg_system.c`)

ducalex, 2025-09-16:
I've looked at your repo and doh I just removed double audio buffering in SNES, I guess I'll have to add it back (or wait for your code)!

> Maybe you could take a look and tell me what I need to change to give my code a better chance of getting successfully PRed.

I've only skimmed but so far most of the changes seem fine.

Your changes to SNES break compilation on og esp32, (IRAM overflow). Once you mirror my change of making snes9x an independent app it should resolve the issue. I'll do it on my side and test later on.

I've now added a timeout parameter to the task functions. I didn't like your `non_blocking` implementation because it doesn't really imply that the message will be lost, just that the call won't block. Timeout is a bit clearer (to me) and it matches the existing mutex functions.

The only thing I'm not sure how I feel about is having to carry esp_lcd. It's a bit annoying to have so much extra code and ensure it keeps working every time a new esp-idf version breaks something. 

I haven't looked at what you've patched in it exactly (yet) but is there no way to accomplish the same thing from either lcddrv or dsi.h's code?

fcipaq, 2025-09-16:
> I've looked at your repo and doh I just removed double audio buffering in SNES, I guess I'll have to add it back (or wait for your code)!
> 
> > Maybe you could take a look and tell me what I need to change to give my code a better chance of getting successfully PRed.
> 
> I've only skimmed but so far most of the changes seem fine.
> 
> Your changes to SNES break compilation on og esp32, (IRAM overflow). Once you mirror my change of making snes9x an independent app it should resolve the issue. I'll do it on my side and test later on (Edit: I meant that I'll patch my local version, not steal your changes and commit them here).

Oh, that's too bad, sorry for the inconvenience. But I guess the ESP32 will benefit from that, too. BTW: How fast is Snes9x on the ESP32? On the ESP32-S3 I get 45 fps (that is 22 rendered frames per second) for Super Mario World's intro. Oh, and forget everything I said about the tile cache. You were right, it won't speed up things but might slow the emulator down. I now remember that I did this to make it fit into the Teensy 4.1's RAM (as I had not PSRAM attached). So it was not about speed but size. Sorry for the discombobulation :)

> I've now added a timeout parameter to the task functions. I didn't like your `non_blocking` implementation because it doesn't really imply that the message will be lost, just that the call won't block. Timeout is a bit clearer (to me) and it matches the existing mutex functions. It also gives more control to the caller, if they're willing to wait 10ms rather than 0 or infinity.M

Cool.

> The only thing I'm not sure how I feel about is having to carry esp_lcd. It's a bit annoying to have so much extra code and ensure it keeps working every time a new esp-idf version breaks something.

And can see why and you are absolutely right about that. Maybe I can find some way to extract the "essentials". But that's probably going to take me a while.

> I haven't looked at what you've patched in it exactly (yet) but is there no way to accomplish the same thing from either lcddrv or dsi.h's code?

I think that would probably be a difficult task. I'll definitely check it out.

ducalex, 2025-09-16:
> Are you trying to adapt the generated samples depending on how long it took to render the frame? I don't understand what you are doing with the following:
> 
> ` samplesRemaining += samplesPerFrame;` ` int samples = (int)samplesRemaining;` ` samplesRemaining -= samples;`
> 
> Won't this result in always the same number of samples (since you add `samplesPerFrame` and then subtract `samples` which is always the same if you start out with `samplesRemaining`) - am I missing something?

I think you missed that samplesPerFrame and samplesRemaining are floats. Reason is that 32040 / 50 = 640.8. So we have to either round to 640/641, or we carry the .8 for a more accurate speed overall. But anyway those changes were reverted in 845c6ea0f59c54b2bca80afac9b41a71bd052a1e. A better option might be to use 32000hz for PAL and 32040 for NTSC?

> My suggestion also includes a different approach to the frame dropping mechanism.

Frame skipping in retro-go is very primitive as you've seen. At some point I kept a rolling average of frame times to adjust the auto skip which worked better but I never completed it. I don't think I've attempted anything fancier. So I would welcome any enhancement even if it's just for one app!

> I wasn't sure, for example, about the launcher, which seems to resize surfaces

The launcher resizes things like images but the surface that is actually sent to the display is treated like a raw continuous uint16[] buffer by both the launcher and rg_gui (iirc).

`rg_surface_create` has an alloc flag, perhaps we could add a bit to say that you want the buffer in display memory. And then behind the scene that buffer can be provided by the display driver. Surfaces used as framebuffers could specify that bit whereas surfaces used for drawing will remain as is. Although that won't do any good for apps where the format isn't the exact same as the display (possibly most of them), it will have to be transformed anyhow 🤔.

>  So, instead of making RAM the default, one would have to modify every app to tell the driver to reallocate in RAM and I thought that's something you would not like to happen. But if you agree to do so, then yes, that would be perfectly fine for me.

I've actually merged my config struct yesterday. It's still a work in progress and I don't know how many things I want to include in it, but it already contains `mallocAlwaysInternal` so why not add another one to dictate display driver allocation if it helps you? I think it's a better solution than reallocating after the fact because as you've said, our tiny heap is very easy to fragment.

https://github.com/ducalex/retro-go/blob/dev/components/retro-go/rg_system.h#L136-L149

fcipaq, 2025-09-21:
Hi, I tried a couple of things to improve the glue between Retro-Go and the display driver. However,I didn't really come up with a proper solution. `rg_surface_submit` is no problem. What is problematic is the way `lcd_buffer_send `(in conjunction with `rg_display_write_rect` etc.) works. That's why I can't just send a surface directly to the screen. It makes a back buffer that replaces the external display's GRAM mandatory. And in some scenarios there just isn't enough RAM for that.

So, what I tried is: a. to have a back buffer in SPIRAM (was too slow, causes flickering in some situations), b. triple back buffering in SPIRAM - causes data inconsistencies between core 0 and the display call back function running on core 1, which I was unable to resolve c. to have a back buffer in internal RAM on some apps - it works but has visible glitches due to lack of double buffering or is super slow if you wait for a vsync on every `lcd_send_buffer`.

The question is, how we're gonna deal with this. If you don't want to "just somewhat make it work", one possible solution might be: a. use the P4's graphics accelerator ("PPA") to rescale and rotate the frame buffer. *If* I can make this work (never used it), there would be no need to run the display on core 1 and that would help with the data inconsistencies (the display driver causes a heavy IRQ load on core 1 with an IRQ triggered for every line - that's 12,000 per second @ 50 fps!) Also, this would make the "hacked" esp_lcd driver superfluous and would be a more generic approach that basically supports all kinds of MIPI display (which I'm sure we're gonna see a lot in upcoming P4 products due to facilitated PCB routing)  b. we would need to find a way in `rg_display.c` to mitigate direct writes to the external frame buffer so that only complete surfaces are transmitted. Writes using `lcd_buffer_send` are difficult to handle when using triple buffering. Perhaps something like, say, `rg_blit_buffer`, and - depending on the display type attached - this goes directly to the screen or otherwise goes to a buffer.

Ultimately, it all comes down to whether you want Retro-Go to support displays without external GRAM or if you prefer to keep it centered on SPI. I mean there are good reasons for both options.

That being said, Snes9x with Retro-Go runs great on the P4, I uploaded a short video [here](https://www.youtube.com/watch?v=m7ioy6VdjsU). (I hope I pronounced your name correctly, sorry if I did not).

I suggest that I start out with the commits that *should* speed up Snes9x on the older ESP32s. And I would commit the P4 only after it is in a better working state.

Please let me know me what you think about it.

fcipaq, 2025-09-24:
> Yeah retro-go sets a window and then sends color data to fill it. But it's not thread safe, so depending on how you're emulating the lcd_* behavior, it's easy for something else to update the window whilst the display task is still sending pixel data. I've had that issue when working on the SDL2 port.

Ok, that's what I thought. And that makes perfectly sense when using a display with GRAM.

> Another thing that causes partial updates are, well, partial updates. The display task does a diff between the new frame and the previous one and sends only differing blocks. Partial updates are very helpful on the SPI side, they reduce tearing and increase the effective ramerate (otherwise we can barely reach 30fps at 40Mhz. 80Mhz is too glitchy and the APB doesn't allow us to use 60Mhz...). But it's definitely a waste of resources if your display is fast enough. If you `#define RG_SCREEN_PARTIAL_UPDATES 0` it should neuter it.

Ok, I'll give it a try.
 
> I'll catch up with your branch in the upcoming days and see how your display code is currently working. I also saw that you seem to have fixed the audio issue. Was the main issue just the size of the dma buffers? If so then I could probably find a nicer way than having a target-specific value.

That is what I suspected, but it turns out the DMA buffers are not the issue (but increasing them might still help with taking some load off of the CPU). My working theory for the crackling is as follows: Since the sound generation is bound to the signal (task message or semaphore) from the main thread, there is no way for the sound to get ahead of the picture. If, however the image generation falls behind, it can catch up by dropping the next frame. But the sound cannot. So the buffer will run empty and you will be able to hear a crackling. That must have been the reason why I used multiple buffers back then, effectively letting the image run ahead a couple of ms of the sound so that in case the image lags for a single frame, the sound buffers won't run empty. That being said, I just decoupled the sound generation from the main thread and to my surprise it works (I would have guessed you cannot generate sound at any point but apparently at least in some games it possible). Still, it would be better have the sound triple buffered and let the image run one frame (i.e. 17-20ms) ahead of the sound.

> If you want you can open a PR with just your snes9x improvements so we could get those merged whilst the details of the new P4 display are still being worked on.

I could try to PR the changes this weekend (this is a busy week; I'm going at full speed :) ). However, if you prefer catching up yourself, please feel free to do so. It would probably be easier this way since I have no way to test it on an ESP32/ESP32-S3.

HeathenUK, 2025-09-29:
> Hi! I made some progress over the weekend. First of all, I had to discard my previous idea of using the PPA. There only is little documentation available online, but as it seems, the PPA won't output a stream of pixels that you can forward to the display FIFO but rather processes blocks of pixels, and so the PPA needs an additional target frame buffer, and that won't do us no good. At least that's how I understood it. Now, I was able to contain the display driver and only keep the "hacked" part. The rest of the driver (like bus setup, and so on) is just the default `esp_lcd` driver from the IDF. It's now all contained in `retro-go/drivers/display/dsi`. Since I was unable to reliably make the `rg_display` system work, I've switched to one of your earlier suggestions:
> 
> > From what I'm understanding, for your device to work well you just need a way to ditch the display_task mechanism entirely and have rg_display_submit let your driver do whatever with the raw frame buffer?
> 
> However, this also requires all apps to be slightly modified. Because then one has to pass a frame buffer to the DSI driver (all previous attempts with the driver auto-detecting these were subpar and might lead to unexpected behavior in certain situations, also it would be counterintuitive). The launcher and Snes9x are already working pretty well. But, as you've mentioned before: I still need to implement the palette lookup, amongst some other things.
> 
> Maybe you can take a look at the code and tell me what I would need to change for a PR. That would help me a lot, as I wouldn't have to manually keep up with your repository. Also, maybe it's time to come up with names for the `RG_SCREEN_DRIVER` values (the numbers might get confusing at times...)

I might have misunderstood what you were hoping to use the PPA for, but isn't the DMA2D block specifically intended to automatically shuffle memory into and out of the PPA automatically to any valid dma destination, including SPI FIFO?

G

HeathenUK, 2025-09-29:
> PSRAM is probably by far too slow if you're aiming for 50-60 fps. Please bear in mind that the screen resolution is 720x720, which means 120 MB/s have to be juggled around (reading and writing that is).

The on-package psram for the esp32p4 defaults to 200MHz and is wired for hex (Hyper/HPI, 16 bit data) mode, so on a pure wire basis you're talking 400MB/s (and that's if it doesn't inherit the DDR mode from other hyperram). I'm sure the real world numbers are lower and in any case it won't compare to the on-die sram, but it may well be fast enough here.

I'm on holiday at the moment but this makes me want to run some benchmarks when I get home...

ducalex, 2025-09-30:
I've reviewed your changes so far:

1. Nice to see esp_lcd gone!
2. I think the `#if RG_CUSTOM_PLATFORM_DEINIT  ... endif` should be in shutdown_cleanup() instead.
3. Concerning the changes in rg_display, I think you should rely more on rg_surface. It should be as fast (or faster) than your current loops plus it has the bonus of supporting mismatched formats (not preferable, but it could happen). Example: [diff-to-0f83f0f.txt](https://github.com/user-attachments/files/22625949/diff-to-0f83f0f.txt)

The dsi driver itself seems fine though I don't have the knowledge to assess its correctness. I would require two changes:
1. Fence all the .c files with `if RG_SCREEN_DRIVER == 2`, because it currently fails to build on older esp-idf. Not the most elegant solution but the easiest.
2. LCD_H_RES, LCD_V_RES, LCD_ROTATION, PIN_NUM_LCD_RST should probably point to RG_SCREEN_WIDTH, RG_SCREEN_HEIGHT, RG_SCREEN_ROTATION, RG_GPIO_LCD_RST respectively. Maybe have `.rgb_ele_order = LCD_RGB_ELEMENT_ORDER_RGB,` use `RG_SCREEN_RGB_BGR` too.

I understand your frame buffer situation better now. I think that, by default, the driver should create the frame buffer. Because right now most applications will simply not work until you modify them, right?

However I understand that in some cases you want the app to control/create it.

So what I would suggest is adding a flag to rg_config_t, perhaps named `noAllocDisplayBuffer` or similar that (currently only) snes9x would set. Then the driver can do this:
```c
    if (!noAllocDisplayBuffer)
        rg_display_setup_fb(rg_surface_create(RG_SCREEN_WIDTH, RG_SCREEN_HEIGHT, RG_PIXEL_565_LE, MEM_FAST), 0);
```

That way all the other apps no longer need to be modified at all.

ducalex, 2025-09-30:
> Maybe you can take a look at the code and tell me what I would need to change for a PR. That would help me a lot, as I wouldn't have to manually keep up with your repository.

So to be clear if you submit a PR as is, I'd want the dsi changes done and the buffer allocation sorted out so that all apps work. The rest of the things I've suggested would not prevent me from merging your PR.

The Snes9x changes will take me longer to review because it will need more testing. Even as a separate app I don't think all your IRAM_ATTR will fit, I'm pretty sure I tried exactly what you did before. So I'll have to see what's the most impactful, and replace the rest with something like `P4_IRAM_ATTR` that defines to nothing on regular esp32. I also have to refresh my memory concerning the various screen buffers in snes9x, to make sure your changes affect no (or at least only a few) games.

ducalex, 2025-09-30:
> LCD_H_RES, LCD_V_RES, LCD_ROTATION, PIN_NUM_LCD_RST should probably point to RG_SCREEN_WIDTH, RG_SCREEN_HEIGHT, RG_SCREEN_ROTATION, RG_GPIO_LCD_RST respectively. 

You can also add new constants because I know that RG_SCREEN_WIDTH and LCD_H_RES aren't the same in your setup. Either a new set of width/height, or a multiplicator. Essentially what I want to achieve is allow other people to use your driver in their device without having to edit the driver code.

ducalex, 2025-09-30:
It occurred to me that it would be trivial to add support for a framebuffer in rg_display with very minimal changes. I've made a temp branch to show you: https://github.com/ducalex/retro-go/commits/fb_ptr/. Let me know if you think this could work, I'm not suggesting that we have to do it this way.

fcipaq, 2025-10-07:
Hi, sorry for my delayed response.

>     2. I think the `#if RG_CUSTOM_PLATFORM_DEINIT  ... endif` should be in shutdown_cleanup() instead.

I have two cases here: When rebooting only, the LCD and the amp have to be turned off in order to prevent visual or audio glitches. When actually shutting down the machine, then the MCU must turn off the power (the P2 has a load switch connected to a GPIO). What would you suggest would be the best way to differentiate between these two cases?

>     3. Concerning the changes in rg_display, I think you should rely more on rg_surface. It should be as fast (or faster) than your current loops plus it has the bonus of supporting mismatched formats (not preferable, but it could happen). Untested example: [diff-to-0f83f0f.txt](https://github.com/user-attachments/files/22625949/diff-to-0f83f0f.txt)

Ok, I'll look into it.

> The dsi driver itself seems fine though I don't have the knowledge to assess its correctness. I would require two changes:
> 
>     1. Fence all the .c files with `if RG_SCREEN_DRIVER == 2`, because it currently fails to build on older esp-idf. Not the most elegant solution but the easiest.

Ok

>     2. LCD_H_RES, LCD_V_RES, LCD_ROTATION, PIN_NUM_LCD_RST should probably point to RG_SCREEN_WIDTH, RG_SCREEN_HEIGHT, RG_SCREEN_ROTATION, RG_GPIO_LCD_RST respectively. Maybe have `.rgb_ele_order = LCD_RGB_ELEMENT_ORDER_RGB,` use `RG_SCREEN_RGB_BGR` too.

Here's one of the big problems: LCD_H_RES doesn't necessarily have to match RG_SCREEN_WIDTH, and that's where most of the problems come from (LCD_H_RES is 720 in my case, and so the image MUST be scaled). For the rest, I totally agree.

> I understand your frame buffer situation better now. I think that, by default, the driver should create the frame buffer. Because right now most applications will simply not work until you modify them, right?
> 
> However I understand that in some cases you want the app to control/create it.
> 
> So what I would suggest is adding a flag to rg_config_t, perhaps named `noAllocDisplayBuffer` or similar that (currently only) snes9x would set. Then the driver can do this:
> 
>     if (!noAllocDisplayBuffer)
>         rg_display_setup_fb(rg_surface_create(RG_SCREEN_WIDTH, RG_SCREEN_HEIGHT, RG_PIXEL_565_LE, MEM_FAST), 0);
> 
> That way all the other apps no longer need to be modified at all.

That's a good idea. Is it possible to tell the rg system to change the resolution (so it's no longer RG_SCREEN_WIDTH), because when allocating a frame buffer after rg system was initialised, the frame buffer won't generally match the RG_SCREEN_WIDTH/HEIGHT but will match the emulator's native resolution, would that require substantial modification?

fcipaq, 2025-10-07:
> So to be clear if you submit a PR as is, I'd want the dsi changes done and the buffer allocation sorted out so that all apps work. The rest of the things I've suggested would not prevent me from merging your PR.

Ok.

> The Snes9x changes will take me longer to review because it will need more testing. Even as a separate app I don't think all your IRAM_ATTR will fit, I'm pretty sure I tried exactly what you did [...]

Dual using the screen buffer as subscreen buffer will lead to visual artifacts in certain games. For example in Donkey Kong Country (e.g. background messed up in the first level). We could make it an emulator setting like "Subscreen buffer on/off" and in case it's on, we could rg_malloc it from the PSRAM. So everyone gets to choose.

fcipaq, 2025-10-07:
> It occurred to me that it would be trivial to add support for a screen/frame buffer directly in rg_display with very minimal changes. I've made a temp branch to show you: https://github.com/ducalex/retro-go/commits/fb_ptr/. Let me know if you think this could work, I'm not suggesting that we _have_ to do it this way.

I thought this was exactly the way I was trying to do it and I ran into trouble as soon as the stride didn't match (e.g. if an LCD window was set) - or is that what disabling the partial updates if for? What happens if RG_SCREEN_WIDTH doesn't match the buffer's stride?

fcipaq, 2025-10-07:
> I've tested the SNES changes (albeit on a single game) and there's definitely some improvements!
> The IRAM_ATTR changes seem to have decent benefits and cause no side effects, so I'd be happy to merge that part immediately (either commit it myself or through a PR you submit). In the future it might be worth exploring different optimization levels to reclaim some of that IRAM (your changes cause it to be 98% used with esp-idf 5.3. -Os on cpuops.c alone drops that to 67%, which leaves a fair amount free to be used for more tile/gfx code or heap/cache purposes).

Ok, that's a fair amount. I suppose having more of that hot code in RAM outweighs most certainly the speed optimisation in most cases, but thorough testing would be required to know for sure.

> The unlocked audio task shows potential in allowing it to run ahead. However the current implementation has some side effects that I don't like:
> 
>     1. there's no longer any speed cap, so emulation speed keeps going up and down, from 30 to 80fps.

Do you mean the audio fps? Because the main thread shouldn't be able to exceed the target fps, I get a steady 50 fps no matter what:

`
          if (framedrop_balance < -550)   // Allow some jitter
          {
            int64_t fpsLimiter = rg_system_timer();
            while (rg_system_timer() - fpsLimiter < (-framedrop_balance))
              ;
            framedrop_balance = 0;
          }
`

>     2. It makes save states unstable (I do not have an explanation for that one but I have a few theories...).

In what way? I observed the audio had trouble sometimes in save states, is that what you saw (or heard :).

> So I'm in favor of uncoupling it more from the main emulation loop but evidently we still need some coupling. I don't know if you have any ideas on your side? My first thought is to go back to the message queue but increase the depth from 1 to 3, allowing the audio task to be at most 1-2 frames ahead. Another thought is to keep counters on both sides and periodically compare and stall if needed.

That's probably how it needs to be done. But, I think the best way is let the audio playback fall one frame behind. This way, we can tightly couple the audio generation to the main thread while still preventing an audio buffer underrun if the emulation lags: When a frame is rendered too slowly, the frame time will exceed the target duration and the audio can be played back from that 1 buffer that the audio is running behind. The next frame will then be dropped and the frame time is below the target duration so that the audio buffer can be refilled. This way the audio buffer can never run empty while still having the audio running in lockstep with the main thread.

fcipaq, 2025-10-07:
> Sorry, I know this is a sidebar on this wider issue but to confirm (now I'm home): a quick and dirty set of tests on my esp32p4 kit gets 130-140MB/s (or 75-80 FPS) when flushing from the P4's on-die psram (which as I say, clocks at 200MHz and runs in HPI mode) to the MIPI peripheral.
> 
> While good for your use here this is actually disappointing, as I'd hoped for closer to 200MB/s real world performance and the PSRAM->MIPI DMA2D route is about the fastest available in hardware, so it's downhill from 140MB/s for other operations...

I'll run a quick test on this, as this is similar to how ESP did it with their Quake port, which they claim runs at ~ 20 fps. However, I have a bad feeling that this is because of the PSRAM buffer. :)

fcipaq, 2025-10-07:
> I'll run a quick test on this, as this is similar to how ESP did it with their Quake port, which they claim runs at ~ 20 fps. However, I have a bad feeling that this is because of the PSRAM buffer. :)

Maybe it can be done with the PPA after all. I ran some tests with ESP's Quake port. I checked and they're getting around 90 fps just out of the "lcd frame drawing code" (so not rendered frames, but only drawn buffers). In case you're interested [here's](https://github.com/espressif/esp32-quake/blob/fe81f2840bc658bc4b009e77b1c6302572446e46/main/display.c#L66) the key part of the code.

ducalex, 2025-10-17:
> I thought this was exactly the way I was trying to do it and I ran into trouble as soon as the stride didn't match (e.g. if an LCD window was set) - or is that what disabling the partial updates if for? What happens if RG_SCREEN_WIDTH doesn't match the buffer's stride?

My proposal would use retro-go's entire rendering pipeline, so mismatched strides or pixel sizes wouldn't matter. I think you had issues because you bypassed some of it.

Basically my proposal was just to have a framebuffer mode that would allow all apps to "just work" with your driver. And if performance is good enough then they can remain that way and if not then on a case by case you can switch apps to use your direct pipeline instead.

But I admit that the patch I shared was untested, so it's possible you saw a problem that I didn't intend ;) Maybe I need to spend time making my example work to convince you!

>> 2. It makes save states unstable (I do not have an explanation for that one but I have a few theories...).
> In what way? I observed the audio had trouble sometimes in save states, is that what you saw (or heard :).

Crashes when loading the save state, something gets badly corrupted in memory. Especially if you save in stable retro-go, then load the state in your patched snes9x (or vice versa). In my mind it's likely a pointer issue but I haven't investigated it yet.

> Ah... I now get it, it's because you're using app->frameskip. Could you please try with app->frameskip = 0 and report back how the fps "algorithm" works on the ESP32?

Ok I'll double check my code! I was pretty sure that I copy pasted your entire snes9x folder onto my branch but maybe I made a mistake.

ducalex, 2025-10-17:
> Is it possible to tell the rg system to change the resolution (so it's no longer RG_SCREEN_WIDTH), because when allocating a frame buffer after rg system was initialised, the frame buffer won't [...]

Currently no. But I think it would make sense to have a function that allows to change the resolution system-wide during run time. Thankfully very few of the code uses the constant directly so it shouldn't be hard, there isn't that much to do beyond what you already do (update display.screen.(real_)width), maybe nudge rg_gui to update some things? And maybe send an event for apps that do care about resolution, like the launcher.

fcipaq, 2025-12-07:
Hi, just a quick note: I haven't lost interest, and I'm still working on the port. I just have very little time right now. November and December are always busy at work, and I need to manually catch up with your branch because I'm quite a few commits behind.

DynaMight1124, 2026-02-21:
Looking forward to seeing the Held 2, not sure if I mentioned it but I was super close to making the Pico Held (and the few different names it had :) ) but some of the components, I think it was the SD slot, was super expensive, then the urge passed. The Held 2 looks awesome tho, lovely big screen.

A coder friend of mine has recently told me about CLI based AI clients, I'd previously just used website based ChatGPT and Gemini but only so much info/code you can give it, but CLI, you can give it an entire code base and tell it what you want and it'll just go to town, even auto compiles, gets for failures, adjusts code, recompiles and keeps going. Obviously once stuff compiles you need to test. So about a week ago, I started to see if it could get ducalex's old port of Duke3D running as I thought the P4 would have enough grunt, when I tried to compile in the latest dev on 5.5, it had loads and loads of warnings and fails. It took maybe 30mins before it compiled (with warnings) but obv didnt boot, it read the serial log, understood the issue, updated, recompiled and kept going etc etc anyway, it got Duke3D working.

So a few days ago, I tried to get it to do the same again but with Quake. Amazingly, took probably 45mins and it was displaying an image on an S3. Spent about 75% of the time trying to get the ESP32 to run but the 4MB PSRAM limit makes it very hard and I dont know if it will ever run without some extensive work moving stuff around, but the S3 does get a solid 15-25fps and the P4 even smoother.

Both of those are the first time I've seen such a divide between the OS ESP32 and the S3, I guess its mostly the much faster 8MB PSRAM but its a massive jump in both Duke3D. Pretty much double the performance.